jgyates/genmon #1: Part Numbers

madisdev, 2017-03-18:
Hey Jason,

This is awesome! I see you are missing a wiring diagram. If you want to post something simple I could mockup a fritzing(or similar) for you.

jgyates, 2017-03-18:
Thanks. A wiring diagram would be excellent. Pin 8 on the Evolution Controller is Transmit from the Evolution Controller (on mobile link this is black). Pin 7 (white on mobile link cable) on the Evolution Controller is Receive. Since I used the built in serial port on a Raspberry PI, which is a TTL signal on the expansion header, I used a level converter to get the signal to a RS-233 level. I used this one:
https://smile.amazon.com/gp/product/B00EJ9NAKA/ref=oh_aui_detailpage_o03_s00?ie=UTF8&psc=1

From there connect RS-232 transmit (DB9 pin 3) to Evolution Receive (Molex pin 7), and RS-232 receive (DB9 pin 2) to Evolution transmit (Molex pin 8).

The cable used for mobile link and my implementation do not use a ground in the cable so it is important to make sure that the raspberry pi (or other embedded device) is using a ground common to the generator battery, otherwise communication errors will result. I used this to power the raspberry pi:

https://smile.amazon.com/gp/product/B01DYE54LI/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

This allows for a common ground between the raspberry pi and the generator battery.

madisdev, 2017-03-19:
> detailpage_o03_s00?ie=UTF8&psc=1
> detailpage_o00_s00?ie=UTF8&psc=1

I'm glad I can assist. I've had this project in my head for awhile now. But
never got around to messing with the generator modbus stuff(I bought the
parts though :|)

Please bear with me I want to make sure I got the wiring correct....Im
ignoring some of the adapters I just want to get the wiring down first.

Raspberry PI RX -----> |RX-(RS232 to TTL)-TX|-----> Molex Pin 7(RX)
Raspberry PI TX -----> |TX-(RS232 to TTL)-RX|-----> Molex Pin 8(TX)

Is that it?

Madis

jgyates, 2017-03-19:
That is almost there. Here is what I would put:

Raspberry PI RX -----> |RX-(RS232 to TTL)-RX|-----> Molex Pin 8(TX)
Raspberry PI TX -----> |TX-(RS232 to TTL)-TX|-----> Molex Pin 7(RX)

This keeps the TX and RX a strait path  until it leaves the enclosure (i.e. TX from RPi -> TX level converter -> RX on Controller.

Thanks for you help. I updated the readme with some clarity (I hope).

madisdev, 2017-03-19:
Here is what I came up with. Nothing crazy. I'm not so focused on the DB9
and molex connectors, I just want to make sure people understand the flow
of TX-RX.

[image: Inline image 2]

I think in the next version I should show the molex pin outs as I'm not
sure people know which pins are which. I had seen someone on the gardenweb
discussion had posted a pic of the molex for the mobillink. Just to confirm
its 7/8 from this pic:
[image: Inline image 3]

If you agree on the wiring diagram I'd like to do a pull request to add it
to github if that's ok.
​

jgyates, 2017-03-19:
You are correct about pins 7/8 from the gardenweb discussion picture. I agree that clarification is needed on the pins. Once I got my connectors from Digikey the are marked with the pin numbers but they marking is extremely small. I have a mobile link with the cable so this helped me figure things out regarding the pin out. 

Regarding the pull request, you bet. Fire away when you are ready.

madisdev, 2017-03-19:
Cool, do you need to give me permission to do pull request or commit? I
tried an it seems like I'm either doing something wrong or I don't have
permission. Thanks.

jgyates, 2017-03-19:
I added you as a collaborator. I think that will do it. This is my 1st github project so if that does not work let me know.

jgyates, 2017-03-20:
Thanks for the diagram!

madisdev, 2017-03-21:
I see you added some pics that's cool & helpful. I'm going to update the diagram to show you using the 5v and ground from PI to the RS232 converter.

closed 2017-03-23 by jgyates