jgyates/genmon #3: Power From 8-Pin molex?

madisdev, 2017-03-23:
Hey Jason,

I'd rather do this over email, but I don't have yours :) And github doesn't really have any other method to communicate.

I had a question for you, did you check to see if there was power coming out of the generator's 8-pin molex?

I saw [this](https://www.amazon.com/Generac-Basic-Wireless-Remote-Monitor/dp/B00I9HJ6HE/ref=sr_1_2?ie=UTF8&qid=1490285593&sr=8-2&keywords=generac+mobile+link) on amazon. And it looks like it's pulling power from the 8-pin molex. Did you by any chance check the port on the generator with a multimeter? Thanks

jgyates, 2017-03-23:
My email is jason.yates (at) gmail

jgyates, 2017-03-23:
Power and ground may be on the controller molex connector. The position of the controller on my generator makes it difficult for me to validate this. If it is there then this would likely be an alternative way to power the raspberry pi or at minimum supply a common ground so a normal USB wall power supply could be used.

closed 2017-03-23 by jgyates

rhombixt, 2017-04-08:
I have found some information that might be useful. I have the generac manual for repair, shows the pinout of the 8 pin molex. Pin 1 and 2 are power according to this install manual.
http://soa.generac.com/manuals/7819446/0J3075

jgyates, 2017-04-09:
I only see photos of the cable in the link you sent. Can you send attached a picture of this diagram in the repair manual?

krisasmith, 2017-04-23:
Pin 1 is 5v and pin 2 is ground. I can successfully power my pi through the GPIO.

jgyates, 2017-04-23:
Thanks! Just curious, do you have a Nexus or Evolution controller?

krisasmith, 2017-04-23:
Evolution. It's a 2017 model: G0070430 generator.

jgyates, 2017-04-23:
If you would not mind, once you get it setup can you send me the output of a couple of commands? I have yet to validate a couple of things are working on air cooled evolution. I have a liquid cooled Evo and have worked with a couple of people  getting the nexus stuff working. 

The commands I am interested in are:
generator: registers
generator: status

Also, if you see any values that have "unknown" in the output, they are easy to fix, I just need a small bit of info, but since you have an Evolution you will likely get no unknowns. 

Let me know if you have any questions.

krisasmith, 2017-04-23:
Registers:
Num Regs: 129
Not Changed: 74126 Changed 1285 Total Changed 0.02
05f1:0000                       01f1:0000
05ed:0064                       023b:0000
023e:0000                       0059:0000
0058:0000                       0053:0000
0057:0000                       005d:0000
003c:0000                       000e:150b
000f:0416                       000a:0088
000c:0000                       0005:0a00
0006:0301                       0000:0009
0001:00000000                   0008:0000
0009:00f6                       01f2:0000
0012:0000                       0011:009c
0010:0611                       0007:0000
002a:6a75                       001a:00c7
01f4:33303031343534373630

jgyates, 2017-04-23:
wow that was fast. Thanks. I updated the wiki with the info on pin 1 and 2. 

The output looks mostly right. It looks like one thing is not quite right. You pickup voltage and battery charging status are not being detected properly. This would be normal for a Nexus controller however you said that you have an evolution. Let me look closer at the registers. I will reply with my findings shortly.. in the mean time can you post a pic of your controller front panel. Just want to validate this is an Evolution and not a nexus.

krisasmith, 2017-04-23:
If you zoom in you can see it from this picture. If you can't see enough let me know and I'll take another one.

https://goo.gl/photos/SVV1rjf53j12n4Ct5

jgyates, 2017-04-23:
yep, that is an evolution. Looks like there are a couple of things that are not working on air cooled evolution models. Right now the software thinks you are running a Nexus model because the code I put in for autodetecting nexus vs evolution made an assumption about air cooled nexus models. You should have a lot of functionality. I can probably get the rest of it working if you would be willing to help with some data points. Here is what is not working on your system:

Alarm Log is not working
The ability to read the onboard sensors and output relays is not working, this includes the state of the transfer switch relay and the battery charger.

Based on the number of registers you are returning it looks like you may be out of date on the software by a about a week. I updated the number of registers that are being read. If you can update your software and give me another "generator: registers" command that would be helpful.

Once you update your software, try this:
modify your genmon.conf file to uncomment this line:

#evolutioncontroller = True
change to:
evolutioncontroller = True

Reload genmon.py and then send the "generator: registers" output again.

If for some reason you get notices about transfer switch status changing out once you reload you can re-comment that line. I don't think you will just being cautious. You transfer switch will not be changing, just the software is reading the wrong registers.

The core issues is that the evolution controller has two variants, one for liquid cooled and one for air cooled. Since I don't have a liquid coold to test one I have to do it like this :)

The good news is that I built in some options for debugging registers so we have some thing to work with.

Of course if you don't have time to do any of this I understand. 

Thanks
Jason

krisasmith, 2017-04-23:
Happy and willing to help. I'll pull the new code tomorrow and send you the output of the test.

Do you want me to continue to post to this thread or contact you through a different method?

jgyates, 2017-04-23:
Thanks for helping. you can contact me at jason.yates@gmail.com

rhombixt, 2017-04-23:
I have a 2017 air cooled Honeywell. I assume I would be missing these functions as well. Will it be difficult to fix?

jgyates, 2017-04-23:
Probably not hard. I have the code instrumented to help debug this type of thing. I have a mode I can enable in the software that collects all the registers and logs changes, then every 10 min it will spit out an email with the changes to the registers. This is how I got it working on my gen and the Nexus model (thanks to Jeremy who let me run the test on his Nex system). The test just reads mod bus registers so it is not invasive. 

I have another mode in the code that will send and email when a specific register changes value.

the general approach is to run the first mode mentioned above, do something meaningful with the generator or transfer switch, look for registers that change, then run the second test.

The more data points I get the more solid we can make this, for example I am pretty sure register 0000 is a register that tells the model but I only have four data points (and two of them are nexus).

ts4s, 2017-04-23:
Hello krisasmith, i'm gonna be noisy here for a minute and ask about the pi power method you're using. I'm assuming you are using pin 2 or 4 for the 5 volts and one of the ground pins to power the Pi? 

I have to say its nice to see they went to one slab instead of 2 for setting the unit on. I have an old model like yours on 2 square slabs.

jgyates, 2017-04-23:
I deleted my email from the thread. If you are following the thread feel free to contact me. I just don't want a bunch of spam bots picking up my email :)

jgyates, 2017-04-23:
I created a new issue for the Evolution Liquid Cooled problem kris is seeing on his generator.

krisasmith, 2017-04-23:
@ts4s - I am using GPIO pins 2 & 14 on the pi connected to pins 1 & 2 on the accessory port of the evolution controller.

jgyates, 2017-04-23:
good to know. I added that to the wiki hardware page in the power supply paragraph.

krisasmith, 2017-04-23:
You got a typo on the wiki: https://github.com/jgyates/genmon/wiki/Making-a-Cable

"Pin 7  - Received to the Controller, connect to Receive on the Pi"

I believe that should read "... connect to the Transmit on the Pi"

jgyates, 2017-04-23:
Thanks!  Fixed now.