jgyates/genmon #6: Not really an issue, but a wuestion. :-)

kfonda, 2017-04-06:
Hi Jason, I just recently stumbled across you project and it looks great. I have a question/idea that you might be able to help me with. How hard would it be to use an esp8266 serial to WiFi link on the generator and run everything else on a remote Pi or other linux computer? Esp modules like the esp-07 or the Wemos d1 mini pro have external antenna connectors and can be had for less than $5.00 It should make development easier since once the serial link is set up, everything else would be remote. 
Either way, I intend to set up you software this spring. Keep up the good work and thanks for the effort. There was no way I was going to pay Generac 2 or 3 hundred dollars for something like this. :-)

jgyates, 2017-04-06:
Hi Kevin,
As long as you can use python on the esp8266 you should have not problems as far as I can tell. The project was designed to run genmon.py on one system and the web interface and console app on another system (or the same one if you prefer). The low level requirements are a tcp stack (for email and socket comms) and serial. If pyserial works on esp8266 you could be golden. Once you get to setting things up let me know how it goes.

Thanks
Jason

kfonda, 2017-04-07:
Thanks for the quick response. I know MicroPython runs on the 8266. I'll have to do a little more research. I'll let you know how it goes when I get things going.

closed 2017-04-07 by jgyates