jgyates/genmon #8: A little lost

rhombixt, 2017-04-07:
As a first time PI user.. I am very confused as to what is needed to make this work.
I have the PI setup and installed
I cloned the repository and configured the conf files and moved them to the etc directory
I followed the instructions to setup the PI port and enable the uart
I installed minicom
I installed crcmod
I installed pyserial
all the modules are in pi/genmon
so now i need to write a script to run genmon and do I need to run the others a well for the web interface? How does the web interface work or whats needed for that? I assume install the Flask 012 software. does the script needs to run genpio and genserv,,,

can you help me as I have never used or setup a pi or anything before

jgyates, 2017-04-08:
Hi,

I would run the program to validate everything works with your setup before putting it in a script.

To launch the program you will need to run it as root (administrator). To do so you would type the following:

sudo python genmon.py

to launch the program as a background process you can type:

sudo python genmon.py &

sudo launches programs as root, you may be prompted for the root password or you may not, depending on linux is setup.

If you serial is working, your cable is working, you have your config files setup, then it should just run. To see any output you can either make settings in genmon.conf display the output to the screen, or use ClientInterface.py.

You can run ClientInterface.py on another system on your network, assuming you have python installed. Python is installed on linux and mac, but you will need to install it on a windows system.

you can run ClientInterface.py by typing

python ClientInterface.,py IP-ADDRESS-OF-YOUR-RASPBERRY-PI

you will need ClientInterface.py,mylog.py, myclient.py (assuming you have the latest release, myclient.py was added in the last few days) in the same directory.

or you can run the test program on the same system as your pi
python ClientInterface.py localhost

you will need to run this on a second terminal console or run genmon.py in the background.

Once you validate that it is working you can setup to where it runs at boot. crontab is a the method for doing this. I can point you in that direction once you get the rest validated and working.

Also, to stop a background python program type:
sudo pkill -f NAMEOFPROGRAM

for example:
sudo pkill -f genmon.py 
will stop an instance of genmon.py started with this command: sudo python genmon.py &

If you do not run the python program in the background (i.e. without the & at the end) then you can just type Ctrl+C to exit or type "sudo pkill -f genmon.py" from another terminal window.

you can put the above commands in a script and then run the script.

for example I have a script named startgen.sh on my system. It is called via crontab on boot. The script contains the following:

#!/bin/bash    
sudo python /home/pi/genmon.py  &	
sleep 10
sudo python /home/pi/server/genserv.py &

I hope this helps. Let me know if you have any other questions.

rhombixt, 2017-04-08:
One more question .. the link to the modbus appears to connect to the rs233 to ttl adapter. Is the modbus actually using rs233?

jgyates, 2017-04-08:
Yes. RS-232 is the electrical interface, modbus is the protocol or format of the bits and bytes.

jgyates, 2017-04-08:
one other note to your original questions: You can run genmon.py (with it's requirements or pyserial, crcmod, mymail.py, mylog.py) alone. All other programs (genserv.py, gengpio.py, ClientInterface.py) are optional. genmon.py will respond to email commands and send email regarding the generator status. I would recommend using ClientInterface.py to validate things are working as it is just a simple way to communicate with genmon.py. Once that is setup if you want to have a web interface, then setup genserv.py with Flash. gengpio.py is only used if you attach LEDs or relays to the raspberry pi GPIO headers.

rhombixt, 2017-04-08:
What do you mean "setup genders.py" with flash? How do you do that? I do want to run all those programs together so do I need separate script for each to run in the background or is ther a way to run them all together wth one script? What would that look like?

jgyates, 2017-04-08:
Sorry, I meant to type Flask instead of Flash. The web interface requires you install the package Flask (http://flask.pocoo.org/). I reference this in the section on genserv.py in the wiki and readme. You follow the directions on the Flask site to install Flask. Then you just run "python genserv.py &". I copied and pasted the only script needed in the above post:
<begin script>
#!/bin/bash
sudo python /home/pi/genmon.py &
sleep 10
sudo python /home/pi/server/genserv.py &
<end script>

The above script will run both genmon.py and genserv.py in the background. The "sleep 10" just waits 10 seconds between loading genmon.py and genserv.py. The line #!/bin/bash is required for any bash script on Linux.

Flask is a framework that allows a web site to be run from a simple python script. It is much easier than setting up a full blown web server. According to their web site you should install Flask by typing "sudo pip install Flask". If memory serves it may ask you about virtual environments, I opted not to use virtual environments for reasons I go into in the readme.

On your raspberry pi type:

sudo apt-get update
sudo apt-get install python-pip
sudo pip install Flask

This should setup Flask. Then you can just run the above script to execute genmon.py and genserv.py.

In order I would set things up would be:
1) Validate your serial cable by performing a loopback test through the cable by connecting the cable to the raspberry pi and connecting the far end of the cable's tx and rx together. Use minicom to perform a simple test that transmits and receives. This test is mentioned in the read me under "Setting Up Your Raspberry Pi (Serial Port)". If this test is successful then connect one end of the cable to your generator and the other to your raspberry pi.
2) Modify /etc/genmon.conf and /etc/mymail.conf per the specifics of your generator and email account
3) Run "python genmon.py" and validate your setup is working by connecting to it from a separate computer. You can connect to genmon.py on your Pi by running "python ClientInterface.py <your raspberry pi IP address>" in a separate console window or from another computer.
4) Once all of this is working then then you can install Flask and run "python genserv.py". From there you should be able to pull up the web page as described in the documentation.
5) After the above steps are working, then create a single script to run at boot time.

Without performing these steps in the order above you will have a lot of work trying to figure out what is wrong if your did not setup something correctly.

rhombixt, 2017-04-08:
I have compiled a set of point instructions for the software install. Let me know if anything is a miss:

•	Assemble the Raspberry Pi 3 and all the physical connections as per the schematic
•	Down load SFD Formatter https://www.sdcard.org/downloads/formatter_4/ onto your PC or laptop and install this utility for setting up the mini SD Card.
•	Insert the SD card into your PC and format the mini SD Card
•	Once complete, leave the SD card in your PC and go to https://www.raspberrypi.org/downloads/ and select the NOOBs download and extract the files from the zip. Copy all the files from the extracted folder to the SD Card
•	Eject the SD card from your PC or Laptop and insert into the PI ensuring that you have and HDMI monitor, usb keyboard and mouse connected to your unit. Power up the PI and let the installer setup the Rasbian PI software. FYI – this takes a while and wait for the OK window to pop up to confirm the install was successful. Unit will reboot
•	Connect network patch cable to the PI unit and setup your timezone, time and date. Click on terminal window >_ and then type the following:
1.	sudo raspi-config.
2.	Select Internationalisation Options.
3.	Select I2 Change Timezone.
4.	Select your Geographical Area.
5.	Select your nearest City.
6.	Select Finish.
7.	Type Reboot at the terminal window.

•	You need to setup the serial port on the PI 3. Open the terminal window again and type sudo nano /boot/config.txt and add the following line at the bottom: enable_uart=1 and then press cntrl-o to save the file 
•	Leave the terminal window open and confirm the name of the serial port of ttys0 has the alias serial0. Type ls –l /dev and confirm. 
•	Disable the console function of the serial port by typing : 
sudo systemctl stop serial-getty@ttys0.service
sudo systemctl disable serial-getty@ttys0.service
•	Now remove the console from the command line boot file. Type sudo nano /boot/cmdline.txt and remove the text “console=serial0,115200. Press cntrl o and write the file
•	At the terminal window now type reboot
•	Install Minicom to test the serial port. Open a terminal window and type: sudo apt-get install minicom
•	Now configure minicom . On a terminal window type sudo minicom –s –c on . change the serial config by highlighting with the arrow keys and edit the serial port to be /dev/serial0 and then save the settings as DFL . (if this fails, make sure you are running with sudo command)
•	Go to the https://github.com/jgyates/genmon and click the CLONE or download button. You can also just transfer the files via USB after downloading to the pi directory. Create a folder called genmon and paste. The other more efficient way is to click the clipboard icon to copy the link and then open a terminal window and type:
 sudo git clone https://github.com/jgyates/genmon.git
This will transfer the files to the PI and create the directory
•	Edit the .conf files as required and copy them into the /etc directory via the terminal window. Open a terminal window and change directory to /pi/genmod and then copy the .conf files to the /etc directory. Type sudo cp *.conf /etc
•	Now go to https://github.com/gsutil-mirrors/crcmod and either download and transfer the files to /pi/crcmod directory or use the terminal windows  sudo git clone https://github.com/gsutil-mirrors/crcmod.git
•	Now install the application via the terminal window: sudo python setup.py install
•	Now install the pyserial application via the terminal windows: sudo apt-get install python-serial . Once completed test – type  python -m crcmod.test
•	Go to http://flask.pocoo.org/ and download the latest release , in this case 0.12
•	When completed type sudo apt-get update
•	When completed type sudo apt-get install python-pip
•	When completed type sudo pip install Flask
•	Now test the serial connection with minicom to make sure it works
•	Open a terminal window and test genmon py : sudo python genmon.py
•	You can change the genmon.conf file : displayoutput = true and then change displaystatus = true and displaymonitor = true to be able to validate the data coming from the genset Modbus for testing
•	To run the genmon.py in the background and leave it running: sudo pyton genmon.py &
•	To stop the program from running in the background sudo pkill –f genmon.py
•	To use the client to interface with the application: python Clientinterface.py localhost (on the machine)  or if remote python Clientinterface.py <ip address of the raspberry pi> (don’t use DHCP – assign static IP address if using this method)
•	Now a script needs to be made that allows all these applications to run together
Open a terminal window to the genmod directory. In the directory type: nano startmon.sh
Cut and paste this into the script or re-type it:
#!/bin/bash
sudo python /home/pi/genmon/genmon.py &
sleep 10
sudo python /home/pi/genmon/HTTP/genserv.py &
Sleep 10
Sudo python /home/pi/genmon/OtherApps/genpio.py &
**or if you have arranged all the *py files into a different directory, then you can reference that directory above in the script**
Press Control-X and return to save
•	Now make the script execute able : chmod 775 startmon.sh
•	Now test by typing sh startmon.sh – script should run and test with clientinterface.py
•	Install Crontab gui: sudo-apt get install gnome-schedule to install the task scheduler gui . This will install a gui based scheduler on your raspberry start programs. Open this application to add the startmon.sh to run every time the unit reboots. You can also skip this step and use the crontab without the gui via command line. 
From a terminal window:
First create a logs directory in home/pi called logs. Navigate to home/pi and mkdir logs
Now type: Sudo crontab –e 
From the contab window, add the following to the script
@reboot sh /home/pi/genmon/startmon.sh >/home/pi/logs/cronlog 2>&1
And save the crontab schedule.

rhombixt, 2017-04-08:
One other item - not sure if this as easy to compile or do - a manual on/off  via GPIO would be handy.
If a GPIO x is grounded, the generator starts (sends the command to the controller to start)
If a GPIO y is grounded, the generator stops (sends the command to the controller to stop)

jgyates, 2017-04-08:
Hi,

The write up of the setup is great. I will likely add that to the wiki. After a quick read two minor thing I noticed that we may want to change if it goes in the wiki. 

1) NOOBS is just one choice, I use Debian Jessie Lite because it is smaller but others may want to use NOOBS. We may just want to give anyone reading this a few options rather than implying that one is required.

2) If memory serves, is you use pip to install Flask (e.g. sudo pip install Flask), pip should download it for you from https://pypi.python.org/pypi. pip is sort of like apt-get in that it automatically downloads, installs and configures packages, except pip is just for python modules.

As for the GPIO question, yes that should be easy to add to gengpio. I will add that in a few days. I have a couple of other items on the list to get to first.

Were you able to get things working? If so what type of generator do you have? Just looking for some data points to validate coverage.

rhombixt, 2017-04-08:
It appears to be working - this is just the PI board at the moment. So far no errors, web page works - no data of course. I am still waiting on parts to complete the fabrication. My actual new Honeywell/generac 11kw is in my garage and I will install it next month or in June to replace my older 2006 unit that was all switch and relay driven (very easy to add external monitors, controls etc). With my new system, I can't use the normal monitoring solution because I live in Canada and they killed the CDMA protocol here. This application gives me the ability to do all the things I used to do that I was going to loose on the swap out. Once the new genset is installed, I will just plug my fabricated PI unit in and test it  and then re-connect the automation controls.
 I thought of one more input that would be handy - the ability to just enable the generator and disable it via GPIO - GPIO w is grounded, its like pressing the auto button, gpio z is grounded, its like pressing the OFF button.
So in total 4 inputs - start input /stop input / auto button / off button
Right now, with my home automation system I can start/stop/ take the genset offline and online
[image]

jgyates, 2017-04-09:
Thanks for the info. 

Regarding the GPIO inputs. I don't presently know a way to set the state of the enable / disable via the modbus registers. Right now all I can do is start / stop / start and transfer / start quiet mode.

Sounds like you have a very nice setup, I like home automation as well.

rhombixt, 2017-04-09:
No problem .. just a wish not a need at this point. The system is connected to an hai controller run by a touchscreen - hai Omni pro that's all home brew.
[image]

rhombixt, 2017-04-09:
as an FYI- I believe that based on this information below, pin 1 and pin 2 are power and ground. I just need to confirm with a DVM which connection is which.
[image]
[image]
[image]

jgyates, 2017-04-09:
Sounds good. Let me know if you get a chance to validate. I would do this myself but on my generator (48wk diesel) the controller would need to be removed to probe these pins, it is quite difficult to reach.

madisdev, 2017-04-09:
Hey guys. I took some measurements today on my 20kw. It looks like there is 5v on a couple of the pins. I will report in later with numbers.

madisdev, 2017-04-09:
OK on my evolution controller there are voltages i recorded. Please check for yourself before doing any wiring. I am not responsible for anyone messing up their generator.

<table>
<tr>
<td>Pos Lead Pin</td>
<td>Neg Lead Pin</td>
<td>Voltage</td>
</tr>
<tr>
<td>1</td>
<td>2</td>
<td>5v</td>
</tr>
<tr>
<td>4</td>
<td>6</td>
<td>8.5v</td>
</tr>
<tr>
<td>1</td>
<td>6</td>
<td>13.5v</td>
</tr>
<tr>
<td>1</td>
<td>4</td>
<td>5v</td>
</tr>
<tr>
<td>1</td>
<td>5</td>
<td>5v</td>
</tr>
<tr>
<td>2</td>
<td>5</td>
<td>0v</td>
</tr>
<tr>
<td>2</td>
<td>6</td>
<td>8.5v</td>
</tr>
<tr>
<td>1</td>
<td>-batter term</td>
<td>5v</td>
</tr>
<tr>
<td>-batter term</td>
<td>4</td>
<td>0v</td>
</tr>

</table>
From the manual it states pin 4 is ground and 6 is power. But i'm not sure since it just shows the mobile link connector and not the evolution. I'm not sure if they map accross to the same number pins on the evolution controller 8pin.

rhombixt, 2017-04-09:
I assume this pinout you are testing is on the evo controller itself . The picture above is looking at the controller itself from the repair manual. Can you humor me and put you meter on the ground of the battery positive lead to  2 and then do the same ground  and positive to pin 1. repeat on the +lead of the battery. The pins don't map across .. It does look like pin 4 is a ground which makes me wonder how the remote monitor works. that's the wiring picture from above.

rhombixt, 2017-04-09:
I'm not so much about powering the unit from the connector - rather I want the common ground so I can tie the serial connector in with 3 wires properly vs just 2.

madisdev, 2017-04-09:
Ok here is what i got. Hope it helps.

<table>
<tr>
<td>Neg Lead</td>
<td>Pos Lead</td>
<td>Voltage</td>
</tr>
<tr>
<td>-battery</td>
<td>1</td>
<td>4.99v</td>
</tr>
<tr>
<td>-battery</td>
<td>2</td>
<td>-1.4v(counts down to)</td>
</tr>
<tr>
<td>+battery</td>
<td>1</td>
<td>-9.03v</td>
</tr>
<tr>
<td>+battery</td>
<td>2</td>
<td>14.03v</td>
</tr>
</table>

rhombixt, 2017-04-09:
I bet the wireless monitor controller works on +5v. It appears that pin 1 is +5v , and pin 2 is negative. 
I see your previous results for pin 4 are above - that looks to be like a true ground. If you put your neg lead on pin 4 and + lead to the battery - do you get 13.5 - 14v?

rhombixt, 2017-04-09:
and if you measure between pin 2 and pin 4 it should be near 0v.

jgyates, 2017-04-09:
you may get as low as 12 out of the battery. If the charger is active it should be above 13.

Also, rhombixt, I put your setup guide in the wiki:

https://github.com/jgyates/genmon/wiki/Setting-up-your-Pi-and-genmon

Thanks!

madisdev, 2017-04-10:
I found this on Lowe's website. If you look at the image zoomed in it looks like its using pins 1 & 2 and 7 & 8.

https://www.lowes.com/pd/Generac-Basic-Nexus-Wireless-Remote-Monitor-Harness-Adapter-Kit-for-Standby-Generators/3558716

jgyates, 2017-04-10:
the description says it is for nexus controllers. I wonder if there is another cable for evolution controllers. If so then that would mean that the power and ground may be different between the two controllers. I don't know how likely that is that there are two, if so it would appear to be a bad design choice if Generace did change the power and ground between Nexus and Evolution. I am betting they did not and this cable works on Evolution as well.

rhombixt, 2017-04-10:
I am almost convinced that the 8 pin monitor can supply 5v for the wireless monitor kit. For the Mobile link, they don't use these connections because the mobile link needs 12v hence the reason they have a break out of the harness to connect to the +12v battery and ground. I don't think the connector is different. I will do a continuity test on my new Honeywell that's in my garage at the moment to confirm

madisdev, 2017-04-10:
I am working on a personal project similar to what is going on here, but I'm working on doing it through a ESP8266. I'm planning on using the 5v from the 8pin molex and stepping it down to 3.3v. Ill let you know how it goes.

jgyates, 2017-04-10:
Hey Madis, feel free to use this repo if you think it matches your needs if you end up changing anything in the code and want to share your updates. I have never done this but I think we can create sub projects.

madisdev, 2017-04-10:
Will do, i'm not looking to steal your thunder. Just want to explorer a bit on my own. But the work you did will def be helpful. 

I have a project plan :) 
First step was to verify I can create a UART-to-TCP relay on the ESP8266. I have that working in theory, I can send and receive from the UART using a loopback.

Second step is to create my circuit. This is almost completed as i just got my dc-to-dc buck converter the other day. Once I connect that to the ESP8266 and the <a href="http://www.ebay.com/itm/1x-MAX232-RS-232-MAX3232-to-Serial-UART-TTL-Voltage-Converter-Module-3-3V-5-0V-/131840924017?hash=item1eb254d971:g:TJUAAOSwnFZXVhfv">MAX232</a> I should be able to put it in the generator. 

Hopefully this weekend. From there ill either make a program that connects to the TCP or look at writing the modbus code on the ESP8266.

JeremyStevenson, 2017-04-10:
> ESP8266 and the MAX232

I don't know if this helps anyone in their thinking but instead of getting
the 12v adapter recommended by jgyates I just got a 12v car usb adapter,
broke it open, soldered some leads onto it and then connected to +12V and
ground on the generator connection block. The USB car chargers are circa $8
each.

I suppose if 5V isn't available on the Molex connector but some other power
supply is, you could use something like this:
https://www.amazon.com/dp/B01GJ0SC2C?psc=1

madisdev, 2017-04-10:
Jeremy, 
That is actually the buck converter that I got to bring down the 5v to 3.3v for the ESP8266. Worst case scenario is I will get power off the battery like you mentioned.

rhombixt, 2017-04-11:
I bought this device:
http://www.ebay.ca/itm/Car-Charge-DC-Converter-Module-12V-To-5V-Micro-USB-Output-Power-Adapter-3A-15W-D-/172119968417?hash=item2813264aa1:g:cQ4AAOSwYlJW1rso

jgyates, 2017-04-11:
I am sure that will work fine. I ended up going with the one that I did because it had and inline fuse. The adapter that came with mobile link had a fuse as well so I just went that direction. Plus my mechanical skills are a bit lacking. I can solder but it never looks that great when I am finished. :)

rhombixt, 2017-04-11:
Yes, that's why I like the idea of NOT using the molex for power... in case something gets shorted etc. it's wiser to be connected a segregated fuse assembly. I should be able to get the unit assembled this weekend for install into my automation panel.  How are the inputs coming along on the GPIO for start / stop etc - i assume it's a switch to ground to trigger them?

jgyates, 2017-04-11:
I should be finished tomorrow or wednesday. this page has an input circuit listed at the top that will work:
http://elinux.org/RPi_GPIO_Interface_Circuits

rhombixt, 2017-04-11:
Simple enough... I will build a small pull up circuit and switch an use an opto iso board to switch the grounds.

jgyates, 2017-04-11:
FYI: I will be using these GPIP for input:

RPI Header pin 11,  STOP GPIO 17
RPI Header pin  13  START GPIO 27
RPI Header pin  15   START/TRANSFER GPIO 22

Also, I am setting the broadcom chipset of the raspberry pi to use the internal pull up resistors on each input so you don't need to add them.

The GPIO library does have the concept of a software debounce, but it is not included with the default package on raspbian. It have the debounce stuff in the code but it is commented out (easy to add back in). Unless you update your RPI.GPIO library you may want to add a debounce to the switch (add a 0.1uF cap across the switch).

I have a first cut finished. I will check it in a little later and add some info to the readme and wiki. I put the input stuff in separate source file (gengpioin.py) so if people want only input or output they can pick and choose.

jgyates, 2017-04-11:
I had some time tonight so I just added gengpioin.py and updated the wiki and readme for each GPIO program. I also made a small tweak to gengpio.py so you will want to refresh that one also. 

I have only done partial testing (ran each of them on a PI, but I don't have the LED or switch wired in. I simulated execution so I know the commands issued to genmon.py are working. The GPIO interface is pretty simple so it will likely work.

rhombixt, 2017-04-12:
Question: will the i/o "poll" for status on boot? Let's say I reboot the board and the program starts. Will the gengpio.py program ask for status to set the output accordingly?

jgyates, 2017-04-12:
If I understand your question, you are asking if gengpio.py and gengpioin.py will start on reboot?

If that is your question then, yes they can. You just need to have the program start as a background app on boot. I have a crontab script that loads the programs on boot. 

The script that gets called by crontab looks like this:

   sudo python /home/pi/genmon.py  &	
    sleep 5
    sudo python /home/pi/HTTP/genserv.py &
    sudo python /home/pi/OtherApps/gengpio.py &
    sudo python /home/pi/OtherApps/gengpio.py &

Note the last three entries are optional. genserv, gengpio and gengpioin are only needed if you want their functionality but they all require genmon.

The & makes it run as a background process. The crontab makes it run when the system boots.

This page in the wiki describes setting up the system along with crontab entries and a startup script.

https://github.com/jgyates/genmon/wiki/Setting-up-your-Pi-and-genmon

If this is not your question or if it is still unclear let me know.

jgyates, 2017-04-12:
Sorry, just dawned on me you wrote that wiki entry. :)

If the & at the end of running gengpio and gengpioin does not answer your question let me know.

rhombixt, 2017-04-12:
No.. the question is if I reboot the board which will cause all the gpio to change to logic low and then the board comes back online. Will the program poll the generator to turn the gpio for the "ok" green status to be on (logic high)? Or will the board not update until a modbus message comes accepts that the generator is ok?

jgyates, 2017-04-12:
modbus registers are continuously polled so every register gets updated about once or twice a second within gernmon. The support apps (including the gpio apps) poll genmon about every 2-5 seconds, it varies between apps and what they are polling. Short answer, as soon as you reload gengpio it will reinitialize the gpio and poll for status.

rhombixt, 2017-04-12:
Great - sounds good. If I wanted to , I can break out the IO for specific alarms to different GPIO's using your code as an example -

jgyates, 2017-04-19:
Closing issue. Please create a separate thread for any follow up on the issues in this thread.

closed 2017-04-19 by jgyates

rhombixt, 2017-04-19:
Just waiting on parts now from china.

Closed #8<https://github.com/jgyates/genmon/issues/8>.