jgyates/genmon #9: Cabling question

griffths, 2017-04-08:
Could the following cable be substituted for the serial to TTL adapter by crimping the correct Molex pins on the end of this cable?  It would seem to be a cleaner solution than using the TTL to serial adapter, serial to Molex adapter cable then the cable with Molex connectors on both ends.

This is assuming that the program could be modified to add an option to look as USB rather than serial or maybe by installing a USB to serial driver.

https://smile.amazon.com/dp/B010K5NC68/ref=pd_luc_rh_spmrai?psc=1&dpSrc=huc&preST=_SX300_QL70_&dpID=41rPpFgEBBL

Regards,

Jay

rhombixt, 2017-04-08:
The Generator puts out RS232 Modbus. That cable looks like it converts TTL to USB. The RS232 adapter converts the signal to TTL to connect to the PI.

jgyates, 2017-04-08:
You can use a usb adapter instead of the onboard port. I don't think this particular one will work however. It looks like the output is on the non-usb side is TTL voltage levels. The generator controller is expecting non-TTL voltage levels. It is expecting RS-232 voltage levels. 

To use a USB adapter that outputs RS232 the only change you would need to do on the software side would be to change the serial port defined in genmon.conf. I used a USB serial port in some of my initial testing. USB serial ports should show up /dev/ttyUSB0, so you would change the following lines in genmon.conf:

# the serial device name of your serial port. Normally  /dev/serial0 or
# /dev/ttyAMA0 for onboard Raspberry Pi Serial ports (required)
port = /dev/serial0 .     # change this

port = /dev/ttyUSB0 . # to this

griffths, 2017-04-08:
Thanks for the tips and clarification.

 

So would creating a cable that goes from Molex directly to DB9 work?  I am unclear why you have an intermediate cable to appears to go Molex => Molex then Molex => DB9.

 

Regards,

 

Jay

griffths, 2017-04-08:
Thanks for the clarification.

rhombixt, 2017-04-08:
Molex to molex was just for having an easy 2 pin disconnect at the case. You could go right to a DB9 and then use cat5e and a hood like me.
http://www.mycablemart.com/store/cart.php?m=product_detail&p=6406&gclid=CjwKEAjwlKLHBRDztKr6wMnRthMSJAALcT-s08laY7MFhfA1BSxxfpQxSWaNP1p-SHeJ9s618FCH3xoCOkzw_wcB
-- the only part I don't know distance wise is how good the line driver is. RS232 should be able to do 50ft but I might go with an rs485 to 232 adapters to make things work better

griffths, 2017-04-08:
Thanks for the information and link.

 

Regards,

 

Jay

 

http://www.mycablemart.com/store/cart.php?m=product_detail <http://www.mycablemart.com/store/cart.php?m=product_detail&p=6406&gclid=CjwKEAjwlKLHBRDztKr6wMnRthMSJAALcT-s08laY7MFhfA1BSxxfpQxSWaNP1p-SHeJ9s618FCH3xoCOkzw_wcB> &p=6406&gclid=CjwKEAjwlKLHBRDztKr6wMnRthMSJAALcT-s08laY7MFhfA1BSxxfpQxSWaNP1p-SHeJ9s618FCH3xoCOkzw_wcB

jgyates, 2017-04-09:
Closing this issues as the original questions has been answered

closed 2017-04-09 by jgyates