jgyates/genmon #10: Compatibility Testing (request for input)

jgyates, 2017-04-09:
This is a request from the author of this code to anyone using it for some information to help make the software more robust.

 From ClientInterface.py please type the following command: "generator: getregvalue=01f4". This will return the contents of the register that holds your generator serial number. If you would not mind please post this value along with your model number to this thread. This information will allow me to validate the format of register 01f4 and hopefully allow me to decode the format of Generac serial numbers. If we can figure this out it may make the setup / configuration easier since the software could autodetect your generator type.

Thank
Jason

JeremyStevenson, 2017-04-09:
OK : Auto, Off - Ready
>generator: getregvalue=01f4
30303037373232393837
>

jgyates, 2017-04-09:
Thanks Jeremy! I updated the above comment for anyone else who would like to help out. Going forward please send your model number along with the contents of register 1f4 as described above if you can.

sgtmaj27, 2017-05-27:
Model Generac 5870. Nexus controller

Registers:
Num Regs: 158
Not Changed: 12488 Changed 2663 Total Changed 0.21
0038:fffe                       0034:ff3e
0037:12b6                       0032:4000
05f5:03fb                       05f4:03fb
05f1:03fb                       01f1:0c0a
05ed:0c0a                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:03fb
0052:03fb                       0057:0000
0056:0000                       0055:0c0a
0054:0000                       0208:03fb
005c:03fb                       005a:03fb
005f:03fb                       005d:03fb
003c:0000                       000d:0000
000e:0b2d                       000f:051b
000a:008c                       000c:005a
0005:0a00                       0006:0600
0000:0003                       0001:00000000
0008:0000                       0009:00e8
01f2:0c0a                       0012:0000
0011:009c                       0010:0611
0007:0000                       0019:0000
002b:0000                       002c:0000
002a:646b                       002f:0000
002e:0000                       0020:43e9
0021:0179                       001f:00b1
001e:00b1                       001d:43e9
001c:007f                       001b:43e9
001a:00b1
01f4:30303036303430303632

rhombixt, 2017-06-25:
This is for a new 2017 Honeywell 11kW Model G0070580 with Evolution Controller

Registers:
Num Regs: 208 
Not Changed: 1606 Changed 12 Total Changed 0.01
0038:0001			0034:0000
0037:10a5			0032:4000
05f5:0000			05f4:0000
05f1:0000			01f1:0000
05ed:0064			023b:0000
023e:0000			003b:0000
05fa:0000			0059:0000
0058:0000			0053:0000
0052:0000			0057:0000
0056:0000			0055:0000
0054:0000			0208:03f1
005c:0000			005a:0000
005f:0000			005d:0000
003c:0000			000d:0000
000e:130a			000f:0618
000a:007f			000c:0001
0005:1000			0006:0000
0000:0009			0001:00000000
0008:0000			0009:00f0
01f2:0000			0012:0000
0011:009c			0010:0611
0007:0000			0019:0015
002b:0005			002c:1000
002a:6a75			002f:0000
002e:0000			0020:0000
0021:0000			001f:4956
001e:018e			001d:0000
001c:0000			001b:467c
001a:00c6			
01f4:33303031323930373335

drandyhaas, 2017-07-15:
Generac 10kw air-cooled (sorry, don't know model number), Nexus controller
01f4:30303036333330383237
PS - you are my HERO for making this software! I'm interfacing it to SmartThings!

JeanB51, 2017-08-07:
Hi.
I just got my genmon connected and running. Here is my info:

Model : Generac QT036, 36KW, water-cooled, Nexus controller, Propane fueled. Installed 2014.

I'll post some unknown alarm log codes in the #13 issue.

Here is the output from registers;

>generator:registers
Registers:
Num Regs: 158
Not Changed: 6483 Changed 858 Total Changed 0.13
0038:01f4                       0034:01f4
0037:01f4                       0032:01f4
05f5:001e                       05f4:0032
05f1:01f4                       01f1:0010
05ed:0018                       023b:0000
023e:0000                       003b:0000
05fa:01fe                       0059:01f4
0058:01f4                       0053:03f2
0052:03f2                       0057:01f4
0056:01f4                       0055:01f4
0054:01f4                       0208:03f2
005c:01f4                       005a:01f4
005f:01f4                       005d:01f4
003c:001e                       000d:0000
000e:0a3b                       000f:0807
000a:008c                       000c:0036
0005:0700                       0006:0200
0000:0006                       0001:00000000
0008:0000                       0009:0000
01f2:031d                       0012:0000
0011:0090                       0010:0111
0007:0000                       0019:0000
002b:0000                       002c:0000
002a:65d4                       002f:01f4
002e:01f4                       0020:0000
0021:0000                       001f:46bc
001e:00f9                       001d:4498
001c:0031                       001b:454f
001a:0063
01f4:30303038353635333133

JeanB51, 2017-08-07:
Further info and a question.
Here is the output of the status and maint commands.
My question: Why does it show 0V for the Line State Utility Voltage ?

>generator: status maint
Status:

Engine State:
    Switch State: Auto
    Engine State: Off - Ready
    Battery Voltage: 14.0V
    RPM:     0
    Frequency: 0.0 Hz
    Output Voltage: 0V

Line State:
    Utility Voltage: 0V, Min: 0V, Max: 0V
    Utility Threshold Voltage: Low Voltage: 144V

Last Log Entries:
    Start Stop Log: 07/31/17 10:39:42 Stopped - Auto
    Alarm Log:      07/31/17 10:35:24 Unknown 0x01

General:
    Monitor Time:   Monday August 7, 2017 12:01:12
    Generator Time: Monday August 7, 2017 11:43
Maintenance:

Generator Serial Number : 0008565313

Exercise:
    Exercise Time: Tuesday 07:00 Quite Mode Off

Service:
    Service Scheduled: Next service in 99 hours
    Total Engine Run Hours: 54
    Firmware : V2.12, Hardware : V1.01

JeanB51, 2017-08-07:
Finally, I want to say how much I appreciate this project you have done. 
I had been searching for something similar, when I did not want to go with the company's cellular monitoring solution.
Very nice project - I read the python code, it is very neat work - love the threading stuff!
Thanks very much !
Jean B

jgyates, 2017-08-07:
Hi Jean B,
Thanks for the kind words. 
I created a new issue for the utility voltage:

https://github.com/jgyates/genmon/issues/35

I will respond to this utility voltage issue in the new new thread.
-Jason

JeanB51, 2017-08-22:
Hi Jason

So today, the transfer switch initiated the first automatic self-test since I installed the monitor for good.
I received 6 emails, with the successive statuses shown below.
Everything worked perfectly.
This is exactly what I installed this for.
Thanks again for the great project.

Switch State: Auto
Engine State: Startup Delay Timer Activated

Switch State: Auto
Engine State: Cranking

Switch State: Auto
Engine State: Running

Switch State: Auto
Engine State: Cooling Down

Switch State: Two Wire Stop
Engine State: Stopped

Switch State: Auto
Engine State: Off - Ready

jgyates, 2017-08-22:
Thanks for the feedback and kind words. Also, thanks for taking the time to help test.

shavep, 2017-09-04:
Great work on this, after some serial cable battles all up and running!

Generac Model 7043
Registers:
Num Regs: 208
Not Changed: 13547 Changed 317 Total Changed 0.02
0038:00f0                       0034:0000
0037:2fc0                       0032:4000
05f5:0000                       05f4:0000
05f1:0000                       01f1:0000
05ed:0064                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:0000
0054:0000                       0208:03f1
005c:0000                       005a:0000
005f:0000                       005d:0000
003c:0000                       000d:0000
000e:110e                       000f:0904
000a:0085                       000c:0000
0005:0e00                       0006:0300
0000:0009                       0001:00000000
0008:0000                       0009:00f6
01f2:0000                       0012:0000
0011:009c                       0010:0111
0007:0000                       0019:0011
002b:0005                       002c:0e00
002a:6a74                       002f:0000
002e:0003                       0020:0000
0021:0000                       001f:49b3
001e:018f                       001d:0000
001c:0000                       001b:46d9
001a:00c7
01f4:33303030373532333030

ebzimny01, 2017-09-21:
I have an older Generac model with an R200 series control board. I'm sure it is less capable than the newer controllers but hoping I can still get some basic going. Know of anyone who has tried? Or have any other pointers to get started?

jgyates, 2017-09-21:
I do not know of anyone who has tried this on an R200 control board. My guess is that if other products like Generac MobileLink support the R200 then it could work. After a quick search it looks like there is an [OmniMetrix Monitor](https://www.omnimetrix.net/wp-content/uploads/2017/03/installation_Generac_R100-R200_Residential.pdf) that would support your generator, but based on the PDF in the link it looks like there is not serial interface so this project will not work for you. There are also some other products on the market that attach to the transfer switch that use WiFi and do not require monthly service but you have to buy the hardware. The [iGenerator](https://igenerator.ca/) monitors the transfer switch, which has it's pros and cons, and has a WiFi only version.

If you have any other questions on the R200 series topic please create a new thread.

jeepjungle, 2017-09-22:
I also have a QT036 Water cooled, propane fueled gen with Nexus controller, installed 1/2013.  My output differs slightly from JeanB51's.
Also, jgyates, I sent you some beer money.  Enjoy.
OK : Auto, Off - Ready
>generator: registers
Registers:
Num Regs: 158 
Not Changed: 12567 Changed 2946 Total Changed 0.23
0038:01f4			0034:01f4
0037:01f4			0032:01f4
05f5:001e			05f4:0032
05f1:01f4			01f1:0d0f
05ed:0018			023b:0000
023e:0000			003b:0000
05fa:01fe			0059:01f4
0058:01f4			0053:01f4
0052:01f4			0057:01f4
0056:01f4			0055:01f4
0054:01f4			0208:03e8
005c:01f4			005a:01f4
005f:01f4			005d:01f4
003c:001e			000d:0000
000e:0e12			000f:0916
000a:0081			000c:0039
0005:0b3b			0006:0301
0000:0006			0001:00000000
0008:0000			0009:00ef
01f2:0911			0012:0000
0011:0090			0010:0511
0007:0000			0019:0000
002b:0000			002c:0000
002a:65d4			002f:01f4
002e:01f4			0020:0000
0021:0000			001f:46de
001e:00f5			001d:44b2
001c:002c			001b:4571
001a:005f			
01f4:30303037363536383033

jgyates, 2017-09-23:
Thanks @jeepjungle! You rock.

A few comments on register differences: The differences are likely things like different exercise time, different time and date, run hours and serial number. Also, some registers are unused in some variants so the register contents are undefined and change randomly.

batisca1, 2017-10-08:
Hi Jason, I really appreciate this project. I implemented it with a Raspberry Pi0 here in Panama City, Panama. There is not international service of mobile link here, so this is really a great option for me.  I have an 11 KW air-cooled, LP, Evolution controller. Generator Model 6439.
OK : Auto, Off - Ready
>generator: registers
Registers:
Num Regs: 208 
Not Changed: 2119 Changed 80 Total Changed 0.04
0038:0001			0034:0000
0037:10f4			0032:4000
05f5:0000			05f4:0000
05f1:0000			01f1:1808
05ed:0064			023b:0000
023e:0000			003b:0000
05fa:0000			0059:0000
0058:0000			0053:0000
0052:0000			0057:0000
0056:0000			0055:0000
0054:0000			0208:03ec
005c:0000			005a:0000
005f:0000			005d:0000
003c:0000			000d:0000
000e:0932			000f:0a08
000a:0087			000c:0001
0005:0a00			0006:0000
0000:0009			0001:00000000
0008:0000			0009:00f7
01f2:0927			0012:0000
0011:009c			0010:0011
0007:0000			0019:000f
002b:0005			002c:0a00
002a:6a72			002f:0000
002e:0000			0020:0000
0021:0000			001f:49b2
001e:018e			001d:0000
001c:0000			001b:46d8
001a:00c6			
01f4:33303030303835373730

jgyates, 2017-10-08:
Thanks for the input and comment. It is great to hear why (and where) people are using the software.

moparks, 2017-10-23:
Model 7043
serial # 3001242961
Registers:
Num Regs: 208 
Not Changed: 1065570 Changed 17473 Total Changed 0.02
0038:0000			0034:0000
0037:0000			0032:4000
05f5:0000			05f4:0000
05f1:0000			01f1:050c
05ed:0064			023b:0000
023e:0000			003b:0000
05fa:0000			0059:0000
0058:0000			0053:0000
0052:0000			0057:0000
0056:0000			0055:0000
0054:0000			0208:03f0
005c:0000			005a:0000
005f:0000			005d:0000
003c:0000			000d:0000
000e:0520			000f:0a17
000a:008d			000c:0016
0005:0c00			0006:0300
0000:0009			0001:00000000
0008:0000			0009:00f6
01f2:060d			0012:0000
0011:009c			0010:0111
0007:0000			0019:0011
002b:0005			002c:0c00
002a:6a75			002f:0000
002e:0003			0020:0000
0021:0000			001f:4904
001e:0179			001d:0000
001c:0000			001b:462a
001a:00b1			
01f4:33303031323432393631

raymondjclark, 2017-11-08:
Great project, thank you. The only thing missing as far as I can tell is the ability to set the exercise frequency (Monthly, Biweekly, Weekly)

I have a model 7033, 11KW Propane model.

Num Regs: 208
Not Changed: 1831930 Changed 30171 Total Changed 0.02
0038:0000                       0034:0000
0037:0f78                       0032:4000
05f5:0000                       05f4:0000
05f1:0000                       01f1:0000
05ed:0064                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:0000
0054:0000                       0208:03ec
005c:0000                       005a:0000
005f:0000                       005d:0000
003c:0000                       000d:0000
000e:071f                       000f:0b08
000a:0094                       000c:0000
0005:0e00                       0006:0000
0000:0009                       0001:00000000
0008:0000                       0009:00ed
01f2:0000                       0012:0000
0011:009c                       0010:0311
0007:0000                       0019:000f
002b:0005                       002c:0e00
002a:6a76                       002f:0000
002e:0001                       0020:0000
0021:0000                       001f:49e0
001e:018f                       001d:0000
001c:0000                       001b:4706
001a:00c7
01f4:33303031373431313137

jgyates, 2017-11-08:
I can add the monthly, weekly, etc setting, however the implementation is different for the four variants of the firmware (air cooled Nexus, air cooled Evolution, liquid cooled Nexus, liquid cooled Evolution). I would need one person with each variant to send me a few register dumps after setting the weekly,biweekly, monthly option via the controller interface, from there I should be able to determine what the settings would be. If there is interest and some volunteers we can start a new thread on this topic. 

There are two types of registers in the controller(s): direct and indexed. If this setting is in the direct registers then it should be easy, if this setting is in the indexed registers then this will be harder to find. I can easily poll the direct registers, however the index registers appear to be write only so finding these would require a more indepth level of debugging.

As a point of reference, my generator (liquid cooled Evolution) does not have a setting for weekly, biweekly, monthly exercise. 

If you want to volunteer to help provide data points, then create a new thread we can go from there.

Thanks
Jason

raymondjclark, 2017-11-08:
I would be happy to. Thanks
Ray

Raymond Clark
Expect tYopS

jgyates, 2017-11-08:
See this thread:
https://github.com/jgyates/genmon/issues/49

jonathanpisarczyk, 2018-04-11:
Great work on this project.  Thank you.  Here's my registers.

Generac 20 kw with Evolution controller G0062500

sudo ./ClientInterface.py
OK : Auto, Off - Ready
>generator: registers

Registers :
    Num Regs : 217
    Not Changed : 283
    Changed : 28
    Total Changed : 0.10

    Base Registers :
        0038 : fffe
        0034 : ffff
        0037 : 0f24
        0036 : 0000
        0033 : ff38
        0032 : 4000
        05f7 : 05d7
        05f6 : 05ce
        05f5 : 0000
        05f4 : 0000
        05f1 : 0000
        01f3 : 1115
        01f1 : 3210
        05ed : 0064
        023b : 0000
        003a : 0000
        023e : 0000
        003b : 0000
        05fa : 0000
        0059 : 0000
        0058 : 0000
        0053 : 0000
        0052 : 0000
        0057 : 0000
        0056 : 0000
        0055 : 0000
        0054 : 0000
        0208 : 03f0
        005c : 0000
        005a : 0000
        005f : 0000
        005d : 0000
        003c : 0000
        000d : 0000
        000e : 0b38
        000f : 040b
        000a : 0088
        000c : 0054
        0005 : 0b00
        0006 : 0300
        0000 : 0009
        0001 : 00000000
        0008 : 0000
        0009 : 00f5
        01f2 : 0a29
        05ee : 0009
        0012 : 0000
        0011 : 009c
        0010 : 0312
        0007 : 0000
        0019 : 0004
        002b : 0005
        002c : 0000
        002a : 696d
        002f : 0000
        002d : 0000
        002e : 0000
        0022 : 45a1
        0020 : 0000
        0021 : 0000
        001f : 457e
        001e : 013b
        001d : 0000
        001c : 0000
        001b : 459c
        001a : 0092
        01f4 : 30303038393339333131

jgyates, 2018-04-11:
Thanks @jonathanpisarczyk , These register dumps help me out with testing and finding new features.

BTW, any register dumps are good, however if folks are on the fence, these are the ones that will help the most (in order):
1) Evolution Liquid Cooled units 
2) Any Nexus units (air cooled or liquid cooled)
3) Evolution Air Cooled (I have the most of these type, but more is always better..)

JustApu, 2018-05-01:
@jgyates Thanks for the software and the continued improvements. This is from a Model QT03624, 36KW Liquid Cooled with Nexus controller.

BTW, are the commands documented somewhere in the Wiki? I couldn't find them. https://github.com/jgyates/genmon/wiki/Appendix-3--Interfacing-Generator-Monitor-to-External-Applications has hyperlinks to "ClientInterface" and "commands" that appear broken in that they just bring me back to the main Wiki page.

Registers : 
    Num Regs : 170
    Not Changed : 77657
    Changed : 17478
    Total Changed : 0.23

    Base Registers : 
        0239 : 0000
        0039 : ffff
        0038 : ffff
        0034 : ffff
        0037 : ffff
        0036 : ffff
        0237 : 0000
        0033 : ffff
        0032 : 0000
        05f7 : ffff
        05f6 : ffff
        05f5 : ffff
        05f4 : ffff
        05f3 : ffff
        05f1 : ffff
        05ed : ffff
        023b : 0000
        003a : 0000
        023e : 0000
        003b : ffff
        05fa : ffff
        0059 : ffff
        0058 : ffff
        0053 : ffff
        0052 : ffff
        0057 : ffff
        0056 : ffff
        0055 : ffff
        0054 : ffff
        0208 : 0400
        005c : ffff
        005a : 0000
        005f : ffff
        005e : ffff
        005d : 0000
        003c : ffff
        000d : 0000
        000e : 0a21
        000f : 0501
        000a : 0080
        000b : 0000
        000c : 0124
        0005 : 0b00
        0006 : 0301
        0007 : 0000
        0000 : 0006
        0001 : 00000000
        0008 : 0000
        0009 : 00f0
        05ee : ffff
        0012 : 0000
        0011 : 0090
        0010 : 0212
        0019 : 0000
        002b : 0000
        002c : 0000
        002a : 65d2
        002f : ffff
        002d : 0000
        002e : ffff
        0022 : 0000
        0020 : 0000
        0021 : 0000
        001f : 4677
        001e : 00a9
        001d : 459d
        001c : 0031
        001b : 450a
        001a : 0013
        01f4 : 30303036363139383634

jgyates, 2018-05-01:
Thanks for the feedback. I have corrected the broken links. The commands listed in the wiki (https://github.com/jgyates/genmon/wiki/1----Software-Overview) are the base commands that are supported via the email interface. The full command list includes commands that are only available via the socket interface (these are used by the web interface). You can see them in the code here:

https://github.com/jgyates/genmon/blob/master/genmon.py#L288

smeisner, 2018-06-07:
I believe you already have my info from the activity over the past few days, but just in case:

Honeywell 15 kw with Nexus controller, Model: G0060331

```
pi@generator:~/genmon $ sudo python ./ClientInterface.py
OK : Auto, Off - Ready
>generator: registers

Registers :
    Num Regs : 170
    Not Changed : 730102
    Changed : 105297
    Total Changed : 0.14

    Base Registers :
        0239 : 0000
        0039 : 0000
        0038 : 0001
        0034 : 0000
        0037 : 12a9
        0036 : 0000
        0237 : 0000
        0033 : 0064
        0032 : 2d57
        05f7 : 03fb
        05f6 : 03fb
        05f5 : 03fb
        05f4 : 03fb
        05f3 : 03fb
        05f1 : 03fb
        05ed : 0000
        023b : 0000
        003a : 0000
        023e : 0000
        003b : 0000
        05fa : 03fb
        0059 : 03fb
        0058 : 03fb
        0053 : 03fb
        0052 : 03fb
        0057 : 03fb
        0056 : 0000
        0055 : 03fb
        0054 : 03fb
        0208 : 03fb
        005c : 0000
        005a : 0000
        005f : 0000
        005e : 0000
        005d : 0000
        003c : 0000
        000d : 0000
        000e : 072c
        000f : 0607
        000a : 0086
        000b : 0000
        000c : 0058
        0005 : 081d
        0006 : 0101
        0007 : 0000
        0000 : 0003
        0001 : 00000000
        0008 : 0000
        0009 : 00e8
        05ee : 0000
        0012 : 0000
        0011 : 009c
        0010 : 0412
        0019 : 0003
        002b : 0000
        002c : 0000
        002a : 668e
        002f : 0000
        002d : 0000
        002e : 0000
        0022 : 45e9
        0020 : 4576
        0021 : 016c
        001f : 00a4
        001e : 00a4
        001d : 4576
        001c : 0072
        001b : 4576
        001a : 00a4
        01f4 : 30303036393032383232

```

ewouk61, 2018-06-08:
Nexus
model: 0058851

pi@raspberrypi:~ $ sudo python /home/pi/genmon/ClientInterface.py
OK : Auto, Running
>generator:registers

Registers :
    Num Regs : 170
    Not Changed : 83697
    Changed : 11293
    Total Changed : 0.13

    Base Registers :
        0239 : 0000
        0039 : 0000
        0038 : 0000
        0034 : 0000
        0037 : 12ed
        0036 : 0000
        0237 : 0000
        0033 : ff9c
        0032 : 2cab
        05f7 : 0400
        05f6 : 0400
        05f5 : 0400
        05f4 : 0400
        05f3 : 0400
        05f1 : 0000
        05ed : 0000
        023b : 0000
        003a : 0000
        023e : 0000
        003b : 034d
        05fa : 0400
        0059 : 0400
        0058 : 0400
        0053 : 0400
        0052 : 0000
        0057 : 0400
        0056 : 0000
        0055 : 0400
        0054 : 0400
        0208 : 0400
        005c : 0000
        005a : 0000
        005f : 0000
        005e : 0000
        005d : 0000
        003c : 0258
        000d : 0000
        000e : 0905
        000f : 0605
        000a : 008b
        000b : 0000
        000c : 0011
        0005 : 0a00
        0006 : 0200
        0007 : 0e12
        0000 : 0003
        0001 : 00030000
        0008 : 003b
        0009 : 0000
        05ee : 0000
        0012 : 00f1
        0011 : 009c
        0010 : 0212
        0019 : 0003
        002b : 0000
        002c : 0000
        002a : 6691
        002f : 0000
        002d : 0000
        002e : 0000
        0022 : 458e
        0020 : 46cc
        0021 : 017e
        001f : 00b6
        001e : 00b6
        001d : 46cc
        001c : 0084
        001b : 46cc
        001a : 00b6
        01f4 : ffffffffffffffffffff

ewouk61, 2018-06-08:
Generac iustrial, h-panel, natural gas gen
model: QT02524KNSNA
25kw
277 leg to ground, 480 leg to leg, 3 phase
12v battery
2 wire start transfer switch

my pi 3 b cpu is at 30% on the status page, but 5% on the others

pi@raspberrypi:~ $ sudo python /home/pi/genmon/ClientInterface.py localhost
OK : Auto : Ready. Stopped.
>generator: registers

Registers :
    Num Regs : 145

    Base Registers :
        0134 : 6f70
        0135 : 7065
        0136 : 642c
        0137 : 204b
        0130 : 0000
        0131 : 0733
        0132 : 0008
        0133 : 5374
        0035 : 3030
        0034 : 4857
        0031 : 2e30
        0030 : 2034
        0033 : 2c20
        00ef : 0733
        004a : 2c20
        004b : 4857
        004c : 3030
        004d : 2c20
        004e : 2444
        004f : 6174
        00f2 : e762
        00f3 : 9ec2
        00f0 : 0000
        00f1 : 014c
        00b1 : 0064
        0141 : 696e
        0140 : 6e6e
        0143 : 2051
        0142 : 672c
        0145 : 6574
        0144 : 7569
        0147 : 7374
        0146 : 5465
        0148 : 2020
        0048 : 2e30
        0049 : 4331
        0040 : 504d
        0041 : 2d44
        0042 : 4350
        0043 : 2052
        0044 : 656c
        0045 : 6561
        0046 : 7365
        0047 : 2034
        013f : 5275
        013a : 5720
        013b : 4175
        013c : 746f
        0059 : 303a
        0058 : 3a30
        0053 : 322f
        0052 : 3031
        0051 : 2032
        0050 : 653a
        0057 : 3231
        0056 : 3320
        0055 : 2f31
        0054 : 3131
        010a : 0219
        010b : 0318
        00af : 0000
        00ad : 0000
        00ab : 0000
        029c : 0000
        029e : 00b9
        0109 : 56c9
        0108 : 4709
        0105 : 41c8
        0104 : 0036
        0107 : 0318
        0106 : 0219
        005b : 2024
        005a : 3036
        0299 : 0000
        00a7 : 0000
        00a5 : 0000
        00a3 : 0000
        00a1 : 0000
        00a9 : 0000
        0110 : 0000
        0066 : 2e30
        00b3 : 0000
        0064 : 332e
        0065 : 3136
        0062 : 2e33
        0063 : 2c20
        0060 : 332e
        0061 : 3138
        00b9 : 0000
        00b7 : 2710
        0002 : 0000
        00b5 : 0000
        0008 : 0000
        0139 : 2053
        002b : 4350
        002c : 2052
        002a : 2d44
        002f : 7365
        002d : 656c
        002e : 6561
        009f : 0000
        009d : 0504
        009b : 0006
        0088 : 0210
        0089 : 0001
        0084 : 0000
        0085 : 0000
        0086 : 0000
        0087 : 0420
        0080 : 841f
        0081 : f8f2
        0082 : 0540
        0083 : 0000
        0022 : 3020
        0023 : 436f
        0020 : 482d
        0021 : 3130
        0026 : 6c6c
        0027 : 6572
        0024 : 6e74
        0025 : 726f
        0028 : 2c20
        0029 : 504d
        0138 : 6579
        00e3 : 1200
        00e1 : 3104
        02a1 : 0000
        02a0 : 0000
        02a2 : 0055
        008d : 0070
        0093 : 0000
        008f : 0001
        00dc : 0000
        00dd : 0000
        008b : 0000
        00e5 : 16d3
        0099 : 0000
        0097 : 00bb
        00e2 : 0606
        0095 : 0000
        00e0 : 0d10
        00e7 : 0618
        00e6 : 0406
        0091 : 02fd
        00e4 : 3140

>

ToltingColtAcres, 2018-08-26:
Generac 20kw 7039 pre-wifi model, purchased July w/ a factory build date in January.

Registers :
    Num Regs : 240
    Not Changed : 607948
    Changed : 13774
    Total Changed : 0.02

    Base Registers :
        0239 : 0000
        0238 : 0000
        0039 : 0000
        0038 : 0000
        0034 : 0000
        0037 : 1ceb
        0036 : 0000
        0237 : 0000
        0033 : 0000
        0032 : 4000
        05f7 : 05fe
        05f6 : 05cb
        05f5 : 0000
        05f4 : 0000
        05f3 : 00c8
        05f2 : 46d4
        05f1 : 0000
        023e : 0000
        023d : 0000
        05ed : 0064
        0235 : 0000
        023b : 0000
        023a : 0000
        003a : 0000
        003c : 0000
        003b : 0000
        05fa : 0000
        0059 : 0000
        0058 : 0000
        0053 : 0000
        0052 : 0000
        0057 : 0000
        0056 : 0000
        0055 : 0000
        0054 : 0000
        0208 : 03f1
        0209 : 0000
        005c : 0000
        005a : 0000
        005f : 0000
        005e : 0000
        005d : 0000
        020f : 0000
        020d : 0000
        020e : 0000
        020b : 0000
        020c : 0000
        020a : 0000
        000d : 0000
        000e : 0706
        000f : 081a
        024a : 0000
        000a : 0089
        000b : 0000
        000c : 0000
        0012 : 0000
        0242 : 0000
        0243 : 0000
        0006 : 0601
        0007 : 0000
        0246 : 0000
        0001 : 00000000
        0244 : 0000
        0245 : 0000
        0248 : 0000
        0249 : 0000
        0008 : 0000
        0009 : 00eb
        0005 : 0a00
        002a : 6a78
        05ee : 008d
        0258 : 0000
        0011 : 009c
        0010 : 0012
        0019 : 0004
        002b : 0005
        002c : 0a00
        0241 : 0000
        002f : 0000
        002d : 0001
        002e : 0006
        0000 : 0009
        0247 : 0000
        0022 : 46d4
        0020 : 0000
        0021 : 0000
        001f : 4b1b
        001e : 018f
        001d : 0000
        001c : 0000
        001b : 4841
        001a : 00c7
        025a : 0000
        01f4 : 33303032363230373030

closed 2018-10-12 by jgyates

trimixdiver1, 2018-11-20:
Hello 

Im an automation engineer, I would like to bring this into my PLC, I understand the serial pins in the connector, but can you tell me what the registers are? Im used to seeing 40100 type registers for MODBUS. Also the machine ID? 

Thanks so much!