jgyates/genmon #11: genmon on Hakaday!

JeremyStevenson, 2017-04-10:
Some well deserved recognition to jgyates for his work.

http://hackaday.com/2017/04/09/generator-monitor-gives-the-phone-company-the-boot/

madisdev, 2017-04-10:
Slow Clap.....good job. Kudos for all the hard work and free tech support :)

jgyates, 2017-04-10:
Aw shucks :). madisdev was the one who suggested I submit this to them.  Good call Madis. Thanks everyone for helping get this off the ground. All of your contributions, testing, feedback, and questions have made this project much better.

madisdev, 2017-04-10:
I mean you should be proud. Considering the commercial options for generator monitoring, this is friggen awesome. 200+ for the wireless monitor with 3 colored lights on it? No thanks.

rhombixt, 2017-04-10:
I agree - this is HUGE plus for the Canadian market because the wireless monitor service is not available here. It also allows for 3rd party integration - something the Kohler does with their PIM but generac does not.

ts4s, 2017-04-11:
Congrats on the project! I saw talks of something like this a few years ago on a forum that no longer exist.  Glad to see somebody pickup the ideal and ran with it. Now those of with one of the generator can mod them. so dislike the monitor unit i have, always having to reset it. Looking around, I am wondering where to? i have most o f the hardware, just need the instructions to get the pi setup while the rest of the hardware arrives.

jgyates, 2017-04-11:
Hi,

The wiki is a good place to start. It has a section on setting up your Pi.

Hardware info (Pi, cable , etc)
https://github.com/jgyates/genmon/wiki/Hardware

Setup of the Pi and software:
https://github.com/jgyates/genmon/wiki/Setting-up-your-Pi-and-genmon

Description of the software (requirements and settings)
https://github.com/jgyates/genmon/wiki/Software

Feel free to post a question if you have a specific question.
Thanks
Jason

LynnSchott, 2017-09-17:
I also read the comments from the persons that do not understand this monitoring application. To me it does way more than the standard Cell monitoring system does and (if you do not mind) the portable idiot monitor. To me I would like the up front information on what I am calling trouble in for. 

If I call in for a problem, and tell the Generac technicians these words "It is not working". is helpful. They may not have the correct parts to repair the system on the above comments. besides how do you fix "It is not working" :) I am a technician by heart and would like more detailed information on what the problem is and what you created here is fantastic and is also beyond the persons that made comments on that review. 

I also do not think paying $99.00 a year on cell service is doing justice for monitoring where your application is a great assistance.

jgyates, 2017-09-17:
Thanks @LynnSchott, for me, the cell monitoring solution was not working well (notification delays, lack of detailed information being the primary two issues).

bowood, 2018-03-04:
Anyone added a SmatThings integration?  I've made a few for other devices.  I'll make one for this and share the source code next.

The most amazing part of this software is the ability to shut off the generator remotely.  My generator runs on propane and I tend to be away from the house a week at a time.  If I know there is no pipe freezing risk the ability to shut off the generator for prolonged outages saves me real money.  Fantastic job with this.

jgyates, 2018-03-04:
@bowood , Sounds great. I will add this to the repository once you post it.

closed 2018-10-12 by jgyates