jgyates/genmon #12: "Unknown" Log Entries - (Request for input"

jgyates, 2017-04-11:
The purpose of this thread is to allow people using the software to help resolve log entries that display the words "Unknown". For example:

     02/01/12 08:12:57 Unknown 0x01 

To resolve this issue, please post the following information to this thread:

1) The line in the genmon.py log output (example above)
2) A picture or verbatim text of the corresponding log entry from the generator controller  (match the time and date from the unknown entry.
3) If you are using an Evolution or Nexus controller
4) If you generator is liquid cooled (it has a radiator) or air cooled (no radiator)

The reason this information is needed to resolve the problem you are seeing is that the log entries, when read over the modbus serial interface only contain a hex value that represents a displayed string. Since we only have access to the hex value and I don't know the mapping of the hex code to the strings, this information allows the string to hex value mapping to be completed if enough people contribute their unknow log entry data.

Once your information is posted it will be added to genmon.py.

Thanks
Jason

derekatkins, 2017-06-15:
I have the following entries in my Service Log:
    12/15/16 13:13:49 Unknown 0x3E 
    12/14/16 13:05:49 Unknown 0x18 
(and a bunch more 0x18 codes)
This is an Evolution controller with an air cooled generator (Generac model 0065510).   I can't seem to attach my images so:

0x3E is "Battery Maintained"
0x18 is "Inspect Battery"

jgyates, 2017-06-15:
I checked in an updated genmon.py that added these codes. Thanks!

derekatkins, 2017-06-15:
Huh.  I pulled and restarted but it still says:
    Service Log:    12/15/16 13:13:49 Unknown 0x3E 
Does genmon cache the "Unknown" state?

derekatkins, 2017-06-15:
Ah, you put it into the wrong section.  This should go under the ServiceLogDecoder and not the AlarmDecoder.

derekatkins, 2017-06-15:
Try this patch..

[service-log.diff.txt](https://github.com/jgyates/genmon/files/1078769/service-log.diff.txt)

jgyates, 2017-06-15:
Sorry, I missed the fact that these came from the service log. Update is checked in.

USAFPride, 2017-11-17:
01/01/09 00:00:30 Unknown 0x00 - Nexus controller.  
Probably a nothing code that just needs to be ignored.

jgyates, 2017-11-17:
@USAFPride, I can fix this in the code however I need the following information:

1) The exact text from the corresponding log in your generator controller (match the time and date 01/01/09 00:00:30)
2) If you generator is air cooled or liquid cooled. 

You already said you have a Nexus controller and provide the unknow code of 0x00 so with the above info I can add this to the software and remove the unknown condition in the software.

The intent of this thread is that the software can read the controller logs, but when it reads the log we only get a code, not a text description of the log entry. The text output can be read from the controller user interface screen by matching the date and time in the log entry.

This is likely a code that was generator when your installed, hence the 01/01/09 date, i.e. an error occurred before the time and date was set on the controller. Probably one you can ignore, but it may help someone else.

Thanks
Jason

USAFPride, 2017-11-17:
High Engine Temp.
Air Cooled

jgyates, 2017-11-19:
@USAFPride, I just checked in a version that includes this fix. Please update the web app static files if you are using that as well. If you just update your entire repository and keep you /etc/genmon.conf file you should be set.

derekatkins, 2017-11-21:
@jgyates -- I just had maintenance on my generator and now the service log has code 0x3F which is not understood by genmon.
I'll try to find the time to get to the system and see what it says in the log.

jgyates, 2017-11-21:
Sounds good. If you have an Evolution controller my guess is that it is likely "Maintenance Reset" or possibly "Schedule C Serviced". These are the only two documented Evolution service codes I don't have yet.

jgyates, 2017-11-22:
@derekatkins, Let me know once you can validate your alarm log in your controller and I will add this to the code. All you need to do is matchs the time and date of the 0x3F log entry in your genmon service log matches the description of the same time and date entry in your controller service log. Let me know if you have any questions. Thanks!

derekatkins, 2018-01-22:
@jgyates -- Sorry for the extremely late reply.  I finally had a few seconds to get to the generator and look at the service log.  The log message (0x3F) was indeed Maintenance Reset.

jeepjungle, 2018-01-31:
Tech swapped out the Nexus Controller on my 36KW Generac Liquid cooled unit:
01/31/18 09:36:28 Unknown 0x0C
Display shows Canbus Error

jgyates, 2018-01-31:
@jeepjungle, I have uploaded a new version of genmon.py that addresses this. Thanks for the input.

jeepjungle, 2018-01-31:
[image]

Rock on

hokie21, 2018-05-07:
02/02/13 10:44:21 Unknown 0x0A 
        02/02/13 09:59:53 Unknown 0x0A 
The display for the alarm log shows "Under Voltage"
I have a Nexus 5870

jgyates, 2018-05-07:
Thanks @hokie21 

I checked in an update that should correct this based on your feedback.

closed 2018-12-05 by jgyates