jgyates/genmon #13: Unknown Alarm Conditions (Request for input)

jgyates, 2017-04-12:
If an alarm condition occurs on your generator and genmon reports an Unknown alarm. Please post to this thread the following information so we can add this alarm condition to genmon:

1) The output of the registers command (generator: registers)
2) The error code (for Evolution Controllers)
3) The a picture or verbatim text displayed on the controller of the alarm
4) Your controller / generator info (model, Nexus or Evolution, liquid or air cooled)

Thanks
Jason

sgtmaj27, 2017-05-27:
>If an alarm condition occurs on your generator and genmon reports an Unknown alarm. Please post to >this thread the following information so we can add this alarm condition to genmon:

    >1. The output of the registers command (generator: registers)
    >2. The error code (for Evolution Controllers)
    >3. The a picture or verbatim text displayed on the controller of the alarm
    >4. Your controller / generator info (model, Nexus or Evolution, liquid or air cooled)

Registers:
Num Regs: 158
Not Changed: 46491 Changed 9774 Total Changed 0.21
0038:fffe                       0034:ff3e
0037:12b6                       0032:4000
05f5:03fb                       05f4:03fb
05f1:03fb                       01f1:0c0a
05ed:0c0a                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:03fb
0052:03fb                       0057:0000
0056:0000                       0055:0c0a
0054:0000                       0208:03fb
005c:03fb                       005a:03fb
005f:03fb                       005d:03fb
003c:0000                       000d:0000
000e:1004                       000f:051b
000a:008c                       000c:005a
0005:0a00                       0006:0600
0000:0003                       0001:00000000
0008:0000                       0009:00e4
01f2:0c0a                       0012:0000
0011:009c                       0010:0611
0007:0000                       0019:0000
002b:0000                       002c:0000
002a:646b                       002f:0000
002e:0000                       0020:43e9
0021:0179                       001f:00b1
001e:00b1                       001d:43e9
001c:007f                       001b:43e9
001a:00b1
01f4:30303036303430303632

Alarm Log:                                  **Actual from Nexus Controller**
    04/22/17 10:00:04 Unknown 0x03                **Overspeed**
    05/26/16 10:53:08 Low Battery                      
    08/08/15 08:01:27 Low Battery
    05/11/15 13:01:11 Unknown 0x02                **Overcrank**
    03/02/15 13:00:04 Unknown 0x03                **Overspeed**
    12/15/14 14:01:11 Unknown 0x02                **Overcrank**
    01/08/14 13:15:54 Unknown 0x04                **RPM Sense Loss**
    01/08/14 12:03:18 Unknown 0x04                **RPM Sense Loss**
    01/06/14 10:01:11 Unknown 0x02                **Overcrank**
    12/30/13 10:01:11 Unknown 0x02                **Overcrank**
    01/01/09 00:00:01 Check Battery               **Inspect Battery**

NOTE: The output appears to have unknown values. Please see the following threads to resolve these issues:
        https://github.com/jgyates/genmon/issues/12
        https://github.com/jgyates/genmon/issues/13

sgtmaj27, 2017-05-27:
I forgot the essentials:

Generac 5870, Air cooled, Nexus Controller

jgyates, 2017-05-27:
Thanks! I added these to genmon.py and tagged a new release.

macbeck22, 2017-06-20:
Hello,

After installing and getting online, I have a few unknown errors as well.  Please see below.

Tks

Evolution Controller

Generac Model #6552
Firmware : V1.17, Hardware : V1.06

Time: 10:20:31 PM
Date: Mon 19-Jun-2017

Registers:
Num Regs: 208
Not Changed: 71691 Changed 1136 Total Changed 0.02
0038:fffe                       0034:ffff
0037:114d                       0032:4000
05f5:0000                       05f4:0000
05f1:0000                       01f1:3a0c
05ed:0064                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:0000
0054:0000                       0208:0402
005c:0000                       005a:0000
005f:0000                       005d:0000
003c:0000                       000d:0000
000e:1612                       000f:0613
000a:0088                       000c:000a
0005:0c3a                       0006:0400
0000:0009                       0001:00000000
0008:0000                       0009:00f3
01f2:0204                       0012:0000
0011:009c                       0010:0111
0007:0000                       0019:0011
002b:0005                       002c:0c3a
002a:6a75                       002f:0000
002e:0004                       0020:0000
0021:0000                       001f:485e
001e:0185                       001d:0000
001c:0000                       001b:4584
001a:00bd
01f4:33303030373230363239

Alarm Log:

03/04/17 21:57:45 Unknown 0x20 : Alarm Code: 0000    **Charger Warning
03/04/17 21:50:50 Unknown 0x14 : Alarm Code: 0000    **Low Battery
03/04/17 21:49:55 Unknown 0x20 : Alarm Code: 0000    **Charger Warning
03/03/17 19:32:59 Unknown 0x14 : Alarm Code: 0000    **Low Battery
09/23/16 14:30:56 Unknown 0x20 : Alarm Code: 0000    **Charger Warning
09/23/16 14:26:00 Unknown 0x20 : Alarm Code: 0000    **Charger Warning

**Error as displayed on controller.

jgyates, 2017-06-20:
I have updated genmon.py with the codes for low battery and charger. Let me know if this does not correct your problem.

Thanks
Jason

macbeck22, 2017-06-20:
I'm still seeing the unknowns after a pull.

jgyates, 2017-06-20:
Did you restart genmon?

macbeck22, 2017-06-20:
Yep, and tried a reboot.

jgyates, 2017-06-20:
ok. try it now. I just checked in a new version. I had put the two new warning codes in your log in the table for error codes instead of the warning table for air cooled models. Let me know how this works. Sorry for the mix up on my end. My generator is a liquid cooled model so I can not test the exact circumstance you have on your generator.

macbeck22, 2017-06-20:
That seems to have fine it.  No more unknowns on my screen.  If you need
anything with an air cooled model, feel free to email me.  Thanks!

zekyl314, 2017-07-13:
I don't have an alarm code on the panel right now, but had this from past history apparently.

>generator: registers
Registers:
Num Regs: 208
Not Changed: 1400 Changed 117 Total Changed 0.08
0038:ffff                       0034:0000
0037:133a                       0032:4000
05f5:0000                       05f4:0000
05f1:0000                       01f1:0c0e
05ed:0064                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:0000
0054:0000                       0208:03fe
005c:0000                       005a:0000
005f:0000                       005d:0000
003c:0000                       000d:0000
000e:132e                       000f:070d
000a:0086                       000c:0024
0005:0e00                       0006:0001
0000:0009                       0001:00000000
0008:0000                       0009:00f4
01f2:020a                       0012:0000
0011:009c                       0010:0411
0007:0000                       0019:0003
002b:000a                       002c:0000
002a:6967                       002f:0000
002e:0000                       0020:0000
0021:0000                       001f:44ca
001e:016b                       001d:0000
001c:0000                       001b:467c
001a:00c6
01f4:30303037393532363337

Honeywell 62600
It is an evolution air cooled unit.

JeanB51, 2017-08-07:
Here are my unknow alarm codes - I wrote the actual Nexus alarms from the Nexus controller next to them

As I wrote in the "registers" issue , this is the model info:
Generac Qt036, 36kw, Liquid-cooled, Nexus controller, Propane fuel.

Alarm Log:
    07/31/17 10:35:24 Unknown 0x01 --> Low oil pressure
    07/31/17 10:33:35 Unknown 0x0B --> Low cooling fluid
    06/10/17 18:20:15 Unknown 0x17 --> Inspect Air Filter
    09/19/15 19:04:57 Unknown 0x21 --> Service Schedule A

jgyates, 2017-08-07:
This is good info, one question:

Since you have a liquid cooled nexus I want to make sure that the nexus Liquid and Air cooled models share the same codes for the given alarm (i.e. 0x01 = Low oil pressure on both LC and AC Nexus models). Do you have any other entries in your log that the are correctly displayed in the web interface or with the "logs" command or are these the only entries you have in your alarm log? I want to match the against the existing ones I have taken from users with a Nexus Air Cooled model.

JeanB51, 2017-08-07:
I have two alarm log entries that show "RPM Sense loss". Otherwise a bunch of 0x17 and 0x21

all the other logs seem good : 
Start Stop Log:
    07/31/17 10:39:42 Stopped - Auto
    07/31/17 10:35:49 Running - Manual
    07/31/17 10:35:04 Running - Manual
    07/31/17 10:33:57 Running - Manual
    07/31/17 10:31:41 Running - Manual
    07/31/17 09:39:12 Stopped - Auto
    07/10/17 17:19:39 Stopped - Auto
    06/20/17 11:07:53 Stopped - Auto
    06/20/17 11:07:28 Running - Manual
    06/20/17 11:04:27 Stopped - Auto
    06/20/17 11:04:08 Running - Manual
    06/20/17 10:58:07 Running - Manual
    06/20/17 10:54:49 Running - Manual
    06/20/17 10:22:24 Stopped - Auto
    06/20/17 10:10:14 Running - 2 Wire Start
    06/20/17 10:07:13 Stopped - Auto
    06/20/17 09:51:37 Running - 2 Wire Start
    06/20/17 09:42:41 Stopped - Auto
    06/20/17 09:40:20 Running - 2 Wire Start
    06/20/17 09:32:44 Stopped - Auto
    06/20/17 09:31:06 Running - 2 Wire Start
    06/20/17 09:27:48 Stopped - Auto
    06/20/17 09:12:44 Stopped - Auto
    06/11/17 18:15:06 Running - Manual
    06/10/17 18:20:15 Stopped - Auto
    06/10/17 18:03:53 Running - 2 Wire Start
    06/03/17 18:19:59 Stopped - Auto
    06/03/17 18:03:37 Running - 2 Wire Start
    05/27/17 18:19:40 Stopped - Auto
    05/27/17 18:03:18 Running - 2 Wire Start
    05/20/17 18:19:22 Stopped - Auto
    05/20/17 18:03:00 Running - 2 Wire Start
    05/19/17 07:01:05 Stopped - Auto
    05/18/17 23:04:33 Running - 2 Wire Start
    05/13/17 17:19:04 Stopped - Auto
    05/13/17 17:02:42 Running - 2 Wire Start
    05/06/17 17:18:46 Stopped - Auto
    05/06/17 17:02:24 Running - 2 Wire Start
    04/29/17 16:18:26 Stopped - Auto
    04/29/17 16:02:05 Running - 2 Wire Start
    04/22/17 16:18:07 Stopped - Auto
    04/22/17 16:01:46 Running - 2 Wire Start
    04/15/17 16:17:49 Stopped - Auto
    04/15/17 16:01:27 Running - 2 Wire Start
    04/08/17 16:17:31 Stopped - Auto
    04/08/17 16:01:09 Running - 2 Wire Start
    04/01/17 16:17:09 Stopped - Auto
    04/01/17 16:00:48 Running - 2 Wire Start
    03/25/17 16:16:50 Stopped - Auto
    03/25/17 16:00:29 Running - 2 Wire Start

jgyates, 2017-08-07:
ok. that helps. I have a fix for this, once I get some feedback on https://github.com/jgyates/genmon/issues/35 I can post an updated genmon.py.

JeanB51, 2017-08-14:
Here is another unknow alarm.
Every time the generator stops after a run, I get 3 emails:
1: generator cooling down
2: generator alert : CRITICAL
3: generator off

For number 2, the email body is : 
Time: 12:40:20 PM
Date: Mon 14-Aug-2017

Notice from Generator: 

Switch State: System in Alarm
Engine State: Stopped

Current Alarm: UNKNOWN ALARM: 00090017
System In Alarm: 0001:00090017

Last Log Entries:
    Start Stop Log: 08/14/17 12:24:10 Running - 2 Wire Start 
    Alarm Log:      07/31/17 10:35:24 Low Oil Pressure 

To clear the Alarm/Warning message, press OFF on the control panel keypad followed by the ENTER key.

But if I do the "logs" command, this unknown alarm does not show up. 
And if I ask for a system status, everything seems normal.
Could this be a spurious signal ?

JB

jgyates, 2017-08-14:
Yes, the cause here is the software has never seen register 0001 with a value of 0017 in the lower word. The 0009 in high word means the engine is stopped. I will code this as "stopping". My next update will reflect this. I have a couple of other updates to add for your Nexus Liquid cooled model as well. It appears that the stop sequence is slightly different for the Nexus LC.

jgyates, 2017-08-14:
One question: Are you using 2 wire start, or is your generator detecting the utility voltage and automatically starting? You start stop log has an entry for 2 wire start so I want to validate that this is actually the case because this may help me interpret the value of 17 better. Also, if you are using 2 wire start, what is the state of your switch (auto, off, manual) on your controller?

jeepjungle, 2017-09-06:
Registers:
Num Regs: 158
Not Changed: 59622 Changed 13883 Total Changed 0.23
0038:01f4                       0034:01f4
0037:01f4                       0032:01f4
05f5:001e                       05f4:0032
05f1:01f4                       01f1:0c0c
05ed:0018                       023b:0000
023e:0000                       003b:0000
05fa:01fe                       0059:01f4
0058:01f4                       0053:01f4
0052:01f4                       0057:01f4
0056:01f4                       0055:01f4
0054:01f4                       0208:03e8
005c:01f4                       005a:01f4
005f:01f4                       005d:01f4
003c:001e                       000d:0000
000e:131b                       000f:0905
000a:007e                       000c:0035
0005:0c00                       0006:0300
0000:0006                       0001:00000000
0008:0000                       0009:00f1
01f2:050c                       0012:0000
0011:0090                       0010:0211
0007:0000                       0019:0000
002b:0000                       002c:0000
002a:65d4                       002f:01f4
002e:01f4                       0020:0000
0021:0000                       001f:46de
001e:00f9                       001d:44b2
001c:0031                       001b:4571
001a:0063
01f4:30303037363536383033

Alarm log:
 09/03/17 14:20:28 Unknown 0x0F    ---> GOVERNOR FAULT
 09/03/17 14:16:16 Unknown 0x1E     --->  LOW FUEL PRESSURE
 09/03/17 13:31:34 Unknown 0x22   ---->  SERVICE SCHEDULE B

Generac QT036 - Nexus/Liquid cooled/NG fueled

jgyates, 2017-09-06:
Thanks @jeepjungle. I uploaded a new version of genmon.py with these three alarm codes. Let me know if you have any issues.

jeepjungle, 2017-09-06:
Works great!  Thanks.
Rob

proppilot, 2017-11-18:
Hi jgyates, I am new to the board and Pi 3 programming (but an old school programmer) and love the project now that the Generac cell based link went down in Canada.  I have had a Guardian 11kW air cooled for six weeks and jumped on the project.  Just got it live on the generator today and have an alarm code error when running the ClientInterface

[image]
[image]
[image]

proppilot, 2017-11-18:
[image]

proppilot, 2017-11-18:
> <reply+02039b7aaa41406f2aa95378809752f5e286b6cea66c72da92cf0000000116285a529
> 2a169ce0d2e8a64@reply.github.com>
> <comment@noreply.github.com>
> (#13)
>
> Hi @propilot,
>
> I can fix this, I just need the following:
> 1. Rin this command:
> 2. Generator: logs
> 3. find any unknowns in the list of alarms and find the corresponding
> entries in you alarm log from your controller LCD screen menu. You should be
> able to match the time and date in the log out put from the above command to
> the log entry from your display. I will need the  any unknown codes ( i.e.
> 0x21) and the matching text description from your controllers alarm log
> display.
> Let me know if you have any questions.

Hi Jason,

Logs are attached.  The hex code 0x21 is 'Charger Missing AC'

What is strange is the date as it is 01/01/2013 so that must be the default
date when Generac first powers up the controller as the generator was
installed six weeks ago.

Love your project.  Look for the Paypal donation !

Regards,  Tony

TI7SrlBks5s3z5SgaJpZM4M69j5> .

  
b","name":"GitHub"},"entity":{"external_key":"github/jgyates/genmon","title"
:"jgyates/genmon","subtitle":"GitHub
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
-9aed-b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/jgyates/genmon"}},"updates":{"snippets":[{
"icon":"PERSON","message":"@jgyates in #13: Hi @propilot,\r\n\r\nI can fix
this, I just need the following:\r\n\r\n1) Rin this command:\r\n
Generator: logs\r\n2) find any unknowns in the list of alarms and find the
corresponding entries in you alarm log from your controller LCD screen menu.
You should be able to match the time and date in the log out put from the
above command to the log entry from your display. I will need the  any
unknown codes ( i.e. 0x21) and the matching text description from your
controllers alarm log display.\r\n\r\nLet me know if you have any
questions.\r\n\r\nThanks\r\nJason"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/13#issuecomment-34546
9129"}}}

jgyates, 2017-11-18:
Thanks! I will post an update on Monday. I am out of town today.

proppilot, 2017-11-18:
No problem at all.  Sent you $ 50 dollars for your efforts.

Two questions for whenever

1) on monitor page it says Evolution Controller, Liquid Cooled and it is air cooled. Is this a programming issue by my installer or not relevant ?

2) genserv.py is taking between 98 and 100% CPU whether or not the status web page is called up.  Expected ?

Processor does not seem sluggish and I have the monitor web page open and for SSH sessions open via wifi from my Mac to the generator.

Thanks for the help Jason !

Regards,

Tony

jgyates, 2017-11-19:
Hi Tony,

Regarding question 1, the software should auto detect your controller, unless you have the overrides enabled in /etc/genmon.conf. Based on your registers posted above register 0000 = 0009, which should be evolution air cooled. Make sure your /etc/genmon.conf file has the following lines commented (add a # at the beginning of the line).

     #liquidcooled = True
     #evolutioncontroller = True

If either of these lines does not have the leading # then it will override the corresponding setting.

Regarding question 2, my cpu usage when is the following:

     load average: 0.06, 0.07, 0.01

100% is not normal. I would attempt to resolve the 1st issue and see if the CPU do not return to normal.

Let me know if that does not fix your problem.

I will post and update with the updated alarm later today. I will let you know in this thread once it is there.

Tansk
Jason

proppilot, 2017-11-19:
> <reply+02039b7a237c6974bf347b06429573f714d54b77f9f72f9592cf000000011629aa689

Hi Jason,

I see there are two places to comment out the Liquid Cooled option and I had
one commented out.  Other one done.

As for the CPU issue, a reboot resolved that and the unit has been operating
in the single digit CPU usage per running task.

Regards, Tony

Date:  Sunday, November 19, 2017 at 3:24 PM

Vg1sW3Dks5s4I5ogaJpZM4M69j5> .

  
"icon":"PERSON","message":"@jgyates in #13: Hi Tony,\r\n\r\nRegarding
beginning of the line).\r\n\r\n     #liquidcooled = True\r\n
#evolutioncontroller = True\r\n\r\nIf either of these lines does not have
setting.\r\n\r\nRegarding question 2, my cpu usage when is the
following:\r\n\r\n     load average: 0.06, 0.07, 0.01\r\n\r\n100% is not
return to normal.\r\n\r\nLet me know if that does not fix your
problem.\r\n\r\nI will post and update with the updated alarm later today. I
there.\r\n\r\nTansk\r\nJason"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/13#issuecomment-34554
6578"}}}

jgyates, 2017-11-19:
Ouch, my bad. I had two in the default in the repository. I will removed the one that is uncommented as it is not needed.

jgyates, 2017-11-19:
@proppilot, I just checked in a version that includes this fix. Please update the web app static files if you are using that as well. If you just update your entire repository and keep you /etc/genmon.conf file you should be set.

USAFPride, 2017-11-22:
From Email:

> Generator Alert at x:
> CRITICAL
> Time
> Date
> Switch State: System In Alarm
> Engine State: Stopped
> Current Alarm: UNKNOWN ALARM: 0009001
> System In Alarm: 001:0009001
> 
> Last Log Entries:
> Start Stop Log: date time Stopped - Auto
> Alarm Log: date time Low Battery

Nexus Air Cooled

`>generator: registers
Registers:
Num Regs: 159
Not Changed: 45219 Changed 13151 Total Changed 0.29
0038:0000                       0034:0000
0037:0000                       0032:0000
05f5:03fb                       05f4:0000
05f1:0000                       01f1:000a
05ed:000a                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:000a
0054:0000                       0208:03fb
005c:03fb                       005a:03fb
005f:03fb                       005d:03fb
003c:0000                       000d:0000
000e:1315                       000f:0b15
000a:008c                       000c:00b7
0005:0a00                       0006:0000
0000:0003                       0001:00000000
0008:0000                       0009:00f6
01f2:0805                       0012:0000
0011:009c                       0010:0211
0007:0000                       0019:0002
002b:0000                       002c:0000
002a:668d                       002f:0000
002d:0000                       002e:0000
0020:4724                       0021:0139
001f:00c7                       001e:0071
001d:4724                       001c:0095
001b:4724                       001a:00c7
01f4:30303036343733323239
`

jgyates, 2017-11-22:
Hmm, couple of questions: The email states 

     Current Alarm: UNKNOWN ALARM: 0009001
     System In Alarm: 001:0009001

The numbers in these lines are the contents of the register 0001, however your registers dump shows register 0001=00000000. The email only has 7 digits, instead of 8. Is it 00090001 or 00009001? 

Also, does your generator show an alarm condition? If yes, then what is that condition? If not then we should look at your genmon.conf file. 

Another thing that it could be is a comm issue. Look at you the monitor command and it should tell you if there are any communication errors. While you are looking at the "monitor" command (or Monitor page in the web UI) verify that the "Generator Type Selected" is Nexus Air Cooled. If it is not then I would suspect the genmon.conf file. I had an extra line that was enabled that I updated a few days ago (liquidcooled = True). That line should be commented out with a # .

USAFPride, 2017-11-22:
`00090001` <-dropped a zero
Generator is not in alarm
Nexus Air Cooled is selected
No Communication issues
`Serial Stats: 
    PacketCount M: 22897, S: 22896, Buffer Count: 0
    CRC Errors : 0  Percent : 0.00
    Discarded  : 0  Restarts: 0  TimeOut: 0
    Packets per second: 60.80
    Average Transaction Time: 0.0316 sec`

jgyates, 2017-11-22:
OK. The high part (0009) means the generator is stopped, the low part (0001) is new. I have never seen that before. Is there an alarm log entry that matches the time and date? The low portion (0001) of this register is normally the state of the switch, unless an alarm occurs (then it is an alarm code). Once the alarm is cleared then the low part of the register goes back to reporting the switch state. The alarm log is updated immediately when an alarm occurs. The Evolution controller added a 4 digit alarm code (which are documented by Generac), but the Nexus as only the low part of this register and the alarm log.  If we can match the timing of this email to an alarm log I can add that condition to the code and the email will be more informative.

USAFPride, 2017-11-22:
I'll watch for it and see if I can get it to show something. I have an idea it might be something related to a manual start, then stop ("generator stopping"), and then going to auto.

jgyates, 2017-11-22:
Forgive me if I am explaining something you already know, just want to make sure you understand what I am asking. You can navigate the menu on the front panel of the controller and look at the alarm log, or you can just look at the alarm log from the UI. There is a chance that genmon already decodes the alarm log for this entry, just not the contents of the low part of the register, so just try to match the time/date on the email to a time / date of an entry in the  alarm log.  or just post the time and date of the email with your full alarm log.

USAFPride, 2017-11-22:
I understand.  Nothing matches on the UI.  The email has a time stamp of 04:56:17 PM and the closest alarm on the UI has `16:57:01 Low Battery` (and matches on the controller). The email is 44 seconds before the alarm, so unless your software is quicker than the controller, it is not currently logged. :)

jgyates, 2017-11-22:
That could be it. Do you recall if this alarm stopped the generator or if you had to reset the error to start it again. The issue with the time is the software can set the time, but presently I know of no way to set the seconds (only hour and minute), so the time of the Pi and the generator can be off by 59 seconds. The email is timestamped by the Pi (or your mail server) and the log entries are timestamped by the generator.

For what it is worth, when genmon sets the time, it waits for the Pi's time seconds to be zero so the offset between the controller and the Pi should always be near constant. You can view the offset in real time from the mail web UI page:

     Monitor Time:   Tuesday November 21, 2017 20:44:54
     Generator Time: Tuesday November 21, 2017 20:45

Let me know what you want to do. I can add this code now or if you are having a reoccurring battery issue we can wait until the next alarm.

USAFPride, 2017-11-22:
I hope I don't have a battery issue for awhile as I just swapped it out.  My bet is that it is the battery alarm as I've gotten it for a bit.

I'm not sure you saw this above:

I also have an email with
`No Alarms: 0001:00000000`

jgyates, 2017-11-22:
The "No Alarms" email is just an email letting you know that all is clear. One of the core pieces of logic is to monitor register 0001, if anything changes, you get an email as this is the base status register that monitors the switch state, engine state and alarm state.

I added the Low Battery code and check in the latest. Thanks for you help!!

lqallison, 2018-04-18:
Found Evolution Error codes on Zillerelectric.com. This is a copy & Paste into Microsoft Word file.
[Evolution E-Codes.docx](https://github.com/jgyates/genmon/files/1922384/Evolution.E-Codes.docx)

jgyates, 2018-04-18:
@lqallison ,

I appreciate the information, however the file ALARMS.txt already contains all of these plus others. The issue with "Unknown" Alarms is not related to knowing the the documented E-Codes since we have all of those, but rather the information that is stored internally by the generator is coded to a status register. The mapping between these status register bits and the actual E-Code or the text description displayed on the screen is unknown for some bit positions. 

The way the software works is when running on an evolution controller, we take the E-Code and look up the value in ALARMS.TXT to get the text description of the alarm. If it is not in ALARMS.txt then we try to decipher the status bits. The info we need is in ALARMS.txt in most cases. For the Nexus it is different situation because that controller does not have the concept of E-Codes so we rely solely on the people to tells us what their errors are and the bits associated with them (as displayed in the email sent from genmon when an alarm occurs).

berenguerj, 2018-04-27:
Hi @jgyates 
Let me know if you need additional information:

**From register**s:

Time: 06:11:20 PM
Date: Fri 27-Apr-2018
Registers : 
    Num Regs : 220
    Not Changed : 2504129
    Changed : 55019
    Total Changed : 0.02
    Base Registers : 
        0239 : 0000
        0039 : 0000
        0038 : fffe
        0034 : ffff
        0037 : 1037
        0036 : 0000
        0237 : 0000
        0033 : ff38
        0032 : 4000
        05f7 : 05a4
        05f6 : 05b6
        05f5 : 0000
        05f4 : 0000
        05f3 : 00c8
        05f1 : 0000
        05ed : 0064
        023b : 0000
        003a : 0000
        023e : 0000
        003b : 0000
        05fa : 0000
        0059 : 0000
        0058 : 0000
        0053 : 0000
        0052 : 0000
        0057 : 0000
        0056 : 0000
        0055 : 0000
        0054 : 0000
        0208 : 03fb
        005c : 0000
        005a : 0000
        005f : 0000
        005e : 0000
        005d : 0000
        003c : 0000
        000d : 0000
        000e : 120c
        000f : 041b
        000a : 0081
        000b : 0000
        000c : 001e
        0005 : 0c00
        0006 : 0600
        0007 : 0000
        0000 : 0009
        0001 : 00000000
        0008 : 0000
        0009 : 00f3
        05ee : 01c2
        0012 : 0000
        0011 : 009c
        0010 : 0512
        0019 : 0001
        002b : 0005
        002c : 0c00
        002a : 6a77
        002f : 0000
        002d : 0001
        002e : 0006
        0022 : 45c4
        0020 : 0000
        0021 : 0000
        001f : 4a0b
        001e : 0171
        001d : 0000
        001c : 0000
        001b : 4731
        001a : 00a9
        01f4 : 33303031393933343938

**From Email:**

Notice from Generator: 
Switch State: System in Alarm
Engine State: Stopped
Current Alarm: UNKNOWN ALARM: 0009002b
System In Alarm: 0001:0009002b
Last Log Entries:
Logs : 
    Alarm Log : 04/27/18 08:51:27 Charger Missing AC : Alarm Code: 0000
    Service Log : 
    Start Stop Log : 04/27/18 06:56:59 Switched Off 
To clear the Alarm/Warning message, press OFF on the control panel keypad followed by the ENTER key.

**Text displayed on controller.**
I looked in the controller log, but couldn't find an entry corresponding to the error date/time.

**Generator information**:
Generac Model 7030, 9Kw, Air Cooled wit Evolution controller

Thanks

jgyates, 2018-04-28:
Thanks! Let me look into this and see if I can see. I notice you did have an alarm / warning today (Alarm Log : 04/27/18 08:51:27 Charger Missing AC : Alarm Code: 0000) , but this appears to be a different time.

berenguerj, 2018-04-28:
Yes, the **Missing AC** alarm comes from a power failure we had this morning and lasted until until 5:55PM . I could't find an log entry for the 5:55:05 PM entry. BTW, I missed the following entry:

Time: 05:55:05 PM
Date: Fri 27-Apr-2018
 I just added into the previous thread.

jgyates, 2018-04-28:
@berenguerj ,

Question: Can you look at your log (start stop log or outage log and give me the times the power went out? Here is what it looks like. There are 2 places that an alarm is read in the controller registers: 1) the alarm log, and 2) the lower half of register 0001. Based on the events as I can see them it looks like you had a power outage, the log entry was entered for missing AC (the software knows this value already), when power was restored the alarm condition was displayed in reg 0001. Also, when the Alarm Code is 0000 this denotes a warning, which also matches Charger Missing AC. If the time of the outage matches the log entry for Missing AC, and when power was restored is when the Unknown alarm condition email was sent then it looks like I can say reg0001 value 2b is Missing AC. This would make sense since a warning code would be secondary to a power outage status.

BTW, I received an email because you enabled auto feedback when the event occured so it sent me a copy of your registers as the Unknown entry occured, this was very helpful. Thanks. I will post and update to genmon shortly and let you know when it is there.

jgyates, 2018-04-28:
I checked in an update that should address this error.

berenguerj, 2018-04-28:
Hi @jgyates 

There is a great benefit from all of us by having the Automated Feedback option turned on. It seems you are very passionate about your project, so at least we should contribute with feedback and support. Thanks!!

I'll try to explain what happened yesterday to see if the logs make sense. Here they are for yesterday with my interpretation after the ##:

Logs : 
    Alarm Log : 
        04/27/18 12:33:13 Charger Missing AC : Alarm Code: 0000 - ## **City power still down, I set the unit from Auto into Off**
        04/27/18 08:51:27 Charger Missing AC : Alarm Code: 0000 - ## **City power went down**

    Start Stop Log : 
        04/27/18 17:43:56 Stopped - Auto  - ## City power was restored
        04/27/18 17:13:16 Running -  Utility Loss ## My guess is that this is sensed when I set it to Auto
        04/27/18 17:13:09 Stopped - Auto - ## I switch the unit from Off to Auto
        04/27/18 12:32:57 Switched Off  - ## I switch the unit from Auto to Off
        04/27/18 12:03:22 Running - Utility Loss ## My guess is that this is sensed when I set it to Auto
        04/27/18 12:03:08 Stopped - Auto  - ## I switch the unit from Off to Auto
        04/27/18 06:56:59 Switched Off - ## I switch from Auto to Off

Since we are getting so many power failures, after power was restored following Hurricane Maria, I only set the generator into automatic mode when we are at home.

Every morning, since everyone is going to work and/or school, I switch the generator into off mode (06:59:59). I see no benefit from having the generator turning on when there is a power failure and there is no one is at home, specially when the price for LPG is at $2:50 per gallon and this unit spend around .87 gallon per hour at 1/2 load. Yesterday we had another power failure at (8:51:27). When I get home from work at noon, I switch the generator back in automatic (12:03:08) and then back into manual when I left back to work at 12:32:57. After returning from work I set the unit back into Auto at 17:13:09 when the city power was still down. City power was restored at 17:43:56. 
Hope this help. Let me know if you need more information.

jgyates, 2018-04-30:
Thanks! I have everything I need on this one. This confirms my theory. Let me know if the updates do not fix this issue if it occurs again.

tmorse305, 2018-05-01:
I have a Nexus Controller on an air cooled Generac 20KW generator.
Here are 2 unknown codes:

Change Oil & Filter: 0x16
Inspect Spark Plugs: 0x19

jgyates, 2018-05-01:
btw, if you want to send your registers without posting them to the forum you can do this directly from ClientInterface.py

from clientInterface.py type:

     generator: sendregisters

This was added in v1.7.4

tmorse305, 2018-05-02:
[image]
Here is a screen shot where the events occurred.  I sent the register data via the Client interface

jgyates, 2018-05-02:
Hmm, I have not received your registers. I have the change made in the code, I just need the registers to make the final validation. When you send the registers via Clientinterface it should give you a response back "Registers submitted"

     >generator: sendregisters
     Registers submitted

jgyates, 2018-05-04:
I don't see a screen shot in your last post. To send an attachment, you must post them from the web forum. Github will strip attachments that are sent from an email attachment, so I don't know your response when you tried to use the "sendregisters" command.

You can send your registers a few ways:

1) run ClientInterface.py

     python Clientinterface.py <IPaddress of your Pi>

     generator: registers

 Then copy the output of the listed registers to a file, then paste them into this issue thread.

2) run ClientInterface.py

     python Clientinterface.py <IPaddress of your Pi>

     generator: sendregisters

jgyates, 2018-05-04:
@tmorse305 

I just checked in a change that should 

1) Add the entries for your log that are missing. Let me know if they do not match your actual generator log.
2) A button on the Monitor page of the web UI can be clicked to submit your registers. I would still be very interested in seeing your registers since you have two service items due and you have a Nexus controller. There are now dates on the Maintenance page and I want to validate that the service dates match the service item.

Thanks
Jason

tmorse305, 2018-05-08:
I am running v1.6.0, is that the latest?  I clicked the Upgrade to the latest version button but forgot to note the version prior to clicking it.

I don't see a button to send registers.  I tried the generator: sendregisters again but got the same result, see attached.

Thanks,

TIm
[image]

jgyates, 2018-05-08:
I forgot to mention, once you update the software you need to refresh your browser.

tmorse305, 2018-05-09:
Thanks, all set, software updated, registers sent.  The logs in the app also match the display on the generator.

jgyates, 2018-05-09:
@tmorse305 , glad your logs are working now. I still have not received any registers. I would check your logs, specifically /var/log/genmon.log and /var/log/mymail.log to see if you have any errors. Are you receiving emails when you start genmon?

tmorse305, 2018-05-09:
I don’t have email set up as I am using NodeLink for alerts.  I just sent the registers via the Client app.  This time I got confirmation of submission.

 

Thanks again for all your help,

 

Tim

[4 more replies]

closed 2018-12-05 by jgyates