jgyates/genmon #14: instruction questions?

ts4s, 2017-04-12:
have the instruction been check recently from a fresh build?  referring to [https://github.com/jgyates/genmon/wiki/Setting-up-your-Pi-and-genmon](url)

I ask due to the issue i'm having? currently running a fresh version of jesse pixel.

- this section was already completed when i got to it. 

    > You need to setup the serial port on the PI 3. Open the terminal window again and type 
    >
    >sudo nano /boot/config.txt 
    >
    >and add the following line at the bottom:
    >
    >enable_uart=1 
    >
    >and then press ctrl-o to save the file, ctrl-x to exit nano.>

- not sure what i'm suppose to see here
   >Leave the terminal window open and confirm the name of the serial port of ttys0 has the alias serial0. Type ls –l /dev and confirm.

   I see this:
   >ls: cannot access –l: No such file or directory>
   follow by a list of /dev

- type in: 
    >  sudo python setup.py  install >

    result:

    >python: can't open file 'setup.py': [Errno 2] No such file or directory>

- with "minicom –s –c on" terminal just locks up and unresponsive. open a new terminal window and close out the old, i get "Device /dev/tty8 is locked." so that tells me i need to resolve something in the above steps?

- this command "sudo python setup.py install" not working either. sudo python will get me to a promp. So i'm guess the setup.py comes from one of the github (have to go check this) but didn't get copy when using the git command? i'm grabbing the zip file to try again later.

have to end it there for now and wait for some feed back & try the zips. 

thanks!

jgyates, 2017-04-12:
Hi,

Regarding your questions:

1) "this section was already completed when i got to it." Great. I assume that this means your serial port was already enabled. Let me know if I am not understanding your question.

2) "not sure what i'm suppose to see here". One Linux / Unix devices like the serial port show up as files in the /dev directory. "ls" is the command to list the files and directories. The "-l" is an option to the "ls" command so show details like file size and date. The file you are looking for is "/dev/serial0". So if you type "ls -l /dev/serial*" you should see something like this:

      lrwxrwxrwx 1 root root 7 Apr 12 01:06 /dev/serial0 -> ttyAMA0
      lrwxrwxrwx 1 root root 5 Apr 12 01:06 /dev/serial1 -> ttyS0

The first line is the file / device /dev/serial0. That is your serial port that you will use.

3) Regarding the question about "sudo python setup.py install ". This assumes you have downloaded the crcmod files from "https://github.com/gsutil-mirrors/crcmod" and unzipped them to your file system. Then you must be in the directory that you unzipped that contains the file named setup.py. Another option that may be easier is to type the following:

     sudo apt-get install gcc python-dev python-setuptools
     sudo easy_install -U pip
     sudo pip uninstall crcmod
     sudo pip install -U crcmod

This will should download and install everything for you.

4) Regarding minicom: You can just type "sudo minicom" The "-s -c" are optional and the "on" is a typo. I will correct that in the wiki. once you type "sudo minicom" you will need to tell minicom to use /dev/serial0. To do that, one minicom is running type "Ctrl+A" then "Z". This will bring up a menu. Once the menu is up type "O" to configure minicom. In minicom you can always type "Ctrl+A" then "Z" to get to the menu and select other options such as "exit minicom". Once you tell minicom to use /dev/serial0 you can go back to the main screen. From there if your serial port is setup correctly, and you have your cable connected to your Pi serial port and transmit and received connected at the far end of the cable (i.e. looped back) then when you type on the main minicom screen, what you type will show up. If you remove the loopback, it will not.   This will validate your serial port is working properly before using genmon.py.

5) you last question looks like a repeat of an earlier question. Let me know if I am wrong about this.

This project does require some knowledge of Linux to setup and the documentation does reflect that requirement. Also, that page is relatively new so you feedback will go to making it more clear. 

I hope this helps. Let me know if you have any other questions. 

Also, what type of generator do you have?

Thanks
Jason

jgyates, 2017-04-12:
I verified that "sudo minicom –s -c on" is not a typo. The -c is optional. The "-s on" will bring up the same menu I described earlier "Ctrl+A" then "Z" then "O".

jgyates, 2017-04-13:
Also, I have updated the wiki with some improve formatting and the above info on an alternate way to install crcmod.

ts4s, 2017-04-13:
Great! The doc is a little vague at times as to what action is need. Some items are verbatim, other are intuition. 

  I have some more inquiries: email setup, how important is it to have setup prior to running genmon.py? I live in the country where internet is dsl, cell phone, or nothing. I fixed a few items and tried to launch the app, and am getting error in the emailcommandthread? I'll upload a photo here once I switch over from photo to Mac and shared a few more items.

jgyates, 2017-04-13:
That is an excellent question. I never occured to me to make the email optional. If you can send your log file for genmon. It is located in /var/log/genmon.log that would help. The way things are suppose to work is that emails are queued inside of mymail.py, if there is a temporary problem sending they will be resent. Emails are never discarded so if you don't have an valid email or internet connection then they will just queue forever (until your run out of memory). I can honestly say that I have not tested a situation where email credentials are not valid or internet is not available. I will try to test this and see if I can make email optional.

I looked at the email code again and there will be a lot of logging if your email credentials are not valid, but it *should* not crash or anything. I theory it should work without email, but I have never tried it.

I agree with the setup issues, it is not exactly a breeze to setup. I am working on something that may make part of that easier, but it is not a quick fix.

ts4s, 2017-04-13:
[genmonlog.zip](https://github.com/jgyates/genmon/files/918426/genmonlog.zip)

documentation is always fun... :-) ever changing process as the software changes. 

[image]

above photo is when i try to run the program. thought i got what is my next question, the next photo about the alarm.txt.  

[image]

alarm.txt
- notice this file need to be copy out to the /home/pi folder when trying to run the genmon.py. easy fix but something to note. 

logs are attached

jgyates, 2017-04-13:
ok. Just to validate, you are not connected to the internet, correct? If that is the case I see the problem. Should be an easy fix to keep it from crashing, in the short term your /var/log/mymail.log will still fill be filled with error messages. In the longer term I can look at an option to disable email.

ts4s, 2017-04-13:
yup... no internet connection.

ts4s, 2017-04-13:
dumb question, if i run git from the home directory, that will download the whatever i tell it into the home folder right? just make sure i understand the whole git clone process.

jgyates, 2017-04-13:
As with most things there are several ways to do it. I clone to my PC then use "scp" to transfer the files. I believe the answer is yes. I don't really use git to transfer files.

I checked in a new version of mymail.py that will fix the crash and allow you to run with no internet connection. Let me know if you have any other questions.

I also looked into the ALARMS.txt in the docs. It does state that the file is required, maybe it should be clearer. I am working on a longer term fix, it will automatically download and setup genmon but I have a bit of testing and coding to do on that front.

jgyates, 2017-04-13:
one other note: your genmon.log files look like you are not connected to your generator or your serial port is not working. I have a simple serial test app that I can put in the repository. if it works for you then I can create some documentation. Basically it works like this:

if you serial port is working and your transmit is looped back to your recieve the test will indicate success, otherwise a failure. To run the test type "python serialtest.py". If you are using the built in port on a raspberry pi then there are not command line arguments. I put the test app in /OtherApps/serialtest.py in the github repository.

ts4s, 2017-04-13:
i haven't hooked that part up yet. parts where support to be here today but snail mail is running slow with prime. i grab the new mymail file but it looks like i might be trying start over with another fresh built sd card. Was trying something for work arounds. 

not that it matters at the moment, my generator is a 20K air cool, older controller (can't think of it at the moment). Tomorrow will give me a chance to go reclaim one of my other pi3 from my parents house :-)

jgyates, 2017-04-13:
OK, no problem. I uploaded a new serialtest.py the first version had a bug.

Also, make sure you generator has either a Nexus or Evolution controller. That is the only two that are supported.

It probably has the nexus controller:
https://github.com/jgyates/genmon/blob/master/Diagrams/Nexus_Controller.jpg

The newer models have the Evolution controller:
https://github.com/jgyates/genmon/blob/master/Diagrams/Evolution_Controller.jpg

I setup a spare PI in the last few min and validated the serial port stuff in the docs is working as described. I found one other issue in the wiki. Now that I validated serialtest.py I will add some stuff to the docs on that.

ts4s, 2017-04-13:
nexus is it.... had to replace it a few month after having it... can't run them in quite mode or it mess up the controller.

jgyates, 2017-04-13:
I just checked in new versions of mymail.py, genmon.py and mymail.conf. With these editions if you want to disable email add the following line to mymail.conf
disableemail = true

I have tested this and it appears to work fine.

ts4s, 2017-04-13:
ok, starting over with fresh build an working through the steps and reviewing the info here, now i have a better sense of whats going on. 

here's my notes were i was getting hung up on: 

1. > Leave the terminal window open and confirm the name of the serial port of ttyAMA0 has the alias serial0. Type ls –l /dev and confirm.>

   command should read "ls -l /dev/serial*" otherwise you get a list of all the device on the system. (that what threw me) 

   now one can see this:
    > lrwxrwxrwx 1 root root 5 Mar  3 10:39 /dev/serial0 -> ttyS0
     lrwxrwxrwx 1 root root 7 Mar  3 10:39 /dev/serial1 -> ttyAMA0 >

2. more of an fyi(you may already know this) - this 'enable_uart=1'  is added when in raspi-config and  enabling the serial port which also gives "serial0 ->ttyS0'. (probably shouldn't be doing the enable serial port in raspi-config, habit acquire from other projects) 

sidenote: After working through this and reading up on the pi3, looks like I'll need to set my config for the serial1 since the port are remapped on the PI3, if i am to use the ttyAMA0. 

sidenote: Are the true and false case sensitive? noticing some are and some aren't depending on which file one is in.

3. minicom - instruction was there, now gone. I know when i tried to run it, it froze or was not responsive. I could open a new terminal window and move on. But the minicon window would not do anything. 

Pretty much stuck here i guess til i get the cable built for the serial com. would be nice to include the photo on hackaday here in the instruction, a link to it. I am not seeing in either section, Hardware or Software. 

I did try to move on to the ClientInterface.py localhost steps and failed there... my be due to being at work on dhcp network. I'll try with a static address when i get home.

jgyates, 2017-04-13:
I am guessing your issue with minicom is due to your ttyS0 having an alias to serial0. I have multiple Pi3s and I have two setup with serial, both map serial0 to ttyAMA0. Not sure what that is happening to you. 

I removed the minicom references and added description of /OtherApps/serialtest.py. Try this:

     python serialtest.py /dev/serial1

It will fail since you do not have a loopback, but it should not lock up like minicom. I am thinking that minicom is accessing ttyS0, which is associated with the console so that can cause the system to appear to be locked since nothing is being output to the console anymore.

I will update the docs to make it "ls -l /dev/serial*"

I agree on serial1. You should be using the alias mapped to ttyAMA0. I don't know if having serial0 mapped to ttyS0 will cause you any other issues with your system, but as far as genmon, it should be fine.

jgyates, 2017-04-13:
Photos are in two places in the repo: genmon photos are in Diagrams folder in the root of the project.

Screen caps of the web app are located in /HTTP/ScrenCap

The issue with ClientInterface is that specifying localhost assumes genmon is running on the same system that executing ClientInterface.py. You can specify a remote address however you that address must be available to your network. For example, I can run "python ClientInterface.py 192.168.11.15" on my local network on may laptop in a console window and view generator stats. I also can remotely log into my pi from a console windows via "ssh pi@192.168.11.15" and then enter the password for the pi user. Once I log into the raspberry pi i can run "python ClientInterface.py localhost" since I am running ClientInterface on the pi3 in the latter example.

ts4s, 2017-04-13:
[image]

here's a screen shot of clientinterface.py run locally on the pi3.

sorry about closing this thread, wasn't meaning too.

jgyates, 2017-04-13:
Is genmon.py running in the background or in the foreground from another console window? I does not look like it is. This is the message you what is normally displayed when ClientInterface.py can not connect to genmon.py

ts4s, 2017-04-13:
well that explains it... duh.... just cruising along and didn't have it running. Just ran genmon and then client interface.py, now we're getting somewhere. Guess that's what performing each step :-)

ts4s, 2017-04-14:
Ok, got everything setup (I think) and hooked up to the generator😁 So how do I now if the last part works? The crontab/startgen part. 

Serial pass on the bench, is there a test for testing communication to the generator? And then what the next step for the webinterface?

jgyates, 2017-04-14:
If genmon.py is running in the background then if you use ClientInterface.py to connect to genmon.py and get status:

     python ClientInterface <IPADDRRESSOFPI>

If the above line connects then you will see a prompt ">" that you can enter commands.

First type "generator: help" for a list of commands type "exit" to quit genmon.

if you installed flask per the setup in the wiki "sudo pip install Flask"

They from a console type "python genserv.py &" this will load it in the background. You can add the line "python startgen.py &" to you startgen script if it is not already there.

If genserv.py is loaded they you bring up a web browser to the IP address of your PI at port 8000

    http://YourIPAddressGoesHere:8000

from here you should see the web interface.

jgyates, 2017-04-14:
sorry, left of the extension:

     python ClientInterface.py <IPADDRRESSOFPI>

ts4s, 2017-04-14:
"First type "generator: help" for a list of commands type "exit" to quit genmon."

What if i don't get a list of commands? I have just a blank line. Exit command works.

jgyates, 2017-04-14:
I could be a few things. I would need to see genmon.log, client.log and your genmon.conf file. Also a screen cap of the output of the "ifconfig" command on your PI would be helpful also.

ts4s, 2017-04-14:
Client.log is where? Not seeing it in /var/logs

jgyates, 2017-04-14:
client.log is in the same directory that you executed ClientInterface.py so i could be on your PC or or on the PI, depending on where you executed the program from

ts4s, 2017-04-15:
Genmom.log says "unable to open alarm file: [errno 2] no such file or directory: ./ALARMS.txt"

File is in /home/pi & /home/pi/genmon

Client.log says "error: Connect[errno 111] connection refused" 

Lost here, this was localhost when ran.

I've hardcoded the IP address, my network is a managed network. I can vnc & ssh into the pi without issue. Was think is was my network but I not seeing that to be the issue.

jgyates, 2017-04-15:
OK, here is the problem. I checked in a version of genmon.conf on Thursday that had the following line:

     alarmfile = ./ALARMS.txt

That works when ALARMS.txt is in the same directory as genmon.py, but not when called at boot up from a crontab. I fixed that earlier today and checked in a new version of genmon.conf that has the following line instead:

     alarmfile = /home/pi/genmon/ALARMS.txt

relative paths do not work when called from a script in a crontab job so genmon.conf needs to have the full path.

Sorry about that, the timing of the checking was bad timing for you. You can either edit your genmon.con (sudo nano /etc/genmon.conf) or checkout the updated version of genmon.conf. 

The changes I made recently were to improve the install process. With the latest checkins if you checkout everything in /home/pi  it will work as is. I also have a script that is can be called from  from crontab. The format is a little different that was was in the wiki example from before Thursday evening. Your script should still work but if you wanted to use the new one the format is:

     /home/pi/genmon/startgenmon.sh start

to stop genmon and genserv use:

     /home/pi/genmon/startgenmon.sh stop

Let me know how it goes...

ts4s, 2017-04-15:
Ok, think I got it... made the change for alarm, so we're good there. I'm in ssh and ran the new startgenmon command, I get:
Prompt:/ Warning: not receiving data. : ,
* running on http://0.0.0.0:8000/

Open browser on local machine(my Mac) and gave it the pi address:8000

I can the see in my ssh window that the mac "get / http/1.1" 404 - & "get /favicon.ico http/1.1" 404 -

Thought?

jgyates, 2017-04-15:
Here is what the output of genserv.py should look like when it is running:

root@genmon:/home/pi/genmon# OK : Auto, Off - Ready
 * Running on http://0.0.0.0:8000/ (Press CTRL+C to quit)
192.168.11.106 - - [14/Apr/2017 23:14:47] "GET /cmd/getbase HTTP/1.1" 200 -

The directories named static and template need to be at the same level as genserv.py and myclient.py needs to be in the same directory as genserv.py. This is one of the things that was fixed in the old repository structure. With the new structure it should all run in place. I would recommend just doing a fresh checkout of the repository. You should be able to keep your .conf files in /etc. To do this do the following:

\# this will make a backup of what you have now. From the /home/pi directory type
sudo cp -r genmon genmonold

\# now checkout a fresh repository from the /home/pi directory
sudo git clone https://github.com/jgyates/genmon.git

\# update your crontab
sudo contrab -e

\#and replace your existing crontab line with:
@reboot sh /home/pi/genmon/startgenmon.sh start

Let me know how this goes.

jgyates, 2017-04-15:
one other thing, when you check out the repository, donrt forget to make startgenmon.sh executable. From /home/pi/genmon type:

sudo chmod 775 startmon.sh

Also, when you checkout genmon you don't need the sudo, instead just type:

git clone https://github.com/jgyates/genmon.git

jgyates, 2017-04-15:
It will work if you do type sudo, it is just making all the checked out files owned by root instead of the pi user.

ts4s, 2017-04-15:
Ok... I have a clone copy (the 1.2.6) here on the Mac. Just trying to figure out the best way to remove the old genmon and copy over a fresh copy.

I was able to add the 2 directory to get to the webpage 😉 Just no data on the page. So tomorrow I'll see if I can get 2nd pi3 update, while I go work on a buddy's network and then swap sd card when I return home.

jgyates, 2017-04-15:
Backup the files on your pi, from the /home/pi directory:
sudo cp -r genmon genmonold

this command will copy over a file or directory. To copy a directory:

open a console window on your mac and use the "scp" command:
scp -r  /home/user/Downloads/genmon pi@your.server.ip:/home/pi/genmon

you will needed to change the source directory in the above command.

what is in genmon.log?
what is in /var/log/genserv.log?

Also, please validate that /etc/genmon.conf contains the proper:

alarmfile = /home/pi/genmon/ALARMS.txt

ts4s, 2017-04-17:
view of web page
[image] 

ssh view of Ifconfig & startgenmon.sh start:
[image]

VNC view of the genome.logs since my prompt is tide up at the moment in the above screenshot
[image]

Client.log show nothing. 
gender.log show nothing.

Clean copy of the latest build, pull this afternoon. 

genome screen schots
[image]
[image]
[image]

Not sure where to go from here?

ts4s, 2017-04-17:
grrr... the genmon.conf, is not right... i know i edit it and copy it to the /etc... the alarm setting is fixed and the evolutioncontroller = false.

ts4s, 2017-04-17:
nvm... took screen shot of the wrong pi... shut that one of.. so genmon.conf is setup correctly for those 2 settings.

ts4s, 2017-04-17:
one quick pointer

> Open a terminal window to the genmon directory. The project file startgenmon.sh is a script that will start and stop genmon and genserv. First we need to make the script executable:

 sudo chmod 775 startmon.sh
To start genmon and genserv type:

 ./startgenmon.sh start

> 

the line "sudo chmod 775 startmon.sh" should probably be "sudo chmod 775 startgenmon.sh" with all the changes that's been made?

jgyates, 2017-04-17:
I updated the typo in the wiki. Thanks!.

A couple of things: 
1) Let's take the web interface out of the picture for the moment. Since you are running without an internet connection there is an issue with the main index.html for the web app. It pulls a standard java applet from code.jquery.com to perform JQUERY request to the Flask app genserv.py so this will fail. You don't have to shut down the app genserv.py, just close the browser page making the request.

2) Once all web pages are closed,  restart the scripts by killing them (./startgenmon.sh stop) and restarting (./startgenmon.sh start)

3) Since we are taking the web UI out of the picture temporarily, let's focus on genmon.py and only interface with it with ClientInterface.py. Open a second ssh session and type:

    cd genmon
    sudo python ClientInterface.py 

you should get a prompt. At the prompt type:

generator: help
generator: status

What do you see at each step?

jgyates, 2017-04-17:
One other thing that I noticed in your post: I see that after you loaded ./startgenmon.sh start, there is what appears to be an line on the screen "WARNING" : Not receiving data"..

Is this printed to the screen or is this a remnant of a previous print out? The reason I ask, the only time this should be printed is when data is not being received by genmon.py AND your are running ClientInterface.py. What is really confusing is how this is being printed out but your do not appear to be running ClientInterface.py from the information you show. Are you running ClientInterface.py form startgenmon.sh?

Since you updated your code I have made the following updates to the master branch:
- Updated genmon.py, ClientInterface to make the command line interface more forgiving by accepting a space after "generator:" in the command string and an error to be returned on  zero lenght  commands and commands that are not formatted properly.
- Added a file to the /static directory and updated /static/index.html to the web interface to allow the web interface to operate without an internet connection.

I still would like to validate your genmon.py is working properly with ClientInterface.py. At the moment we have not indication that your serial port and cable is working with the generator so this should be validated.

ts4s, 2017-04-17:
the photo in question, that was taken after swapping to the update sd card, fresh boot, and going through the steps of running the software.  

Should be easier to get file today, have access to window machine with tools ;-) down side is i pull the equipment out to work with since i'll be gone for a few days. 

here the result of your request... looks like the 2 file are playing nicely... now I'm assuming at this point can verify communication to the generator by using the commands?

[image]

jgyates, 2017-04-17:
depending on if you updated to the latest code this morning you will need a space after the colon in the "generator: status" command. If you updated to the latest code you will not need it. I made that change last night to make things a bit more user friendly.

type:
generator: status
generator: monitor

The monitor command should show the number of packets sent and received and any serial errors that have occurred.

ts4s, 2017-04-17:
serial cable: 
[image] figure i go head and get this out there. 

no, i have no update to the new code yet, holding off as that ten-dent to cause the problem and at least be on the same page. 

Will the apt-get update work for updating or do i need to clone and transfer the files.

jgyates, 2017-04-17:
just clone and transfer.

Thanks for the pic of the cable. I can't tell from the angle what pins you are connected to on the white molex connector.
[image]

In the attached picture white is transmit from the controller and should be connected to receive on the raspberry pi serial 9 pin connector (pin 2). If you connections are like the ones in my pic then your cable looks right. When you ran serialtest.py did you loop back pin 7 and 8 in the white molex cable. This would allow the test to transmit through your cable.

ts4s, 2017-04-17:
Green is 8 and orange is 7, I do have better photo.

I will try to clone once our network is restored.  it's definitely Monday, network took a noise dive.

ts4s, 2017-04-17:
[image]
better photo :-)

ts4s, 2017-04-18:
update complete! so now if i had this connected up to the generator and ran genmon, clientinterfaces, and type in gen:status; i would see com status for my gen like: status, engine state, line state, last log entries, and data time, correct? (gen is short for generator,getting lazy on typing it out).

anything thing else i should try? I have a couple of days before i can get this hooked up to the gen again.

jgyates, 2017-04-18:
If you have not run serialtest looping back transmit to receive through your cable, that would be a good test. This will show that the cable and serial are both working.

After you connect pin 7 to 8 in your cable, change directories to  /genmon/OtherApps on your PI then run "python serialtest.py"

Once that is validated then load genmon, then ClientInterface, main commands are

generator: help          show all commands you can run
generator: status       for general status
generator: maint        for maintenance info
generator: outage      for info on last outage and utility voltage
generator: logs          for logs
generator: monitor     for information on genmon stats like packets sent and received, etc

The rest of the commands are mostly for setting exercise time and remote start and stop, etc

These 5 commands will give you most of what is in the web interface. Once we get that validated we can revisit the web user interface.

Let me know how it goes...

ts4s, 2017-04-18:
it pass the loopback test again tonight. 

so a couple of question: 

- is there plans to log the data for data collection purpose for review later down the road? say a year from now, i want to lookup runtime to figure out how much propane I have used during the year. basically data mining on the unit.

- i'm assuming this will work on any of the pi boards? I have a zero w that i could test with? just need to clone this sd card to another and see what happens.

jgyates, 2017-04-18:
The data mining idea sounds cool. I will give some thought on the best way to implement.

Yes, it should work on any pi board.

The only pi specific stuff is in gengpio.py and gengpioin.py. If you are not using that then you can move it to any system  that supports python, has an IP stack, and a serial device accessible via python. Will need to tweak the values in genmon.conf and possibly make the conf file readable form somewhere other than /etc/ but there is no reason the base stuff would not work on Windows or another embedded platform like the esp8266.

jgyates, 2017-04-20:
I added a program named genlog.py to the repository and added instructions to the docs. You should be able to log state changes from genmon.py with this program to a CSV file which can be read my Excel or any other spreadsheet program. It is an optional program that calls into genmon.py to get the data to log. You should be able to add it to startgenmon.sh to have it running all the time if you want to log data 24/7. At the moment it just logs the generator state (running, exercising, ready, alarm and service) and the time the state changed.

ts4s, 2017-04-20:
Evening, well I have the pi connected to the generator. Run the gen on.py and am getting error: timeout receiving slave packet for register 1f4 buffer: 0 
The 1f4 change with each line.

jgyates, 2017-04-20:
What does ClientInterface tell us when you issue the "generator: monitor" command?

jgyates, 2017-04-20:
you should see someting like this:

Packet Count: M: <somenumber> S: <someothernumber>

If the S: number is zero then you are not receiving anything anything back from the generator. This can be caused by a few things:

1) There is something wrong in the cable path from the PI to the generator
2) there is something wrong in the cable path from the generator to the Pi
3) the serial port is not working

The third one is not likley but it may be that the base port is working but there is something wrong in the TTL to RS232 conversion. If you did the loopback test through your cable then this is likely not the issue and it is 1 or 2 above.
 
If you are received something back (S: is non zero) we can go from there. Normally you would either be communicating or not.

The generator: monitor command also shows CRC errors. If you are seeing a lot of those errors, it could be a problem with your power supply / ground.

jgyates, 2017-04-20:
Here is the output of mine:

\>generator: monitor
Monitor:

Genmon Stats: 
    Monitor Health: OK
    Generator Type Selected : Evolution Controller, Liquid Cooled
    Generator Monitor running for 1:40:38.

Serial Stats: 
    PacketCount M:195809: S:195808 Buffer Count:0
    CRC Errors :0  Percent :0.00
    Discarded  :0  Restarts:0
    Packets per second:64.86

ts4s, 2017-04-20:
It's saying:

Monitor health: not receiving data.
Gen type selected: nexus controller, air cooled
Running for 0:01:20

Serial stats:

Packercount m:81 s:0 buffer count:0
Crc:0 percent:0
Discard:0 restart:0
Packets per second: 1.00

ts4s, 2017-04-20:
Guess wiring might be flip flop on the molex/db9 cable?

jgyates, 2017-04-20:
ok. So genmon has attempted to send 81 packets in the 1min and 20 seconds that it has been running and has received zero packets. 

First check /etc/genmon.conf to make sure you have your serial port selected . You are using /dev/serial1 correct?

[56 more replies]

closed 2017-08-30 by ts4s