jgyates/genmon #15: confirming correct Molex plug P/N

griffths, 2017-04-12:
I believe that the male Molex plug listed in the readme is incorrect, at least for my application.  (2013 22KW Nexus controller liquid cooled) .  The part number listed (WM3603-DN) does not appear as if it will mate with receptacle on the rear of my controller.  The part that looks correct is WM3703-DN.

https://www.digikey.com/product-detail/en/molex-llc/39-01-2080/WM3703-ND/61382

Pic of my controller accessory receptacle

[image]

Regards,

Jay

jgyates, 2017-04-12:
Hi Jay,

Thanks for the pic. That always make things easier.

WM3603-ND is the receptacle that corresponds to WM3703-ND plug. The WM3603-ND is the plug on the controller (in your picture above). The WM3703-ND is the mate. You are correct that you want to have WM3703-ND at the end of your cable.

When I made my cable I used both and made the receptacle one the other end of the cable because when I first started the project I was sniffing on Mobile Link to reverse engineer the protocol.

I will make the docs more clear. Thanks for the feedback.

Also, what model generator do you have?

jgyates, 2017-04-12:
I have updated the readme and wiki. Thanks for the feedback!!

griffths, 2017-04-12:
1800 RPM Quietsource 22KW 

 

Manufacture date: 11/27/13

 

Model number: QT02224ANAX

jgyates, 2017-04-12:
Thanks! Let me know how it works for you. I am that since your generator is a few years old it will have some entries in the alarm log. I only have one string decoded for the Nexus controllers alarm log. I can easily add others but some feedback is needed. Once you get everything setup and working this thread will tell you how to resolve any unknowns you might see when genmon outputs your alarm log.
https://github.com/jgyates/genmon/issues/13

griffths, 2017-04-15:
jgyates,

Can you lend some assistance identifying the correct 6 circuit Molex plug to mate with the Generac 6478 extension cable.  I purchased this a while ago thinking I would use the Mobile link product, but could never justify the annual monitoring cost so never did it.  After digging it up, it appears that it would be a perfect solution to connect power and data to the Raspberry Pi.  I think I have identified the correct connector as Molex part 03-01-2060 but cannot find any good resources to confirm my choice.

http://cablesandconnectors.com/PIX/moljr6pin.jpg

Thanks

jgyates, 2017-04-15:
I think this is what you are looking for (6pin plug like the one mounted on the mobile link unit that will mate the the 6 circuit receptacle on the cable):

Molex pn: 39-01-2061 (the 6 appears to be for thefor the number of circuits)

http://www.molex.com/molex/products/datasheet.jsp?part=active/0039012061_CRIMP_HOUSINGS.xml

Digikey pn: WM3602-ND 

https://www.digikey.com/products/en?keywords=39-01-2061

griffths, 2017-04-15:
Thanks for the info, I just ordered the parts to complete the data/power cable.

Didn't realize that my Pi was coming from across the Pond when I ordered so will try setting up a Python/Windows interpreter to test with until the Pi arrives.

Regards,

Jay

griffths, 2017-04-24:
Updating this thread for those who want to use the Generac OEM 0K2220 or 6478 extension cable....

The Molex connector didn't seem to fit really well and an investigation revealed why...Generac uses AMP brand connectors and they are keyed slightly different than the Molex ones.  I was able to force the two together but was afraid that it would eventually lead to problems so I am going to order what I think is the correct AMP connector and give it another go.

I will update this thread with the results.

Regards,

Jay

jgyates, 2017-04-24:
Are you referring to the 6 circuit connector (the one that connect to Mobile link?). I I have one of these cables. The 6 circuit does appear to not be molex. The only thing I plugged that one into was the mobile link.

griffths, 2017-04-24:
I should have been more specific in my previous comment but you are correct, I am referring to the 6 circuit connector that would connect to the Mobile link accessory.

I believe this to be the part I need from Digi-Key that will correctly mate with the Generac OEM cable:

https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/770969-1/A25635-ND/281356

weffz, 2017-05-07:
When reviewing these connectors I noticed something that might explain why the Molex connectors might not fit so well whereas the AMP connectors fit more easily overall.

It also seems strange too that they would use AMP brand connector at one end of the cable and then use Molex brand at the other. It would seem more logical that in manufacturing that they would use the same company for all of their connectors - whichever company was chosen would also go out of their way to provide a good deal.
When comparing the 6-pin AMP connector (part of their Mini-Universal MATE-N-LOK series) with a very similar series by Molex (Mini-Fit Jr 5557 Range) - the Molex brand has a pitch of 0.165" (4.20mm) whereas the AMP brand has a pitch of 0.163" (4.14mm) which is close enough for the pins to be pushed together using the two different brands. With the very slight measurement difference between them - unless there are other mechanical differences for instance where the latch goes, or any other connector keying that might make using them together unworkable.

The Molex WM3703-ND is the 8 pin version with pitch 0.165" (4.20mm) and is the female connector.

The AMP A25575-ND is the 8 pin Male or female version with pitch 0.163" (4.14mm). See https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/770579-1/A25575-ND/298536   For pin/socket inserts see  https://www.digikey.com/products/en?FV=ffec11a6,ffe0014b

Over the width of 4 connections with pitch 0.163" (4.14mm) is 0.652" (16.56mm).
Over the width of 4 connections with pitch 0.165" (4.20mm) is 0.660" (16.80mm).

With just a difference of 0.24 mm across the 4 pins/sockets attempting connection - the difference in fit should be unnoticeable.  Until I have examples of each brand connector I won't be able to check for any other differences (keying/latch positions) between them that makes using the wrong brand connector unworkable.

As yet I don't have a generator nor the #6664 accessory I ordered which should help determine which brand of connector is used on the current Generac generators.

For all we know Generac might have even used both brands at some stage so they can trade-off orders from both companies so as to get a better deal - though logically I think they would use just the one company and the one series of connectors in all their goods so as not to introduce compatibility issues between the 2 products.

weffz, 2017-05-07:
Further to this - I have also found from Digikey searches that there is a brand of AMP connector that shares the same pitch of 0.165" (4.20mm) as MOLEX and that is the AMP VAL-U-LOK series. mechanical differences with keying or latching are unknown as yet.

The 8 pin AMP connector A30593-ND  has the same pitch of 0.165" (4.20mm) as the MOLEX. See https://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/794954-8/A30593-ND/701598 for details.

Until I have a set of connectors of each type (the #6664 cable should be a a good example) and a set of each type of connector from MOLEX and AMP so as to determine which brand they actually used - assuming they have only ever used the one brand and series of connectors.

I think even with a digital micrometer it will be difficult to determine the brand from a measurement as a difference of 0.24mm across the face of the different connector types is quite small to measure and perceive.  

Unless someone has further information to determine which brand of connector has actually been used where - the discovery that the ML cable uses an AMP connector on one end - to me implies that the connector at the other end is also most likely an AMP connector. It doesn't make sense that they would use a different brand .. unless the ML module had used a different brand in error by being designed by a different company but as far as I know both the #6664 and #6483 modules were all designed by Generac - so connector brands and series should all be the same.

The aspect that remains unclear to me is whether the 0.163" (4.14mm) pitch AMP-only VAL-U-LOK connector range has been used - or whether the actual pitch is meant to be 0.165" (4.20mm) allowing the connectors from both AMP (MATE-N-LOK) and MOLEX (Mini-Fit Jr 5557 Range) to be used.

The discovery that the 6-pin 0.163" (4.14mm) pitch AMP-only VAL-U-LOK connector was the only one that fitted the ML plug to me suggests that perhaps all of the connectors should perhaps be AMP VAL-U-LOK and that the close similarity in size of the MOLEX Mini-Fit Jr 5557 range)  may have allowed them to fit with the 8-pin controller plug when it should be the AMP VAL-U-LOK connector range only being used.
Once i receive the 6664 accessory and the #006271-0 Generac Generator I will do some further measurements to see if I can determine which brand connector has been used.

Meanwhile if you can get the MOLEX 0.165" (4.20mm) pitch connector sockets to fit into your generator's controller plug - then the slight difference in connector pin placement overall should have minimum impact on their usage. As long as it will plug in and latch properly it should be good to use.

griffths, 2017-05-07:
I just cheeked my OEM 6478 cable and it is not clear what brand connector is on the 8 pin side but the link weffz provided appears to be the same.  This would made sense since the 6 pin connector is definitely AMP.

The trouble with the Molex brand connector is the keying of the shell.  There is a difference between the AMP and Molex brand which is the reason for the tight fit.  I was able to force the Molex connector onto the AMP end of my cable but it was deforming the key way on the connectors.

Here are pictures of the AMP 770178-0 connector, OEM AMP 6 circuit connector and the Molex 39-01-2061 connector.

[image]
[image]
[image]

weffz, 2017-05-07:
It is no wonder the different brands for the 6 pin port connectors don't fit together - the keying is completely different.

Based on the visible image of the 8 pin controller plug at the very top of this Issue - on the side with the latch (I'll call that the top) - on the top side the two outer connector inserts point down for keying. on the bottom side the two inner connectors face down - a very distinctive pattern.

On the 8-pin 0.165" (4.20mm) pitch AMP A30593-ND VAL-U-LOK connector when I check the keying - using the provided 3D model pdf - it matches that of the controller port. 

On the 8-pin 0.163" (4.14mm) pitch AMP A25575-ND MATE-N-LOK connector when I check the keying - on the top row the two outer edges face out - which is completely different to the controller port.

On the 8-pin 0.165" (4.20mm) MOLEX WM3703-ND Mini-Fit Jr 5557 connector when I check the keying - using the Digikey image (3D file unavailable while molex.com is down) and straining my eyes the keying also matches that of the controller port.

Conclusion - the controller port keying matches that of the 0.165" (4.20mm) pitch AMP connector as well as that of the 0.165" (4.20mm) pitch MOLEX so I would expect that either the AMP A30593-ND or the MOLEX WM3703-ND connectors should be able to be used to connect the controller plug.

For whatever reason though, the 6-pin connector at the end of the ML cable is a 0.163" (4.14mm) pitch AMP connector - the keying is quite distinctive. Generac may have kept with AMP as the only Brand (or not)- but used a different series of connector (MATE-N-LOK) for use with the Mobile-link accessory.

At least now I know to identify a connector series by the pattern of the connector keying.

I am also now confident too that we can use either brand connector for the controller port as essentially the MOLEX and AMP connectors with the same 0.165" (4.20mm) pitch have identical keying and therefore can be interchanged if needed. This makes sense too as the connectors are keyed  in an attempt to prevent incompatible plugs and sockets from mating easily - it doesn't of course prevent the brute force approach from sometimes forcing the issue :)

weffz, 2017-05-11:
@gjyates @griffths  One thing people here need to be aware of is that the images here (especially of these connectors) do appear much larger than their actual size - due to viewing them on a large screen monitor in a lot of cases and the lack of any sizing keys ie. including a common object of known size next to the device. eg box of matches
It would help perceptions of the overall size of these connectors if they were placed next to a ruler or other measurement scale in some of these images.
Without having seen an actual generator controller port or a real cable and only the images here - these have skewed what I thought the connector sizes were - as the images on my screen were a lot larger than actual.
Although I had read (& typed here) that the pitch of these connectors was 4.20mm - I hadn't quite grasped the reality of their overall size. The outer area of the section that will insert the female sockets into the panel male plugs is approx 16mm x 8mm.
The images of the Mobile-link module electronics board is such that the distance centre to centre of the pins of the 6-pin board connector - while that measures 15mm on my screen - in reality that distance is only 4.14mm so while the green electronics board on my large screen monitor measures  across 24mm and vertical 20.3mm (this is on a 30" diagonal monitor) - the reality is the actual size should be 6.62mm across and 5.6mm size vertical .. a much smaller size.
I was expecting to receive from Digi-key a connector similar in size to a white molex hard-drive power connector (old style drive) - what I received today was a much smaller connector with the plug-in part just 16mm x 8mm - such that I thought for a while I had ordered the wrong part.
So - beware one disadvantage of using a large screen monitor - with no size keys visible - your perception may get skewed as a result.

griffths, 2017-05-11:
@weffz 

Thanks, You make a a good point that I hadn't considered since I have access to the parts.  Moving forward I will include some form of reference with my pics to assist others in identifying their real size.

closed 2017-07-01 by jgyates