jgyates/genmon #16: Please read this before posting

jgyates, 2017-04-12:
You  feedback and questions are welcome and they help improve the documentation and software. The following are way we can maximize the feedback response.

1) Review open and closed threads to see if your question has been asked before
2) Please post new questions to a new thread
3) Include your generator model along with what controller you are using (e.g. Generac 6730 with Evolution Controller, Air Cooled)
4) Please include a photo if applicable (i.e. photos work great for enclosure and cable questions) or copy and past output of programs if applicable.

Thanks
Jason

KortP, 2018-07-26:
I am having a 22KW Generac 7042 installed and am very interested in your project. Is there a way I can buy a complete assembled WiFi monitor from you for this unit.

jgyates, 2018-07-26:
There is already a thread on this topic:

https://github.com/jgyates/genmon/issues/62

I would suggest looking in this thread and posting the question there.

closed 2019-07-06 by jgyates

caseyd, 2020-01-18:
Does this body of code work with Kohl generators?

jgyates, 2020-01-18:
The software does not currently support Kohler generators. Adding support for new generators probably possible however that discussion would need to take place in a new issue thread (testing support is needed by generator owners as I do not have a Kohler generator).  The purpose of this thread it to publish the guidelines for posting.

caseyd, 2020-01-18:
thanks, I'll do some more homework and start a new issue thread to see of
folks are interested.

----------------------------------------------------
(aggregation account, which means I check it every few days for email from
various accounts)
caseyd@gmail.com

bbrown-sc, 2020-10-01:
First, thanks jgyates for developing this. My existing online solution is an outdated genlink application running on an outdated OS (Windows 7) and PC. 

I have a 2008 generac propane 25kW industrial liquid cooled QT02524, H-100 control panel (with internal modem) and a 400A HTS "W" Type transfer switch. The Generator Monitor is hosted on a Raspberry Pi 4 Model B Rev 1.2 with 1 × Serial HAT (RS232) running Genmon V1.14.33. I have a couple of questions/comments.

1) I was having difficulty connecting serially until I changed the Modbus slave address to 100 (64Hex)
2) How can I determine which genmon.conf parameters are applicable to my setup? For example, I needed to set "usesensorforfuelgauge = false" so I could get the Tankutilty External Tank Fuel Monitor to function.
3) I have set both "controllertype = h_100" and "hts_transfer_switch = true". Is there anything else special I need to do for my configuration?
4) What do "fuel_units = cubic feet", "full_rate = 142" and "half_rate = 77" (my generator fuel usage specs) parameters do in the H-100 settings? I don't see them in the UI anywhere andI was hoping they would use the External Tank % value to calculate remaining run time.
5) Can you "*" out the cleartext Password in the Tankutility External Tank Fuel Monitor "Add-Ons" setting?
6) "/var/log/gentankdata.log" and "/var/log/gentankutil.log" are always empty. Is there a log verbose level setting somewhere?
7) I don't see where the "favicon = http://www.generac.com/favicon.ico" is used.
8) Utility Power dashboard meter keeps bouncing back and forth between "0 kW" and "36280 kW"
9) Can you explain what the genexercise.py program does? My generator has a builtin exercise function configurable through the H-100 panel. I'm assuming this is an externally hosted program to exercise the generator when the isn't one local to the controller?
10) Is it possible to read/set the builtin H-100 exercise parameters?

Thanks again for developing this and let me know if you ever need my setup for testing.

alabamatoy, 2021-01-18:
Brand new Generac 22KW gnerator.  Appears to be evolution 2 controller, but Im no expert.  The controller has only 1 8-pin connector that looks anything like the one in the writeup.  It has 4 leads, red, white, blue and black, and all 4 go to a wifi module on the back of the unit, with no other connections to that unit.  The 8 pin is marked port1.  There is a port2 with a sticker over it, but under the sticker there is nothing, just a blank circuit board. I have attempted to attach a pic.  Is this "port1" which goes to the wifi puck the one to use with the genmon RPi?  Apologies if this is posted wrongly, please educate me.
[image]

skipfire, 2021-01-18:
Yes that is the right spot, the serial port uses 2 of the pins (for data, optionally a third for ground) that are currently empty there.  I just left my wifi disconnected since it sucks anyway, but you could probably re-pin so that you can use both the serial and wifi.

skipfire, 2021-01-18:
@alabamatoy yes that is the right spot, the serial port uses 2 of the pins (for data, optionally a third for ground) that are currently empty there.  I just left my wifi disconnected since it sucks anyway, but you could probably re-pin so that you can use both the serial and wifi.

liltux, 2021-01-18:
@alabamatoy yes that would be correct.  You can either disconnect the WiFi module and connect the RPi to the board only, or you can just wire in your RPi to pins 7 & 8 for RS232 and use the battery ground for your Pi so your RS232 signals are correct.

What @skipfire said : ) 👍

jgyates, 2021-01-18:
Here is the pinout for the connector. For future questions please create a new thread from the issues page.

Thanks
Jason

aric8456, 2022-07-26:
I have a generac 14kW (exact model isn't at issue here) with 2x 100g tanks monitored via TankUtility. It has been working great until the last month or so when I keep getting low propane alerts from genmon despite having over 70% fuel remaining.

[image]
[image]
[image]

jgyates, 2022-07-26:
You need to create a new thread for a new issue.

mikerenna, 2022-09-19:
i know this is a very late post on this thread but i've been combing through threads for answers before I ask any questions. This thread makes sense to me as it matches my generac but i'm wondering about the difference in pin out. Elsewhere here it expects evolution two to use pins 3456 i think, but this looks like they are using 1256 (or it could be 3478 since I have no idea which direction the pin out diagrams are supposed to be looking). What implications does this have in creating a repinned connector to use both genmon and their existing wifi? (for the record, i've been trying to implment gregmac's esp32 wifi streamer version here that uses a 5v lead in the connector to power the ESP32 but in either case, understanding this pinout is my current sticking point). Thank you in advance for any guidance. This looks like awesome stuff. -Mike

jgyates, 2022-09-19:
This thread is for general a general announcement regarding how to post.  For other topics that you do find a thread related you can create a new thread. 

The wiki has this info:

Main wiki page:
https://github.com/jgyates/genmon/wiki

Wiki entry on pinout and cable:
https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable

If you have any follow up please post to a new thread.

mikerenna, 2022-09-19:
I realized that almost immediately and deleted. I apologize. Asking my question in a new thread.  (I’ll remove this as well if it posts)

https://github.com/jgyates/genmon/wiki<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fjgyates%2Fgenmon%2Fwiki&data=05%7C01%7C%7C78f5e9186f114ef138fe08da9a7f057b%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637992167581445716%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=xqIPPc5QRdUWemE6vD0hwk4eMHZzt1KlrOrIb4NQi8M%3D&reserved=0>

https://github.com/jgyates/genmon/wiki/3.1--Making-a-Cable<https://nam12.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fjgyates%2Fgenmon%2Fwiki%2F3.1--Making-a-Cable&data=05%7C01%7C%7C78f5e9186f114ef138fe08da9a7f057b%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637992167581445716%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=vHgE2LiwLuIxBt7GYem35YeBU5RkxumbuxGI6fVnpX8%3D&reserved=0>

kohenkatz, 2022-10-02:
You should update the link in the post above for "Serial Troubleshooting" - the link that's there now (with "Appendix 6" in the URL) just redirects to the home page of the wiki with a message that "you don't have permission to edit this wiki".

I suppose the correct link is this one: https://github.com/jgyates/genmon/wiki/3.6---Serial-Troubleshooting

jgyates, 2022-10-02:
Thanks. I updated the above link