jgyates/genmon #17: Not really an issue...

dakegg, 2017-04-20:
Cannot find specifics on where this connects on the Evolution controller.  Any screen shots of the molex connection to the controller...  thanks

jgyates, 2017-04-20:
[Here](https://github.com/jgyates/genmon/blob/master/Diagrams/Cable%20(End%20that%20connecets%20to%20Evolution%20Controller)%20(2).jpg) is a picture of the cable that connects to the controller. I believe this one is an air cooled model. Some liquid cooled models the placement of the controller is may not be visible (like on my unit). I will try to puy some better descriptions of the cable in the wiki with pictures.

[This](https://github.com/jgyates/genmon/issues/15) thread has a picture of the controller receptacle 

Some other threads have question and some pictures of cables people have made.

There are two versions of the Evolution controller (liquid cooled and air cooled) so the placement may be different. I believe it is the only connector on the back of the unit with 8 pins, it also could have a sticker or cover over it.

madisdev, 2017-04-20:
On my Evolution controller it was under the panel that has the small LCD display. It also had a sticker over it that said something like port 1.

dakegg, 2017-04-20:
Thanks, I'll take a look.   Mine (11k aircooled) was manufactured in Nov 2016, so it's fairly new.  I saw a usb connection on the top, by the LCD and fuse, but wasn't able to see behind \ underneath without pulling the side panel...

madisdev, 2017-04-20:
I have a 22k and I can just lift up the front panel to gain access. Not sure if yours is similar. But it should be on the underside of the panel that has the USB and the LCD screen.

dakegg, 2017-04-20:
Now I wish I was at home, it didnt look like the panel lifted up, but I didn't think to try.  I figured I'd have to open the side panel where the battery is located and look up underneath from inside there...   I'll report back tonight when I've had a chance to check.  Thanks!

jgyates, 2017-04-20:
I added this page to the wiki
https://github.com/jgyates/genmon/wiki/Making-a-Cable

The part numbers for the connector are here:
https://github.com/jgyates/genmon/wiki/Hardware

dakegg, 2017-04-20:
Ok, so on the 11Kw Generac Guardian air-cooled, if you lift the lid and remove the front panel, you can look up underneath the control panel on the right and see the bottom of the controller.  There are two ports, each covered with a sticker.  One say Accessory Port 1, the other Accessory Port 2.  

I am going to build the solution and then try both, I'll report back in a week or so...   It's a brand new activation, and it's throwing a lot of white smoke when it runs, so I have to dig into that issue first.

jgyates, 2017-04-21:
Sounds good. Let me know if you have any questions. The smoke could be several things, one common thing  could be adding too much oil If you check your oil level and it is over the fill line try removing some oil.

rhombixt, 2017-04-23:
Your molex pinout is wrong in your diagram. The pin 1 is on the lower right . The pin 7 and 8 are correct

jgyates, 2017-04-23:
Thanks, I will try to get that fixed, one of the users created that. I don't have the original.

rhombixt, 2017-04-23:
I sent you an email with the repair manual molex details

jgyates, 2017-04-23:
I updated the wiki with the pinout diagram you sent. Hopefully we will get the other diagram fixed shortly.

rhombixt, 2017-04-23:
Great thanks.

jgyates, 2017-04-23:
The diagram with the error has been updated. Thanks for finding that rhombixt. Good catch!

closed 2017-06-05 by jgyates

rhombixt, 2017-06-06:
I got my generator installed... just waiting on parts still from china. They have been held up with canada customs now for almost 2 months.

Closed #17<https://github.com/jgyates/genmon/issues/17>.