jgyates/genmon #18: Question: Push notification?

krisasmith, 2017-04-22:
Have you considered adding push notifications or text messages in addition to or instead of email?

jgyates, 2017-04-22:
Hi,

I would like to add push notifications via google cloud messages api, although at this point I not started integrating this yet. 

SMS appears to be easy thing to add however it requires an account for some texting service like TxtMagic or Twillo. Google Voice may be an option as you can send txt messages from your google voice account.  Google Voice free so this may be the way to go. 

It this point I have considered both but implementing either of these is not something I am actively working on. Possibly in the future. Certainly if someone wants to add this to the project I would support this as well.

Let me know if you have any other questions.

Thanks
Jason

jgyates, 2017-04-25:
SMS support is not in the repository. See wiki page:

https://github.com/jgyates/genmon/wiki/Software#gensmspy-optional

It does require an account at twilio.com

jgyates, 2017-04-25:
sorry,  *now* in the repository:

https://github.com/jgyates/genmon/wiki/Software#gensmspy-optional

krisasmith, 2017-04-25:
will this load automatically when genmon starts or does it need to be added to the crontab?

jgyates, 2017-04-25:
No, it will not load automatically. It will need to be added to crontab.

I the latest version of startgenmon.sh, if this script is called from your crontab, all you should have to do is uncomment the line with gensms.py. This assumes you are running gensms from the same system as genmon. From your crontab, execute the full path of the script:

/home/pi/genmon/startgenmon.sh start

krisasmith, 2017-04-25:
i'm getting the error. (i've edited the values for privacy)

Error: File contains parsing errors: /etc/gensms.conf
        [line  2]: ' # Twilio account SID. This can be obtained from a valid Twilio\n'
        [line  3]: ' # account\n'
        [line  4]: ' accountsid = xxx\n'
        [line  6]: ' # Twilio authentication token. This can be obtained from a valid\n'
        [line  7]: ' #  Twilio account\n'
        [line  8]: ' authtoken = xxx\n'
        [line 10]: ' # Mobile number to send SMS message to. This can be any mobile\n'
        [line 11]: ' #  number.\n'
        [line 12]: ' to_number = +xxx\n'
        [line 14]: ' # Number to send SMS message from. This should be a twilio phone\n'
        [line 15]: ' #  number\n'
        [line 16]: ' from_number = +xxx\n'

jgyates, 2017-04-25:
Ok. try the version of gensms.conf (or edit the version you have locally. Each line had an extra space. I added spaces at the front of each line for formatting reasons when I was adding language to the wiki. I took almost all of them out when I checked in the gensms.conf. Sorry about that. :)

The version checked in the the repository should fix this.

krisasmith, 2017-04-26:
Works great. My genset did it's weekly exercise today and I got a text message right away.

jgyates, 2017-04-26:
outstanding. Thanks for the feedback. I will close this issue. I still have push notifications on my wish list, just need to get the android piece up and running, that may take some time to get right.

closed 2017-04-26 by jgyates