jgyates/genmon #20: Alarm Code Breakout

rhombixt, 2017-04-23:
Is it possible to add more outputs for specific alarms:

ie) gpio x goes active high when the voltage is out of tolerance
or gpio y goes active high when High Temp Alarm
etc etc

jgyates, 2017-04-23:
sure. I would just need some input on what you are looking for in terms of what alarms or conditions you would like to see. There are more alarms than GPIO and some alarms are specific to propane/lp vs diesel vs gasoline engines. Also, unless the example above was a specific error code we would want to nail down which voltage (generator or utility) and what the tolerance would be.

rhombixt, 2017-04-23:
Overvoltage or Undervoltage of generator output (both trip the same GPIO)
Low Oil
High Temp
Overcrank
Overload
Overspeed/Underspeed
RPM Sense Loss

jgyates, 2017-04-23:
on an evolution, yes this is very do-able. on a Nexus, not without a lot more data from users. I will add this to the list. Right now the main focus is getting Evo Air cooled working fully.

rhombixt, 2017-04-23:
I do have an Evo controller. What if I were to propose a different way of handling things. I noticed that you have an alarm.txt list. My HAI controller has serial ports on it that accept triggers. How hard would it be to add a USB serial port to the Pi and then everytime there is an alarm from the list that get detected, the Pi would send the serial trigger to my panel:
ie) alarm code 1001 is received.
The second serial port would send ascii HAI1 out . I would program the controller on my end to display the message Internal Harness Loop Broken 

This would be actually way easier for me. Even if you just compile an example, I can input the codes for all the triggers HAI1 to HAI128

jgyates, 2017-04-23:
Sending out a code for each alarm is pretty trivial, even more if I just send out the alarm code prefixed by HAI. I can read the current alarm code from existing commands. Another option would be to take the ALARMS.txt file and if you add the code you want at the end of each line "!HAI128", etc. (Rename the file) I can create a small program that will read if the system is in alarm, get the alarm code, find the alarm code in the new file, and send out the command you added to the file over USB serial.

rhombixt, 2017-04-23:
The code sent to the HAI is actually a trigger - you can't sent additional text to the serial port or it will get confused. They engineered the main panel for applications like this to make programming easier. On my end , I program the panel to take the trigger and do something. I would need the PI to simply transpose one for the other. Let me work on the alarm file and ill Add it here.

rhombixt, 2017-04-23:
I created alarmsHAI.txt. Each alarm has an HAI code associated with it - HAI1 to HAI44

jgyates, 2017-04-23:
Hi,

Can you email me the file or attach it to this thread. I am not quite ready to merge this into the main tree as it may cause questions and I would like to do a bit more research before I put it in the project as something I attempt to support. 

I envision this coming together once we get your base setup validated and working. 

I looked at the commit request. Would it be possible to add the HAI commands at the end of the line instead of the middle. If this does get merged with ALARMS.txt it would make integration much easier. Just add "!HAIcode" at the end of each line.
Thanks
Jason

rhombixt, 2017-04-24:
At this point, l think the best way is to stick with GPIO. I will apply the KISS rule .. less work for both of us.  If I need to look at specifics, then I can always check the genset. If you can add the following to the gengpio (keep what is there now of course adding the following)
Overspeed/Underspeed (alarms 1200-1206, 1600-1603) GPIO 9
Low Oil (alarm 1300) GPIO 10
High Temp (alarm 1400) GPIO 11
RPM Sensor (alarm 1500-1521) GPIO 12
Overvoltage/Undervoltage (alarm1800-1803, 1900-1906) GPIO 13
Overcrank (alarm 1100-1101) GPIO 14
Overload (alarm 2100-2103) GPIO 15
Governer (alarm 2500-2502) GPIO 16

jgyates, 2017-04-24:
Roger that.

jgyates, 2017-04-24:
Did a quick check. We will need to adjust the GPIO as they conflict with the serial and SPI. Lets go with the GPIO

READY GPIO 23 (pin 16)
ALARM GPIO 24 (pin 18)
SERVICE DUE GPIO 25 (pin 22)
Overspeed/Underspeed (alarms 1200-1206, 1600-1603) GPIO 5 (pin 29)
Low Oil (alarm 1300) GPIO 6 (pin 31)
High Temp (alarm 1400) GPIO 13 (pin 33
RPM Sensor (alarm 1500-1521) GPIO 19 (pin 35)
Overvoltage/Undervoltage (alarm 1800-1803, 1900-1906) GPIO 26 (pin 37)
Overcrank (alarm 1100-1101) GPIO 21 (pin 40)
Overload (alarm 2100-2103) GPIO 20 (pin 38)
Governor (alarm 2500-2502) GPIO 16 (pin 36)

rhombixt, 2017-04-24:
Is it also possible to break out the exercising to a gpio and the running to gpio as well.. vs grouping it with the ready gpio? How do the states clear? I assume pressing buttons on the controller does the trick.
On the gpio in  part, there is an input to thro the transfer switch, does the transfer thro back on the abscence of an active high on that pin or will your need a separate input?

jgyates, 2017-04-24:
Yes, it is possible.  Genmon is just reflecting the state of the generator and the gpio are reflecting the state of genmon. All conditions are cleared at the generator, once cleared at the generator genmon and gpio will reflect the cleared state.

Regarding GPIO in, on a falling transition the command is sent to the controller. So if you issue the starttranfer command, when you press the button it will start the generator and activate the transfer switch. It will stay running/activated until it is stopped by pressing the stop button.

rhombixt, 2017-04-24:
Ok - so the GPIO IN for Start - this should just be the start generator - not starttransfer ( I just want to run the genset for maintenance remotely - but not transfer the loads)

GPIO IN for Stop - this should just be the stop generator

GPIO for transfer - this should transfer the load to the genset

can you create another GPIO in For transfer back to utility?

jgyates, 2017-04-24:
This is how the genmongpioin works now:
https://github.com/jgyates/genmon/wiki/Software#gengpioinpy-optional

there are no other generator commands to implement (i.e. there is stop transfer but keep the engine running). The controller manages that. In fact after your issue a stop after a starttransfer has been issued the generator will deactivate the transfer switch but the engine will continue to run for about 1 or 2 min in cool down mod. The controller handles all of this. 

 If you want to transfer power back to the generator activate the stop gpio.

I think you will have a better feel for what the program can do once you get your setup work. On the GPIO in the is really nothing more I can do in terms of functionality. On the GPIO out I can make the change to split the exercise, run, and ready to separate GPIO pins.

rhombixt, 2017-04-24:
Sounds good. I have my HAI prepped for programming with GPIO's broken out - along with the list of 8 basic alarms.

I am still waiting for parts of course. The Genset will be installed hopefully next month when its a bit warmer then I can test.

https://github.com/jgyates/genmon/wiki/Software#gengpioinpy-optional

[https://avatars3.githubusercontent.com/u/4513286?v=3&s=400]<https://github.com/jgyates/genmon/wiki/Software#gengpioinpy-optional>

jgyates/genmon<https://github.com/jgyates/genmon/wiki/Software#gengpioinpy-optional>
github.com
genmon - Generac Generator Monitoring using a Raspberry Pi and WiFi

jgyates, 2017-04-25:
gengpio.py has been updated:

https://github.com/jgyates/genmon/wiki/Software#gengpiopy-optional

rhombixt, 2017-04-25:
Sweet. If I do a git clone .. is that the easiest way to handle the deltas on the files?

https://github.com/jgyates/genmon/wiki/Software#gengpiopy-optional

jgyates, 2017-04-25:
No a clone will likely give you errors is you try to merge. If you remove your genmon directory and then clone you will get a fresh copy. If you want to down load deltas only then "git pull origin master" from the genmon directory. There are a lot of resources for using git on the web.

jgyates, 2017-04-26:
just looked this up, "git pull" will update a previously cloned checkout

jgyates, 2017-04-30:
I updated gengpio.py to add a GPIO for the condition when the generator is switched off.

See wiki for pinout.
https://github.com/jgyates/genmon/wiki/Software#gengpiopy-optional

rhombixt, 2017-04-30:
Nice.. that's helps me to make things work better.

https://github.com/jgyates/genmon/wiki/Software#gengpiopy-optional

[https://avatars3.githubusercontent.com/u/4513286?v=3&s=400]<https://github.com/jgyates/genmon/wiki/Software#gengpiopy-optional>

rhombixt, 2017-04-30:
the one item that is missing - is when the transfer switch is thrown. Not sure if you can decode that from the genset however.

https://github.com/jgyates/genmon/wiki/Software#gengpiopy-optional

[https://avatars3.githubusercontent.com/u/4513286?v=3&s=400]<https://github.com/jgyates/genmon/wiki/Software#gengpiopy-optional>

jgyates, 2017-04-30:
The ability to read the state of the transfer switch relay via modbus is only present on liquid cooled evolution models. I have searched for this on both Nexus air cooled and Evolution Air Cooled units. Presently I am out of ideas on how to implement on these models.  We are deep in the uncharted / undocumented territory when doing things that are not even supported by generac products like Mobile Link.  In short, this may not be feasible unless Generac exposes this info via modbus in their firmware for the generator.

rhombixt, 2017-04-30:
I used a secondary switch (contact closure) that's comes as a kit that I install on the transfer switch to send notification.  I think this product is VASTLY superior to any product they would ever sell. Ironically, the Kohler has a PIM module you can buy and program for any array of outputs such as the one that you have designed. Simple to install, program and you have what you need but it's a kohler and when they break, its costly to fix.

jgyates, 2017-10-28:
Hi @rhombixt, I am closing this issue. If you have any outstanding items on this feel free to reopen the issue.

closed 2017-10-28 by jgyates