jgyates/genmon #22: Not so much an issue but have some Wi-Fi Interface questions

weffz, 2017-05-06:
Have been trying to identify the Wi-Fi adapter used and believe this is also from Amazon and appears to be from Detroit Electronics as per https://www.amazon.com/Wifi-Antenna-Raspberry-Pi-Instructions/dp/B00H95C0A2/ref=sr_1_1?ie=UTF8&qid=1494050717&sr=8-1&keywords=B00H95C0A2 but so far not been able to identify the brand/model of antenna used - and the through case fitting used to the outside. Is that a lightning arrester as a feed through on the case to the vertical antenna ? 

I have also yet to have my Generac 13 KVA 240V 50Hz #006271-0 system installed - have engaged installers for gas and electricity last week - will possibly be installed within the next few weeks.  Had given up on having mobilelink anytime soon in Australia - then came across genmon last night - looks exactly what I was after.  Very impressed. 

Have also already separately ordered the #6664 Wireless local monitor - which at least should also provide the correct 8-pin plug to connect into the controller and should help further trace the rest of the pins on that port.  The manual for #6664 and the Mobilelink #6463 Manual are both  quite useful too in identifying  which port is used and where it is located on each version of the controllers as they have evolved between 2008 and 2013.
To ensure the serial port is tied directly to signal ground I also plan on having a 3-wire connection from the controller to the RS232 side of the TTL to RS232 converter. That will also allow a 9-pin breakout box to more easily intercept serial traffic if a proper 9-pin plug/socket is there to intercept. Have considered having an optical isolator in series with the controller output - as added protection but is likely to be redundant as any lightning strike that close would probably destroy generator not just stop at a lightning arrester.

I had also assumed that #6664 merely broke out the led wiring and common to pass on the Red Green Yellow led's via Wi-Fi into the house so it will be interesting to determine how the detection of the alarms is done if not via slaving the RYG leds across Wi-Fi to the 3 coloured LED's in-house and that may also help identify further of the controller port pins too if they aren't using serial comm's to detect message types

Hopefully by the time the generator is commissioned I will have a working genmon system ready to go. Not altogether sold on using Wi-Fi for comms yet - some of the antennas I priced on Amazon cost more than the entirety of the other parts at USD$85-120 whereas an isolated RJ45 connection to nearest switch indoors might be more cost-effective.

krisasmith, 2017-05-06:
I gave up on wifi, I had a terrible time with it. Instead I mounted my raspberry pi in my garage next to the transfer switch. From there I had access to a wired network connection. Then I ran shielded 4 wire from the raspberry pi into the transfer switch then through the conduit with the rest of the wires to my generator. Using the power from the accessory port on the Nexus controller (pin 1 = 5v, pin 2 = ground) I can power the raspberry pi through the GPIO. https://goo.gl/photos/WLUeonGe5JUeDXKz5

jgyates, 2017-05-06:
Yes. That is the wifi adapter I am using. 

My situation is probably a bit unique. My generator is about 60 feet from my home so I opted for wifi.  
The white tubing is just a PVC end cap. My antenna is not mental, it is fiberglass. I once had Mobile Link attached and it left a hole in my generator case so I used that hole to route the antenna. Mobile link uses a threaded PVC to attach the ML unit to the generator case, so I just used the same PVC pipe width and thread, then drilled a hole in the PVC to route the antenna wire.

This company has great wifi antenna prices:

https://www.data-alliance.net/weatherproof-antennas/

This is the one I am using:
https://www.data-alliance.net/antenna-2-4ghz-10dbi-omnidirectional-wifi-mounts-pole-wall-n-f-waterproof/

I prefer ethernet over wireless, but in my case wifi was the simpler solution for me, So far wifi is working well for me. 

I will try to post a close up of the antenna PVC later today.

weffz, 2017-05-06:
Thanks - that is certainly a better priced antenna - it looks like fibreglass [& definitely not "mental"  though all the joins had me wondering! ] - and I expect your Wi-Fi has evolved over time given the number of joins in the external cabling - but if you've managed to bridge 60 feet with Wi-Fi that is a great achievement as it is. 
I read the install process for mobile-link - drilling a hole where the dimple is in some models and adding seals to prevent water ingress - plus the 6  wire molex connector poke through for the mobilelink radio module to plug in.
If you still have the original mobile-link cable I would be interested in seeing which wires they used as apart from the SIM card for Verizon and the CDMA radio circuitry there must be some other smarts in there that are used to connect through to the mobile-link web-address. If only 6 wire come through plus the separate power leads it would be useful to identify which cont
ike fibreglass [& definitely not "mental"  though all the joins had me wondering! ] - but I expect your Wi-Fi has evolved over time given the number of joins in the external cabling - and if you've managed to bridge 60 feet with Wi-Fi with reliable comms that is a great achievement as it is. 
I read the install process for mobile-link - drilling a hole where the dimple is in some models and adding seals to prevent water ingress - plus the 6  wire molex connector poke through for the mobilelink radio module to plug in.
If you still have the original mobile-link cable I would be interested in seeing which wires they used as apart from the SIM card for Verizon and the CDMA radio circuitry there must be some other smarts in there that are used to connect through to the mobile-link web-address. If only 6 wire come through plus the separate power leads it would be useful to identify which controller port leads of the 8 are needed for mobile-link's interface. With MODBUS being an inherently slow protocol perhaps they have implemented more of the RTS/CTS DCD/DTR type hand-shaking to prevent the controller from forwarding data using TX until it has received DCD high for when the Verizon radio is online.
Did you manage to identify how the mobile-link radio module + self-activation circuitry works ? I can't imagine you not opening up the mobile-link module although not that helpful as to work locally you would have needed to decode all the MODBUS protocol as well as the alarms - a massive effort.
Hopefully you were able to debug the interface while being inside in the warm and not beside the generator in the cold.

weffz, 2017-05-06:
@krisasmith  That is certainly a neat solution. With a 13KVA Generator I have a100A transfer switch as there should be sufficient capacity (54A) to run all the normal house power loads so the entire set of power load circuit breakers will be connected to the generator. I just have to make sure that I don't exceed the generator capacity - which I shouldn't do as it would be rare to be consuming 13KVA that often.
They certainly are massive power leads going to utility and generator in those photos.
What capacity is your generator ? - Is the power wiring all 200A rated ?The transfer switch must make a huge CLUNK as it switches once the generator has started up. is that a 22kW system perhaps ?
I like the idea of using the controller to power the Raspberry Pi - a very neat solution and saves messing with the 12V power wiring in the generator - plus the power usage wouldn't be that much either.
Hopefully the Evolution controller provides a +5V supply too - if not will have to borrow off the battery supply too.

krisasmith, 2017-05-06:
Yes I have a 22kw genset with a 100amp breaker. The wiring from the gen to the transfer switch is rated for 100amp. All the rest of the wiring is 200amp. When I load tested everything (2 ac units, electric dryer, stove, microwave, etc) it was only pulling about 55amps most of which was the dryer. When the transfer switch is thrown it is loud, you can hear it inside the house.

The transfer switch I have supports an AC lockout, it won't let it run if the generator is near capacity but I don't anticipate it being a problem. If the generator is on I can live without the dryer.

jgyates, 2017-05-06:
Sorry, I put my spell checker through it's paces. :)

The mobile link cable uses 5 of the 6 connections. 2 for power and ground to the battery, 2 for TX and RX to the controller and one pin is connected to the cable shielding enclosing the TX and RX path. The ML and the remote monitor both appear to read the modbus serial link. The remote monitor just uses what I am guessing is an ISM band RF signal to relay the data.

When reversing the serial protocol, after noticing only two wires were connected from ML to the controller, I set up a pi with two serial ports and captured both  lines and time stamped the data. Once I examined it I looked for patterns and compared them to the modbus protocol. My first revision ran alongside ML, just decoding the info that I could figure out based on what ML was displaying. I started mapping modbus registers to functionality. Once I got enough of it decoded I started transmitting and removed the second serial port and took ML out.

The ML looks like it is using a packaged module for cellular comms (large square metal package in the photo). 
 
I am attaching photos of the inside of the mobile link and my antenna connection.
[image]
[image]
[image]

closed 2017-05-07 by weffz