jgyates/genmon #23: No web interface

sgtmaj27, 2017-05-27:
Generac 5870, Air Cooled, Nexus Controller

Thank you for a great appl;ication.  I have everything working which I can verify through the ClientInterface.py.  I can not seem, however, to get anything other than the template from the web interface.  I installed Flash and have the exact directory structure required and I do have web access to the pi.
This is all I can get:

[image]

I am not that familiar with Flask any direction would be appreciated:

Foster

sgtmaj27, 2017-05-27:
Here are the results from three different computers and browser on my network.  Please notice the directory structure as well
[image]

closed 2017-05-27 by sgtmaj27

jgyates, 2017-05-27:
Hi Foster,

From the display output in your second picture it looks like everything is working. This is the normal output. For debugging purposes let's do the following:

Click "Monitor" on the web interface, then look at the terminal output, you should see a line containing "GET /cmd/monitor HTTP/1.1." in the output.

Also, are you seeing the same results when trying this on separate computers? What is the browser type (Firefox, chrome, etc) and OS that you are using on each computer?

One additional suggestion, since it looks like the server side is responding, I would attempt to disable any browser plugins and see if the results change.

Also, Flash is not needed.

Thanks
Jason

sgtmaj27, 2017-05-27:
Sorry, I meant I installed Flask not flash.
I have the same results from multiple computers, IOS devices and Chrome, Firefox, and Edge.

When I click Monitor this is the results:
192.168.252.103 - - [27/May/2017 09:27:21] "GET /cmd/monitor HTTP/1.1" 500 -
192.168.252.103 - - [27/May/2017 09:27:21] "GET /cmd/monitor HTTP/1.1" 500 -
192.168.252.103 - - [27/May/2017 09:27:23] "GET /cmd/getbase HTTP/1.1" 500 -
192.168.252.103 - - [27/May/2017 09:27:26] "GET /cmd/getbase HTTP/1.1" 500 -
192.168.252.103 - - [27/May/2017 09:27:26] "GET /cmd/monitor HTTP/1.1" 500 -
192.168.252.103 - - [27/May/2017 09:27:29] "GET /cmd/getbase HTTP/1.1" 500 -
192.168.252.103 - - [27/May/2017 09:27:31] "GET /cmd/monitor HTTP/1.1" 500 -
192.168.252.103 - - [27/May/2017 09:27:32] "GET /cmd/getbase HTTP/1.1" 500 -
Thank you for your help

sgtmaj27, 2017-05-27:
Jason:

I am using Windows 10 on my PCs, IOS 10.2 on my devices.  I have disabled browsers and my firewalls no change.  I have also tried a remote desktop to my pi and used Chromium to both the ip address and 127.0.0.1 and the same results; the blank web interface.

thanks

jgyates, 2017-05-27:
Are you starting genserv.py with the startgenmon.sh script?
Does the file /var/log/genserv.log contain any error messages?

After looking closer at the output in your previous message I noticed a problem. The error code for each line is listed at the end of each line, in your case 500 is the code. It should be 200. 

In your previous post you showed the directory listing of your pi. Can you show the contents of the static and template directory? If you open a terminal and type "cd /home/pi/genmon/static" the type "ls -la" this will display the contents of the static directory and the file permissions of each file. This info would be helpful.

sgtmaj27, 2017-05-27:
I am using the startgenmon.sh script to start.  I have also tried to individually start genmon and genserv.

Genserv.log is empty.

/home/pi/genmon/static$ ls -la
total 136
drwxr-xr-x 2 pi pi  4096 May 24 11:00 .
drwxr-xr-x 7 pi pi  4096 May 24 15:52 ..
-rw-r--r-- 1 pi pi  1150 May 24 11:00 favicon.ico
-rw-r--r-- 1 pi pi  3775 May 24 11:00 genmon.css
-rw-r--r-- 1 pi pi 18760 May 24 11:00 genmon.js
-rw-r--r-- 1 pi pi   647 May 24 11:00 index.html
-rw-r--r-- 1 pi pi 97163 May 24 11:00 jquery-1.12.4.min.js

/home/pi/genmon/template$ ls -la
total 12
drwxr-xr-x 2 pi pi 4096 May 24 11:00 .
drwxr-xr-x 7 pi pi 4096 May 24 15:52 ..
-rw-r--r-- 1 pi pi  100 May 24 11:00 command_template.html

jgyates, 2017-05-27:
I couple of other things:

in your browser enter the following address:
http://192.168.252.113:8000/cmd/getbase

What do you see as an output in the browser?

Is your pi using a static address or DHCP?

I did find one problem in the code, but I doubt this is the problem. Flask requires that the template director be named "templates" not "template". I will change this in the repository, but based on the output this is not your problem. This only effects the program if genserv.py receives unknown commands.

jgyates, 2017-05-27:
you can rename the template directory by typing the following:

mv /home/pi/genmon/template /home/pi/genmon/templates

sgtmaj27, 2017-05-27:
http://192.168.252.113:8000/cmd/getbase results:

Internal Server Error
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

I verified in my dhcpcd.conf and my router that the ipaddress is static.

Renamed template dir to templates.

jgyates, 2017-05-27:
Just for a data point, if you use this address:

http://192.168.252.113:8000/xxx

Do you get "Not Found . The requested URL was not found on the server. If you entered the URL manually please check your spelling and try again." ?

Also, from the terminal window, type the following command:

pip freeze | grep Flask

what is the output of this command?

jgyates, 2017-05-27:
you may of tried this already, but try restarting the flask app.

./startgenmon.sh stop
./startgenmon.sh start

I have seen rare situations where flask gets in a strange state.

sgtmaj27, 2017-05-27:
Anything after http://192.168.252.113:8000/ results in the "Not Found etc"

Result from pip freeze | grep Flask

Flask==0.10.1

I have tried stop/ restart the script and individual files as well

jgyates, 2017-05-27:
hmm, that is a dated version of Flask. Let's try updating to the latest. First stop the app with startgenmon.sh. To update flask to the latest version type:

sudo pip install Flask -U

After the update, "pip freeze | grep Flask" should show:
Flask==0.12.2

sgtmaj27, 2017-05-27:
That was it!!! Wrong version of Flask......thank you.

I will send unknown error codes soon to help your data points!

I appreciate all your help

Foster

jgyates, 2017-05-27:
Awsome! Thanks for the feedback. I have never had this issue before so it is always good to find unknow problems.