jgyates/genmon #24: Logging

ts4s, 2017-05-28:
Went to check on generator today to see what events it had, however there were no file for this.

To keep from add to the length thread, I start a separate one for this topic. 

The ideal to log event lead to the following (made a path correction to the original post, it was note a few msg later):

"sudo python /home/pi/genmon/genlog.py -o /home/pi/genmon/Filename.csv &

If you are loading it from crontab you need the full path of the log file. It will work without the path if you call the script your self. Also you need the & if you want it running in the background. pkill will kill the background app.

It will only log to the file if you generator changes state (start, stop, exercise, alarm, or service due warning)"

I went looking for the file and can't find it. So I'm not sure what happen here. I stopped and started the start genmon.sh and manually start the generator to cause an event. Refresh my ls and still no file?

jgyates, 2017-05-28:
I added the & to startgenmon.sh on all loaded modules, including ones that are commented out by default. You are correct that the full path is needed if loaded via crontab. The same thing applies to paths listed in any conf files (full path is always needed if you load via crontab).

ts4s, I could not tell from your post, are you still having issues now that you have the full path listed?

ts4s, 2017-05-28:
[image]

here's a photo of my startgenmon.sh

i haven't change anything, yet. This is still my pi3 from the original thread. The pi zero w is still in the works, (needs a case for it and the serial board to keep it clean in my generator).

jgyates, 2017-05-28:
That looks correct. Let me know if you have any issues. I assume you are saying that you have not connected this to your generator yet so I am guessing you don't have any issue with genlog.py.

ts4s, 2017-05-28:
no, it's been connected. it's not logging any start, stop, or exercise.

jgyates, 2017-05-28:
ok. After looking closer at your startgenmon.sh file I am guessing the problem is the -o. It should be -f instead of -o. This was a typo that was fixed from the initial checkin (command line help printed -o but the actual command line parser code looked for -f). Let me know if that does not fix the problem. I just did a fresh checkout of the repository and tested and it is working for me. If it is some other issue I am sure we can track it down.

ts4s, 2017-05-29:
That fixed it... 👏

I see the fix is in the new version, dropped in on the pi... good to go for the moment. 

Bummed cuz I didn't get the data for the last month, storm season is here 🌪⛈

ts4s, 2017-05-30:
ok, not bummed... found this today after checking for log:

[image]

the csv log is present... it has 3 entry's since yesterday :-)

closed 2017-05-30 by ts4s