jgyates/genmon #25: Direct Serial to Molex (and where to find Molex parts?)

derekatkins, 2017-06-10:
Hi,
I've got a Wandboard sitting around unused.  The wandboard has a built-in serial device with a DB9 already, so I was thinking that I would just use that directly to connect to my Generac with an Evolution controller.  Is this possible?  What code changes would be required?
Also, I'm trying to find the Molex part, but I cannot find WM3703-DN or WM3279CT-DN on digikey.  I can only find the -ND parts (which you say on https://github.com/jgyates/genmon are not the parts I need).  However looking at digikey's website for the WM3703-ND (e.g. https://www.digikey.com/product-detail/en/molex-llc/39-01-2080/WM3703-ND/61382 ) does look like it will fit, based on the picture on the end of https://github.com/jgyates/genmon/wiki/Making-a-Cable
So what part, exactly, do I need?
Thanks.

jgyates, 2017-06-10:
Hi,

I have never used a wandboard but doing a quick search it looks like that should work with no problem. The software (unmodified) requires a linux distribution capable of running python, a serial port and a network adapter (wifi or Ethernet). I see that the Wandboard support both Ubuntu and Debian. I use a debian distribution on the raspberry pi so that would be most like what have been used to date but I see no reason why Ubuntu will not work.

I went back and checked the part numbers. It looks like -ND is what is needed. Apparently I transposed the ND to DN in the docs. Sorry about that. My last digi-key order was for WM3703-ND and WM3603-ND. Thanks for the question, I will update the docs to reflect this change.

jgyates, 2017-06-10:
FYI, this page

https://github.com/jgyates/genmon/wiki/Hardware

Has links to the correct digi-key page for the parts.

jgyates, 2017-06-10:
I updated the wiki and readme. Let me know if you see anything else that does not make sense.

derekatkins, 2017-06-10:
Thanks for updating and adding the links.
The Wandboard also runs Fedora (my distro of choice).  But it should meet all the dependencies.
Just to be sure -- the generator provides RS232 through the molex?  So I dont need a level converter?
I also read that it will supply +5V?  (I'm trying to figure out how best to power the WB)
I'll read through your changes more thoroughly tomorrow.
Thanks

jgyates, 2017-06-10:
Yes, RS-232 through the molex. Fedora should not be a problem. The level converter listed in the project is to bring the Raspberry PI TTL serial port to RS-232 levels. I have tested the serial port on the controller with both the TTL to RS-232 converter and a USB RS-232 serial port.

Regarding power, I supply 12V and GND from the battery and use an automotive to USB power supply to power the raspberry pi. I know another person who has used the software gets 5V and GND from the connector as shown on this page: https://github.com/jgyates/genmon/wiki/Making-a-Cable. They connect the 5v and GND to the GPIO power and ground on the pi.

Both should methods work as long as you have a common ground. Both the 5v from the connector and battery 12V are constant through a utility power loss.

derekatkins, 2017-06-10:
Great.  Parts all ordered.  I just need to figure out how I'm going to run my ethernet cable into the generator cabinet ;)
Can't wait to try this -- I want notifications of when my generator is running (so I have a record of my actual power outages)

krisasmith, 2017-06-10:
I mount my Pi in my garage next to the transfer switch and then ran 4 conductor shilded 22 awg wire (home Depot sells it as alarm wire) through the same wire conduit that holds all the rest of the wires from the transfer switch to the genset. Power the pi from the controller as @jgyates described above. Because my pi is in the garage I can easily run Ethernet to it.

krisasmith, 2017-06-10:
Here is a couple of pictures of my setup. The black and the green wires on the GPIO are the 5v and ground from the evolution controller.

https://goo.gl/photos/D6ew4PiNr4Yo9hmEA

derekatkins, 2017-06-10:
@krisasmith -- I've got an ethernet port outside near the generator, so all I need to do is run a cable from that port to the generator so I can install the wandboard inside.  So the only question is how to get the ethernet cable into the generator housing.

krisasmith, 2017-06-10:
Send it through where the gas line goes in. Make sure you use an outside rated cable for UV protection.

derekatkins, 2017-06-10:
I'll try there.  That looks a little tight but maybe I can squeeze the cable through.
I found the port and know where I can mount the wandboard.

derekatkins, 2017-06-13:
One last question.  The wire link points to 16AWG Primary Wire...  But the crimper tool (and indeed the molex pieces) seem to imply a limit of 18AWG?  So did the wire change?  Or did you actually crimp 16AWG wire into your cable?  I really only need about 12-18" to get from the molex receptacle to where I plan to mount the box.
How warm does it get on the battery side of the generator, anyways?  Could I use 60C wire?  I suspect I'd be safer with 100C wire.

jgyates, 2017-06-13:
16AWG is the best for the molex connector. I will update that link on the hardware page to specify 16AWG.

I will let someone else answer this question regarding heat as my setup may not be a direct comparison, I have a 48KW Diesel generator. As a data point I have not seen any heat issues, but my gen is liquid cooled.

jgyates, 2017-06-13:
FYI, 20awg will work as well. I used 20 on my last cable. The first one I was still learning how to use the crimp tool correctly.

derekatkins, 2017-06-15:
I just built my cable.  The power connector was also 18AWG.  The crimper worked just fine with 18.  Meter shows connectivity at all points.  Next:  Connect it and see if it works.

jgyates, 2017-06-15:
Let me know how it goes. I expect the Wonderboard will work fine but it is nice to have some validation. If it works I will add some text to the docs about using other platforms.

derekatkins, 2017-06-15:
It seems to be working great.  I do have a few unknown error codes but I'll post over on issues 12 and 13 about them.  The system is happily sipping power from the 5V in the molex and talking happily with the controller!  THANK YOU!  Now to wait for the next power outage and see if I get a notification.  :)

jgyates, 2017-07-13:
Closing this issue. Feel free to reopen if you have any other questions.

closed 2017-07-13 by jgyates