jgyates/genmon #26: mymail claims no internet connection?

derekatkins, 2017-06-14:
I want to configure mymail to send mail but not IMAP (I dont care about that service).
So I removed the imap server and password from the config file.
When I start genmon I see this in the mymail.log:

2017-06-14 11:41:42,213 : No Internet Connection! 
2017-06-14 11:41:44,287 : Error in SendMailThread, retrying: STARTTLS extension not supported by server.

Is this because imap_server is set to an empty string?
Is there some wya to turn off IMAP without turning off SMTP?
I'll note that the smtp server *does* support STARTTLS!  So I'm not sure why I'm getting this error.

derekatkins, 2017-06-14:
For the record, apparently my postfix server was misconfigured and STARTTLS *was* turned off..  It had smtp_tls_... instead of smtpd_tls...  So now it doesn't complain about lack of TLS, but it is complaining because it's trying to authenticate when no authentication is required:

2017-06-14 13:58:48,295 : Error in SendMailThread, retrying: (535, '5.7.8 Error: authentication failed: authentication failure')

This was fixed by a simple change to mymail.py:

```
         session.ehlo
-        session.login(self.EmailAccount, self.EmailPassword)
+        if self.EmailPassword != "":
+                session.login(self.EmailAccount, self.EmailPassword)
         session.sendmail(self.EmailAccount, recipient, msg.as_string())

```

I'll look into disabling IMAP in the same way.

jgyates, 2017-06-14:
IMAP can easily be disabled by changing one line in genmon.py

Change this:
 self.mail = mymail.MyMail(monitor=True, incoming_folder = self.IncomingEmailFolder, processed_folder =self.ProcessedEmailFolder,incoming_callback = self.ProcessCommand)

to This:
 self.mail = mymail.MyMail(monitor=False, incoming_folder = self.IncomingEmailFolder, processed_folder =self.ProcessedEmailFolder,incoming_callback = self.ProcessCommand)

I just never formally added a .conf setting to disable. I use the mymail.py module for other projects and do not use IMAP for some of those.

jgyates, 2017-06-14:
I will add your mod to mymail.py

derekatkins, 2017-06-14:
Thanks.
I'll add that workaround in the meantime.  It's certainly easier than changing the monitor thread to "not run" if there is no password.  And that alone wouldn't be the "right" solution, either.

jgyates, 2017-06-14:
I updated mymail.py to allow selective disable of IMAP and SMTP.

Just leave the entry blank to disable, i.e. "imap_server =" or "smtp_server ="

jgyates, 2017-06-14:
FYI, the "No Internet Connection" message is the error code from the python SMTPLib. If an exception occurs due to the address not being resolved or any condition that would make the library think that there is a problem with the connection it will give this error. My mail is just reporting the error the library returns. I disabled my access point and changed the generator state and I can get the library to report this error also. I am assuming that this occurs if a DNS error in the server address, the library is assuming the address is valid and before the mod it was trying to resolve an address of an empty string. Unless you have any other issues I will close this thread.

Thanks for the feedback!
Jason

closed 2017-06-14 by jgyates

derekatkins, 2017-06-14:
No, it's reporting the error because in my case I have "imap_server" set to the empty string, so the connection fails.  I did this as an attempt to "turn off IMAP".

derekatkins, 2017-06-14:
But it looks like you fixed it to handle this case...  Thanks.

derekatkins, 2017-06-14:
Nope, looks like this didn't fix it..  (assuming your change was pushed):

python genmon.py 
Traceback (most recent call last):
  File "genmon.py", line 3209, in <module>
    MyGen = GeneratorDevice()
  File "genmon.py", line 384, in __init__
    self.mail = mymail.MyMail(monitor=True, incoming_folder = self.IncomingEmailFolder, processed_folder =self.ProcessedEmailFolder,incoming_callback = self.ProcessCommand)
  File "/root/genmon/mymail.py", line 82, in __init__
    self.FatalError("ERROR: incoming_callback, incoming_folder and processed_folder are required if receive IMAP is used")
  File "/root/genmon/mymail.py", line 255, in FatalError
    raise Exception(Message)
Exception: ERROR: incoming_callback, incoming_folder and processed_folder are required if receive IMAP is used

jgyates, 2017-06-14:
Sorry about that. I just checked in a fix. I tested this one by disabling IMAP instead of just testing that I did not break anything while leaving IMAP enabled.

derekatkins, 2017-06-15:
Verified fixed.  Thank you.