jgyates/genmon #29: Numerous False outages

sgtmaj27, 2017-06-30:
This is installed in a 5780 w/Nexus controller

Last night the software reported 30 outages lasting from 03 seconds to 1:59 minutes.  We never lost power at all...I have several UPSs that I checked logs on and no "flashing" leds to indicat intermittent outages.  The generator never attempted to start.  The max utility voltage was also recordd as 3190 volts!  This morning I restarted the pi and the genmon software.  
Should I be concerned about  a generator problem or could this just be a "glitch" in the software.

Thanks

jgyates, 2017-06-30:
That does sound odd if you are getting a reading of 3190 volts. Do you see any communication errors (CRC, discarded or timeouts), you can see them from the "monitor" command or on the monitor page in the web UI. Also, do you see any errors in genmon.log? Are you using the latest version of the software? A dump of your registers would also be helpful. Are you still seeing unusual voltage levels?

One limit of the software is that on a Nexus controller the software does not have the ability to read the threshold or pickup voltage so a default value of 143V and 190v respectively. The generator also have a timeout value (typically 5 seconds) that the software can not read via modbus. If the utility voltage is below the threshold voltage for at least the timeout value the generator attempts to start. The software will report an outage if the utility voltage is below the threshold voltage at any time, regardless of the timeout value, so it is possible that an outage will be reported even if the generator does not start due to the inability to read the timeout value.

Having said all of this, it does not sound like you are experiencing normal behavior. I have a UPS that sends emails when the UPS is activated and I get notifications at the same times from the UPS and genmon when the utility voltage is low. Also the utility voltage of 3190 volts should not be occurring.

sgtmaj27, 2017-06-30:
I am running the latest software.

Unfortunately, I did not check the monitor before restarting the pi this
morning.  Don't know if that resets the CRC, timeouts etc.  Currently there
are no CRC or discarded packets.  I combined the 6 genmon.log files that
logged during the "outages" into one file and is attached along with the
registers.

We did have some rain move through the area about the same time....don't
know if we had any lightening though.

There were several outages over one minute reported with the generator not
auto starting.  I rechecked my UPSs for low/high voltage or interruptions
and none were reported on either of the two that I can monitor.

Hope that these files shed a light on this anomaly.

> That does sound odd if you are getting a reading of 3190 volts. Do you see
> any communication errors (CRC, discarded or timeouts), you can see them
> from the "monitor" command or on the monitor page in the web UI. Also, do
> you see any errors in genmon.log? Are you using the latest version of the
> software? A dump of your registers would also be helpful. Are you still
> seeing unusual voltage levels?
>
> One limit of the software is that on a Nexus controller the software does
> not have the ability to read the threshold or pickup voltage so a default
> value of 143V and 190v respectively. The generator also have a timeout
> value (typically 5 seconds) that the software can not read via modbus. If
> the utility voltage is below the threshold voltage for at least the timeout
> value the generator attempts to start. The software will report an outage
> if the utility voltage is below the threshold voltage at any time,
> regardless of the timeout value, so it is possible that an outage will be
> reported even if the generator does not start due to the inability to read
> the timeout value.
>
> Having said all of this, it does not sound like you are experiencing
> normal behavior. I have a UPS that sends emails when the UPS is activated
> and I get notifications at the same times from the UPS and genmon when the
> utility voltage is low. Also the utility voltage of 3190 volts should not
> be occurring.
>
> <https://github.com/jgyates/genmon/issues/29#issuecomment-312252490>, or mute
> the thread

2017-06-30 03:45:39,206 : Error: timeout receiving slave packet for register 054 Buffer:0
2017-06-30 03:45:53,387 : Error: timeout receiving slave packet for register 5f1 Buffer:2
2017-06-30 03:45:57,813 : Error: timeout receiving slave packet for register 05d Buffer:0
2017-06-30 03:46:14,497 : Error: timeout receiving slave packet for register 09 Buffer:1
2017-06-30 03:46:28,914 : Error: timeout receiving slave packet for register 05c Buffer:6
2017-06-30 03:46:38,486 : Error: timeout receiving slave packet for register 052 Buffer:0
2017-06-30 03:46:49,212 : Error: timeout receiving slave packet for register 052 Buffer:2
2017-06-30 03:46:56,354 : Validation Error: Invalid register length (base) 0001:0e13 4 2
2017-06-30 03:47:06,452 : Error: timeout receiving slave packet for register 5f1 Buffer:5
2017-06-30 03:47:12,335 : Validation Error: Invalid register length (base) 0009:00000000 2 4
2017-06-30 03:47:22,927 : Validation Error: Invalid register length (base) 0001:0000 4 2
2017-06-30 03:47:23,975 : Validation Error: Invalid register length (base) 0009:00000000 2 4
2017-06-30 03:47:28,288 : Validation Error: Invalid register length (base) 0001:0000 4 2
2017-06-30 03:47:38,871 : Error: timeout receiving slave packet for register 056 Buffer:3
2017-06-30 03:47:59,900 : Error: timeout receiving slave packet for register 01 Buffer:0
2017-06-30 03:48:05,438 : Error: timeout receiving slave packet for register 5f1 Buffer:3
2017-06-30 03:48:15,866 : Error: timeout receiving slave packet for register 28 Buffer:4
2017-06-30 03:48:33,630 : Validation Error: Invalid register length (base) 0001:03fb 4 2
2017-06-30 03:48:40,913 : Error: timeout receiving slave packet for register 5f1 Buffer:5
2017-06-30 03:49:00,292 : Validation Error: Invalid register length (base) 001f:00000000 2 4
2017-06-30 03:49:05,362 : Validation Error: Invalid register length (base) 0001:00b0 4 2
2017-06-30 03:49:10,227 : Error: timeout receiving slave packet for register 5f5 Buffer:3
2017-06-30 03:49:19,660 : Error: timeout receiving slave packet for register 01 Buffer:0
2017-06-30 03:49:35,295 : Error: timeout receiving slave packet for register 05a Buffer:1
2017-06-30 03:49:41,795 : Error: timeout receiving slave packet for register 053 Buffer:0
2017-06-30 03:49:48,484 : Error: timeout receiving slave packet for register 052 Buffer:5
[... 236 lines cut ...]
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:03fb
0052:03fb                       0057:0000
0056:0000                       0055:0e13
0054:0000                       0208:03fb
005c:03fb                       005a:03fb
005f:03fb                       005d:03fb
003c:0000                       000d:0000
000e:0730                       000f:061e
000a:0086                       000c:005b
0005:0a00                       0006:0600
0000:0003                       0001:00000000
0008:0000                       0009:00eb
01f2:0102                       0012:0000
0011:009c                       0010:0511
0007:0000                       0019:0000
002b:0000                       002c:0000
002a:646b                       002f:0000
002e:0000                       0020:43e9
0021:0178                       001f:00b0
001e:00b0                       001d:43e9
001c:007e                       001b:43e9
001a:00b0
01f4:30303036303430303632

jgyates, 2017-06-30:
Yes, the communication error counters are reset on a reload however your logs show a lot of communication errors, which could cause invalid data  or unreported data. For every "timeout" reported in the log, an attempt to read a value from the controller was made and not value was sent back within the timeout period.  The "Validation" errors indicate that a value was received back from the controller that was not the expected length. This indicates that the software and the controller are out of sync due to a communication issue. I would keep an eye out for other communication errors going forward and re-check and validate the cable and your power and ground (problems here can cause comm errors).

jgyates, 2017-06-30:
one other note, I will review the code to see if there is anything that could be related to this issue, however, considering the information in your logs, I am considering this to be an issue local to your installation. I will still be happy to help provide any insight I can to help you resolve your issue. Let me know if you have any additional data.

Thanks
Jason

sgtmaj27, 2017-06-30:
Thank you for looking at this.  I am not overly concerned about this right
now, especially since it is not related to the software.  Since the pi runs
directly off the battery, I have ruled out dirty power that my surge
protector may have missed. I suppose that if memory was going bad in the
pi/sd card, that could cause problems.  Other than that, I will just let it
go and watch for this to happen again.

jgyates, 2017-06-30:
Here is some additional data that may help you determine if you have on going communication issues:

The comm stats should look like this:

   CRC Errors : 0 
   Percent : 0.00 (CRC error as a percentage of total packets received)
    Discarded  : 0  Restarts: 0  TimeOut: 0
    Packets per second: 64.82 . (or close to this number)
    Average Transaction Time: 0.0305 sec (or close to this number)

Based on your logs, the first seven errors in your log indicate the software sent a request over serial modbus to read a register value and the controller did not respond back for 3 seconds on each error logged. When this occurs in the software and error is logged and the next read is attempted. Each read waits at a maximum of 3 seconds. Ideally each packet is send and each response is received in 30 micro seconds (see Average Transaction Time in comm stats, e.g. 0.0305 seconds = 30.5 micro seconds). According to your logs, after seven timeouts then we have an error that states the length of the response is not the expected size, which indicates that the controller eventually responded to a request that timed out, which makes the software get out of sync and this is likely the cause of the behavior you are seeing (out of range utility voltage and mis-reported outage).

the real question is why is the controller not responding in 3 seconds when it should be around 30 micro seconds.

krisasmith, 2017-06-30:
I find that i always get 3 or 4 errors in genmon.log every time I reboot the pi and/or start the app. Is that normal?

2017-04-30 09:38:18,891 : Error: timeout receiving slave packet for register 052 Buffer:0
2017-04-30 09:38:18,917 : Validation Error: Invalid register length (base) 0001:0000 4 2
2017-04-30 09:38:18,929 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2017-04-30 09:38:19,933 : Error: timeout receiving slave packet for register 09 Buffer:0
2017-05-03 10:09:12,958 : Excpetion in InterfaceServerThread[Errno 22] Invalid argument
2017-05-03 10:09:50,132 : Excpetion in InterfaceServerThread[Errno 22] Invalid argument
2017-05-03 10:13:47,915 : Error: timeout receiving slave packet for register 09 Buffer:0
2017-05-03 10:13:48,955 : Error: timeout receiving slave packet for register 5ed Buffer:0
2017-05-08 13:57:17,627 : Error: timeout receiving slave packet for register 052 Buffer:0
2017-05-08 13:57:17,643 : Validation Error: Invalid register length (base) 0001:0000 4 2
2017-05-08 13:57:18,647 : Error: timeout receiving slave packet for register 5f1 Buffer:0
2017-05-14 12:39:04,457 : Error: timeout receiving slave packet for register 053 Buffer:0
2017-05-14 12:39:04,493 : Validation Error: Invalid register length (base) 0001:0000 4 2
2017-05-14 12:39:07,503 : Error: timeout receiving slave packet for register 5f1 Buffer:0
2017-06-14 17:58:04,856 : Error: timeout receiving slave packet for register 012 Buffer:0
2017-06-14 17:58:07,882 : Error: timeout receiving slave packet for register 010 Buffer:0
2017-06-20 07:56:18,467 : Error: timeout receiving slave packet for register 01 Buffer:0
2017-06-20 07:56:18,483 : Validation Error: Invalid register length (base) 05f1:00000000 2 4
2017-06-20 07:56:21,493 : Error: timeout receiving slave packet for register 09 Buffer:0
2017-06-20 10:51:42,187 : Error: timeout receiving slave packet for register 053 Buffer:0
2017-06-20 10:51:42,223 : Validation Error: Invalid register length (base) 0001:0000 4 2
2017-06-20 10:51:45,233 : Error: timeout receiving slave packet for register 5f1 Buffer:0

jgyates, 2017-06-30:
If memory serves Kris, you have an evolution controller so this is likely a separate issue than what sgtmaj27  is seeing.

This is not what I consider normal but I have seen it happen (timeout on startup). The solution is to add a small delay in the crontab before starting the startgenmon.sh script.

See the very bottom of this page:
https://github.com/jgyates/genmon/wiki/Setting-up-your-Pi-and-genmon

I believe the issue here has to do with the startup sequence of the pi. My best guess is that the serial port built into the chipset is not fully initialized so a delay is needed to let things settle.

I used to not see this issue, but after some software updates I started seeing it on my system. The delay in the crontab fixed it for me. Let me know if this corrects the problem for you.

sgtmaj27, 2017-06-30:
I hope that I don"t have a controller going bad!  I will check all of my
connections and hope for the best.

*Sent with EasyMail
<http://clkde.tradedoubler.com/click?p=259740&a=2839228&g=0&url=https%3a%2f%2fwww.microsoft.com%2fstore%2fp%2fEasyMail+for+Gmail%2f9nblggh626nc%3fcid%3dsignature>*

krisasmith, 2017-06-30:
@jgyates - thanks for the suggestion. I guess I should read the wiki. After reboot I don't see any errors.

jgyates, 2017-07-01:
Closing this issue. Feel free to reopen if you have any other data or questions.

closed 2017-07-01 by jgyates