jgyates/genmon #30: Serial Port Validation

zekyl314, 2017-07-01:
I have been having a heck of a time with my Raspberry pi 3 and the serial port.  I have even tried a raspberry pi 2 I have for another project.  I am looking for some pointers.  I am running jessie lite.  

I have a paper clip looping pins 2 and 3 on the 3.3v TTL to RS232 for RPi.  

Here is the output from the serial test.

/opt/genmon/OtherApps# python serialtest.py
Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 0 bytes

:/boot# cat cmdline.txt
dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=80d62713-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait

/boot# cat config.txt
.....
dtoverlay=pi3-disable-bt
enable_uart=1
core_freq=250

Any other suggestions?  This isn't my first project with a pi, just seems to be tripping me up somewhere.

krisasmith, 2017-07-01:
Can you take a picture of the setup?

zekyl314, 2017-07-01:
[image]

jgyates, 2017-07-01:
what is the output of the following command:
   ls -la /dev/serial*

It should look like this:
lrwxrwxrwx 1 root root 7 Jun 27 22:59 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Jun 27 22:59 /dev/serial1 -> ttyS0

jgyates, 2017-07-01:
It is hard to tell from the angle of the photo, but it looks like you may have looped pins 3 to 4 instead of 2 to 3. You may want to validate that.

zekyl314, 2017-07-01:
ls -al /dev/serial*
lrwxrwxrwx 1 root root 7 Jul  1 16:10 /dev/serial0 -> ttyAMA0
lrwxrwxrwx 1 root root 5 Jul  1 16:10 /dev/serial1 -> ttyS0

Here are 2 more photos.

[image]
[image]

zekyl314, 2017-07-01:
/opt/genmon/OtherApps $ python serialtest.py /dev/serial1

Loopback testing for serial port /dev/serial1...

error open serial port: Could not configure port: (5, 'Input/output error')
Error opening Serial Port /dev/serial1

/opt/genmon/OtherApps $ python serialtest.py /dev/serial0

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
FAILED: Sent data does not match receive. Received 0 bytes

krisasmith, 2017-07-01:
Do you have something besides a paper clip?

jgyates, 2017-07-01:
one thing to try that will let us narrow the problem area: remove the TTL to RS232 converter and clip the TX and RX pins on the header together. This will loop back the TTL signals and if it work then there is a problem with the level converter, if it does not then we should look at the raspberry pi config and hardware.

jgyates, 2017-07-02:
Hi zekyl314,

Were you able to get the serial port working?

Jason

zekyl314, 2017-07-02:
Sorry, was gone all day.  I connected the pi pins together and blam, works like a charm.  So that is good at least.  Thoughts?  Could I have a bad converter?

root@genmon01:/opt/genmon/OtherApps# python serialtest.py

Loopback testing for serial port /dev/serial0...

Serial port opened
write data: sent test string
waiting to received data....
PASSED! Loopback successful

jgyates, 2017-07-02:
Yes, if you have validated the following, then I would guess a bad converter:
- Connections to the converter are correct (they looked right in the diagram. use a volt meter to validate if you have one)
- You convert is a 3.3v to RS232 convert. There are some on the market that are 5v to RS232 and they will not work with the Pi.

zekyl314, 2017-07-04:
I ordered a new RS232 converter.  We shall see.  Thanks for the help so far, didn't even think about looping the pins on the pi.

zekyl314, 2017-07-13:
Sorry it took so long, but I finally got my part in.  Everything is working now as expected.  Looking forward to working on this and getting it all setup the way I want.

jgyates, 2017-07-13:
No problem. Thanks for the feedback. I will close this issue. Feel free to reopen or create a new issue if you have any other questions.

closed 2017-07-13 by jgyates