jgyates/genmon #31: Non-Issue - Build Variation - PI Zero W with MAX3232 board

madisdev, 2017-07-02:
Hey Guys,

So i finally got around to putting the box in the generator and it looks like all is working smoothly. I just wanted to show off my build a little bit to show how I used some different components.

This isn't going to be as cool as some other builds, I don't have a full home automation system or a fiber to serial link ;)

So I decided to use a PI Zero W as I can pick them up cheap at my local Microcenter(10$ each). I also choose to use a MAX3232 board as I didn't want to use a DB-9 as it seemed a little redundant, Considering the DB9 boards use the same chip.

I picked up a waterproof enclosure on ebay and added some rubber cable seals to it. I did a little soldering and wire crimping for the connectors. Other than that nothing too crazy. I am pulling power from the generator off pins 1&2.

Thanks again to Jason for the great work he puts into this project. And everyone else that has contributed.

Here is a pic before I hot glued the PI Zero down and sealed everything up. I'm not sure if heat is going to be an issue off the PI Zero, but its something ill monitor. 
Cables on the left are TX/RX and on the right is Power/Ground.
I'm powering PI using pins 2(5v) and 6(Ground)
And connector to MAX3232 are pins 4,6,8,10 are 5v, Ground,TX and RX respectively.

Parts list after the pic :)

[image]

Enclosure
http://www.ebay.com/itm/NEW-waterproof-Plastic-Project-Box-Enclosure-DIY-100x68x50mm-L-W-H-/151799269041?hash=item2357f106b1

Cable Gland
PG7 Black Nylon Waterproof Cable Connect Cord Grip Cable Gland 3-6.5mm US

MAX3232 Board
http://www.ebay.com/itm/2PCS-MAX3232-MAX3232CSE-Serial-Port-Mini-RS232-to-TTL-Converter-Adaptor-Module-/222084117081?hash=item33b53eb259

jgyates, 2017-07-02:
Very nice. I am glad to hear the feedback on the Pi Zero. It is always good to have people share their experience and implementation details. If I ever need to change out my cable I would likely power the PI via the GPIO header like this.

griffths, 2017-07-02:
madisdev,

Thanks for the followup on your version of this.  I too am planning to use the MAX3232 board but was unsure how to wire it up.  I now see how to do it thanks to your pic so I can proceed with a little more confidence that I am connecting everything correctly as the board didn't come with any instructions.

zekyl314, 2017-07-13:
Great work on this, I like how you used the Zero.  I may have to replace mine with a zero someday, and use the Pi 3 in the house for something else.

JeanB51, 2017-08-15:
Hi Madisdev.
Here is mine. I took inspiration from yours, I used the Pi ZeroW, and the Max3232 board. I soldered everything. I too get the power from my Nexus controller. I did not need waterproofing, as I could bring the 4-wire cable inside the garage next to the generator.

I decided to connect the box to the wire with a molex on the side of the box, in case I ever want to disconnect the thing without too much trouble. My cutout isn't too great - I didn't have a dremel then..

Thanks for the inspiration, and thanks again to Jason for this very well made (-and supported) project.

[image]

madisdev, 2017-08-15:
Jean,

Nice job. Looks nice and clean. I'm glad I provided some inspiration and it worked out for you

Just to add that my layout wasn't that good. I didn't leave much room to remove the SD card. I had to troubleshoot an issue when my PI got off the network and the only way was to mount the SD card in my computer. Good to see you have some nice spacing around the PI. 

I'm assuming under your Pi you glued down some standoffs or did you screw them into the plastic.

Thanks.

JeanB51, 2017-08-15:
Madis,
Thanks for the comments. The spacing was more or less lucky, this is the box that I found at my local electronics supplier. The standoffs are screwed in from the back, as I didn't need waterproofing. And the box is stuck to the wall with some velcro strips.
I'm more of a software guy, and all this hardware nitty-gritty is new to me. I learned as I went along, from Jason's pictures, as well as yours, plus some reading on the Web.
I was actually great fun overall for a summer project !

Now I am hoping for a power failure, to see how this runs in real life. 
Maybe this afternoon, some heavy thunderstorms in the forecast.

Cheers

Jean B

madisdev, 2017-08-15:
I actually need to get a UPS so I can get alerts when the power goes out. Right now I am mainly using it to keep track of the weekly exercise and alerts.

USAFPride, 2017-11-11:
@madisdev, on the bottom of your max3232 board, does it have RS232<->TTL stamped on it?  If so, which side do you have towards the piZero?

madisdev, 2017-11-11:
I'll double check for you.

madisdev, 2017-11-11:
Hey @USAFPride,

Sorry I didn't get back to you earlier I was out for most of the day. The RS232 side is the Generator connection and the TTL is the PI. Be careful as I found that the connections on my MAX3232 board didn't connect to the other side, although there were pads on each side. I used the connections on the back side(side without the chip). Let me know how yours goes, as I had some issues with my WIFI connection dropping. I had to add an external antenna to the PI, I can show my results of that if your interested.

USAFPride, 2017-11-12:
@madisdev, thanks, you solved it for me.  I had it the same direction, but I had is soldered to the top pads.  Now the serial works.  The pads are small enough, so doing it 2X doesn't make my older eyes so happy!?

madisdev, 2017-11-12:
Yea I had the same problem. Glad I could help.

kalex1114, 2017-11-20:
@madisdev @USAFPride  - do you have close up pictures of your wiring and rs232 board connections? I'm about to start this project and wiring is fairly new to me.  Would be great if you can post pictures.  Thank you

madisdev, 2017-11-20:
I put this together in MS Paint. As Im at work.
[image]

[Please refer to this if you have questions about PI Pins.](http://pi4j.com/pins/model-zero-rev1.html)

kalex1114, 2017-11-20:
Ah thats perfect.  Now I need to source molex connectors and plugs and everything.  Thank you

madisdev, 2017-11-20:
Just wondering, Would people be interesting in buying Molex connectors made up with leads? I know buying connectors and crimping tool for a one off really doesn't make sense.

Not really looking to make a profit, just cover shipping exp.

USAFPride, 2017-11-20:
@madisdev, I bought 4 connectors and 8 pins direct from digi-key and spent less than $8.  I do not have a crimper, but I soldered the wire to the connector (which IMO is better)

kalex1114, 2017-11-20:
@madisdev I would be your first customer 😀 I was going to post about buying one. My soldering skills are crap 😀. My plan is to remove mobile link and use that hole to mount the enclosure with raspberry on the outside of the unit. I have 2017 16kw air cooled model

madisdev, 2017-11-21:
I would just throw this out there, what is the distance from the generator to your Wifi access point. As I had to put an external antenna on my Pi Zero to get it to consistently work.

kalex1114, 2017-11-21:
About 10-15 feet through the glass patio door.  Signal is good. I can also boost the signal on my UniFi if needed.  I should have ran an Ethernet cable when generator was being installed but didn’t think about it and didn’t know I can do this

madisdev, 2017-11-21:
You mentioned your soldering skills are not that great, what was your gameplan? Were you going to solder directly to PI? Do you have the same MAX3232 board?

kalex1114, 2017-11-21:
Plan was to look at pictures of how it was done, watch videos and learn to solder. Or buy a Pre made harness. I didn’t get max3232 board yet. Only have raspberry pi 3b that has Wi-Fi built in. I was going to do it as you suggested in your diagram . Molex plug to the accessory port. To the max3232 board and then connect those to the pi. I currently have mobile link with service through April on it so have time to lear, practice and do it 😀. If you have other suggestions I’m listening 😀

USAFPride, 2017-11-22:
[image]

My setup

kalex1114, 2017-11-22:
Very nice.  Thank you.  its not secured inside the enclosure? is that indoors or do you have it attached to the outside of the generator?

krisasmith, 2017-11-22:
Is that mounted on the side of the genset?

USAFPride, 2017-11-22:
@kalex1114 - a bit of hot glue holds it down.  Not enough to make a mess, but enough to hold it in place

@krisasmith - the picture is on my desk, but it is currently mounted directly under the connection port

kalex1114, 2017-11-22:
@USAFPride thank you.  Can you post a picture of how its mounted?  I think this is what i envision my setup to look like.

krisasmith, 2017-11-22:
Nice setup, I went with a pi 3 b so I could use a wired network connection.

kalex1114, 2017-11-22:
@krisasmith - can you post your pictures?  I also have Pi 3b but will use wifi as i forgot to run the wire to the generator :)

USAFPride, 2017-11-22:
Most Pi wifi issues are due to a) wifi power management left on b) weak wifi (duh).  The Pi's generally do not  handle disconnecting and reconnecting well, so I run a simple script every 5 minutes to ping my router, and if it doesn't get a response, to restart `if`.  This has ended all my wifi disconnecting issues which required a restart (pulling the plug) to reconnect

checkwifi.sh
`#!/bin/bash

#router ip
ping -c4 192.168.0.1  > /dev/null

if [ $? != 0 ]
then
#make sure you have the correct wlan
    ifdown --force wlan0
    ifup wlan0
fi
`

krisasmith, 2017-11-22:
I have a network run in my garage that is right next to the transfer switch. I ran a 4 conductor wire to the genset for power and data. In the last picture you can see the gray wire in the upper left going through the key hole shaped mounting hole.

[image]
[image]

[image]

kalex1114, 2017-11-22:
Thank you.  I made a stupid mistake of not running anything to the generator.  My transfer switch is about 35 feet away from the generator.  Stupid zoning doesn't allow generators on the side of the house.

krisasmith, 2017-11-22:
I was able to fish the wire after the fact, it was about 15ft. It took some persistence but I was able to get it. I used shielded security wire from home Depot.

proppilot, 2017-11-22:
Thanks for the script on the wlan reset heartbeat.  This was my last issue to resolve as I get disconnects about every two days.  Here are a few shots of my install.  Can’t do a project without a label on the box !

[image]

proppilot, 2017-11-22:
[image]

proppilot, 2017-11-22:
I added the Active light with a small python script that runs and blips the light every two seconds so I know the processor is active.  I added this when I was losing connection to the Pi and before narrowing it down to WiFi disconnects.  The power light is taken off the GPIO +5 pin so I know that the Pi is powered up by the 12v-5v power converter seen in the box and the active light is off a GPIO pin.

super fun project.  Now looking for other things to do with a RPi 3 !

kalex1114, 2017-11-22:
@proppilot that is awesome.  do you have a link to the enclosure you used? how did you attach it to the  generator?

proppilot, 2017-11-22:
The box is a Cantex (p/n 5133710) 6 3/4 x 6 3/4 in electrical box from Home Depot.  16 dollars

Attached it to the generator with outdoor velcro and took the four conductor cable into the generator via the gland that the gas pipe goes through.  Used molex connectors at either end ( you can see replaced the DB9 with Molex on the MAX232 board ) so it can be removed in a jiffy if my maintenance guy needs to come over !

kalex1114, 2017-11-22:
@proppilot sorry to be a pest but can you show the picture of the cover and wiring attached to it?

proppilot, 2017-11-22:
Cannot do as I am away but this is what is on/in there.  There are two LEDs on it.  The shorter leads on both LEDs are tied together and one wire back to the ground pin on the GPIO connector.  

Each of the long leads ( Anode or + ) have a 330 ohm resistor (so one for each LED) with the power LED going back to the +5V pin on the GPIO connector and the other for the ‘active’ LED back to the GPIO pin that I drive with the python program used to flash it.

krisasmith, 2017-11-22:
@proppilot nice setup, the status led's are awesome.

USAFPride, 2017-11-23:
@proppilot, Nice setup.  When you get a chance, would you be willing to post a parts list and the art for your cover?

proppilot, 2017-11-25:
[genmon.pdf](https://github.com/jgyates/genmon/files/1503050/genmon.pdf)
[genmon.pptx](https://github.com/jgyates/genmon/files/1503051/genmon.pptx)

Here is the PPT and PDF artwork for my box cover

proppilot, 2017-11-25:
[image]

This is the code for the python script that I start at reboot in crontab to flash the light on the box telling me the processor is active.  Like a heartbeat

kalex1114, 2017-11-25:
@proppilot Thank you Tony.  If you can please post the parts list.

proppilot, 2017-11-25:
For the LEDs themselves or ?

kalex1114, 2017-11-25:
Sorry for the actual pi related stuff not leds. at least for me

proppilot, 2017-11-25:
Got it !  Here are the parts.  All from Amazon (prices are in CAD as I am up North) and the box is from Home Depot

[image]

[image]

[image]

[image]

[image]

Some details.  I bought the CanaKit as it had everything needed to get going including the 32 Gb data card which was preloaded and enabled super easy startup and configuration.  

The buck converter converts 12 volts from the battery to 5 volts for the Pi.

(soldering skills needed) I removed the wifi chip antenna from the Pi 3 and cut the end off the antenna connector/cable that came with the antennas and soldered it in place.  I little bit of 5 minute epoxy seals it up nicely and the connector end was mounted through a hole in the box.  I put it on top and ended up having to seal it all up with epoxy and if doing it again would put it on the side and flip the antenna 90 degrees as it has that ability built in.  I bought two sets a) in case I screwed up and b) to build a second unit for my buddy who has a 22 kW Synergy.

The RS232 converter is per the jgyates project description as well as the molex connectors and pins.  If you can solder, don't buy a crimper for such a small project.  Two pins...done and done...

Knowing what I do now about the hardware configuration, I could build it for about $ 100 USD.

As for mounting everything, 3M double sided tape from home depot.  It is sticky so you only need a couple of small pads to mount the power converter and the RS232 converter.  The CanaKit came with a really nice case so I used the bottom of the case which I attached to the box with the 3M tape and then popped the Pi into it and secured it in place with the white nylon screw you see coming through the side of the box.  It just snugs over the USB connector and keeps the board in place.  Easy removal of the Pi if needed.

Four wire cable (two for the serial bus and two for power) with an inline 2 amp fuse connector.  I soldered banana clips to the end of the power leads and clipped them to the battery lugs.  As the box is held to the generator with outdoor velcro, I can remove any trace of the unit in under five minutes if my maintenance guy needed to show up.  The only evidence would be the removal of the little sticker that was covering the connector under the Evolution controller.  

As my unit came with a bumper to bumper 10 year parts and labour warranty ( a September special to drive sales just before the hurricane hit PR and FL - then it disappeared ) I want to keep it as sterile as possible to prevent any potential warranty conflict.

Thats about all I can think of.

proppilot, 2017-11-25:
PS I added a small 3/16" drain hole in the bottom of the enclosure as mine is outdoors.   As I opened and closed the box a number of times while figuring out the fit and finish, the gasket deformed a bit and I got some water in the box that accumulated at the bottom.  As the project is done, I put a light bead of silicone around the gasket before sealing it up and the drain hole in the bottom just in case.  I think it is good to go.  

I added the code from @krisasmith to a sh script that runs on boot so hopefully my disconnect issues are history.  

It is sad when projects are done as the learning is quite fun !

proppilot, 2017-11-25:
Hi Kalex, yes I took power through a fuse from the battery.  My concern was any damage to the GenMon ie water ingress (who knows in a Canadian winter) causing a short could perhaps damage the controller if I was taking power from Port 1 (Evolution controller port).  It is a nice solution to get +5v from Port 1 but....

madisdev, 2017-11-25:
@proppilot Please don't take this the wrong way, you have a nice build. I just had some comments on your build. It sounds a little concerning that you had water accumulate in your work box. :( I opted to use a waterproof work box(see link in first post), even though mine is installed in the Generator enclosure. I understand some people feel more comfortable installing the box on the outside of the generator. I would just suggest that if people go that route to make sure the box is made for the external environment, I.E. water and temperature changes. And also that it can withstand and impacts from branches heavy snow ect.  And I would be careful on how you route your wires to the internal of the generator. Not sure I would mix gas and electric in one opening :|

Since I went the internal install method I just needed to install the antenna on the outside of the enclosure. I used the spot the mobile link would be installed for the antenna and installed the workbox under the control board. I mounted it with double-sided foam tape, but would probably use Velcro if I need to remove it in the future. All components can be installed and removed in a short amount of time.

I got inspired to add the external antenna after a link was posted by @jgyates in a different thread, [data allicance](https://www.data-alliance.net/) has good products with decent prices. The PI Zero W does not come with a antenna connector, so I had to solder on one myself. Going forward I think the [Orange PI](https://www.ebay.com/itm/Orange-Pi-Zero-Plus-H5-Quad-Core-Cortex-A53-Open-source-512MB-DDR3-Development/391921843397) line of PI's would be a better option as the ones I was looking at have external antenna connectors. I found after installing the external antenna that my wifi strength increased enough to keep a constant connection.

Some pics.
View of workbox installed below the main board.
[image]
View of the external antenna from inside enclosure with wire routed back to workbox.
[image]
Outside view of antenna connector, this will allow installation of many different types of antennas since its a common connector.
[image]

proppilot, 2017-11-26:
@madisdev no offence taken at all.  The enclosure is definitely an outdoor electrical box.  

The issue was that the compression gasket failed because I opened the lid probably too many times as I was experimenting.  No way it would happen if the box was only sealed once or twice.  I suspect that the gasket compressed one way and I likely rotated the lid during the process.  I was pretty shocked to see some water in the enclosure and not positive if it came through the gasket or through the antenna connector.  It may have been the antenna connector as I only had it screwed through the hole and not sealed from the inside in case I needed to modify or move it.  

I definitely like your install.  For me, I did not want to leave any traces if I needed to remove it however yours will look 100% stock OEM :-)

USAFPride, 2017-12-09:
@proppilot, did the wifi script solve your connection issues?

jgyates, 2017-12-09:
@USAFPride, the wifi script helped me out. I was having some random disconnect issues but now they are gone. Thanks!

proppilot, 2017-12-09:
Absolutely.  Up 12 days straight with zero issues

Regards,

Tony Varga

USAFPride, 2017-12-16:
All:  

I've found a better script to use for resetting the wifi **_AND_** it will reboot after 5 (configurable) attempts.

[Reboot](https://github.com/ltpitt/bash-network-repair-automation)

kalex1114, 2017-12-28:
so I finally got all the parts and attempted to solder the mini max3232 board with the wires i need.  Wow its tiny and very hard to work with.  
Here is what my abomination looks like :)
[image]

Oh ye and i managed to burn my finger too.  But its expected.  I think i need to get a bigger board or use @krisasmith method.  

Any soldering tips are appreciated :)

USAFPride, 2017-12-28:
@kalex1114, first "tin" the pads (put solder on the pads).  Then tin the wire.  Then lay the wire on the pad and heat the tinned wire.  It should melt both the solder in the wire and the pad quickly and then take the heat off quickly.

kalex1114, 2017-12-28:
@USAFPride ah thank you for the tips. I'll try it.  What temp should the solder be at? is 350c enough ?

USAFPride, 2017-12-28:
@kalex1114 - Just hot enough to melt the solder quickly, but not so hot that is vaporizes :)

[32 more replies]

closed 2018-12-05 by jgyates