jgyates/genmon #32: Got alert generator was starting

krisasmith, 2017-07-09:
I had a quick utility outage this morning. Below are the email alerts I received. My genset is configured to start after 60 seconds.

Question: why did I get an alert the gen was starting although the total utility outage was 6 seconds?

-----
Time: 09:26:30 AM
Date: Sun 09-Jul-2017

Utility Power Out at 2017-07-09 09:26:29

-----
Time: 09:26:37 AM
Date: Sun 09-Jul-2017

Notice from Generator:

Switch State: Auto
Engine State: Cranking
Active Relays:

No Alarms: 0001:00010000

Last Log Entries:
    Start Stop Log: 07/05/17 10:05:49 Stopped - Auto
    Service Log:
    Alarm Log:      04/10/17 09:24:08 Charger Missing AC : Alarm Code: 0000

Last Alarm Code: Error Code 0000: No alarms occured since controller has been power cycled.

------
Time: 09:26:38 AM
Date: Sun 09-Jul-2017

Utility Power Restored. Duration of outage 0:00:06

------
Time: 09:26:43 AM
Date: Sun 09-Jul-2017

Notice from Generator:

Switch State: Auto
Engine State: Off - Ready
Active Relays:

No Alarms: 0001:00000000

jgyates, 2017-07-09:
At the moment, this is the expected behavior of the software. Here is why: The controller detects the utility voltage and if the value is below the controller threshold voltage (a setable parameter in the controller) then the software considers this an outage. The controller also has a delay timer (also a settable parameter) that determines how long to wait before starting the generator. If the utility voltage rises above the pickup voltage (another settable parameter) before the timeout, then the controller does not start, if the voltage stays below the threshold voltage and the delay timer expires, the generator attempts to start. The problem in the implementation is that the software does not have the ability to read the timer delay. This leaves the software with a few choices on how to implement things:

1) ignore the timeout delay and just report when the utility voltage drops below the threshold voltage (notices of short outages, even if the generator does not start due to the short duration of the outage)
2) Add a configuration parameter that mirrors the value of the controller value (this would allow the software to simulate the delay and not notice would be given on utility disruptions that are shorter than the timeout delay)

I chose option 1 as the implementation as it will allow the user to know when there are fluctuations  in the utility voltage. In an outage that last longer than the timeout delay you will receive an additional notice that the generator is starting.

I guess I could make an optional configuration parameter. This would allow the user to chose the behavior of option 1 or 2. Let me know if you think this is something of value.

Thanks
Jason

jgyates, 2017-07-09:
OK. I misread your post. I responded with the way the software works, but you actually got a message that your generator was starting. 

I see that the contents of register 0001 was 00010000, which indicates that the engine is starting. So it looks like your engine was starting.

I would validate your delay timer in your controller settings. I think the default is 5 seconds, which would explain the start notification. If your setting is truely 60 seconds I would validate this with a test of turning off your power at the outside breaker.

krisasmith, 2017-07-09:
I checked the delay timer earlier and it's definitely set to 60 seconds. I'll see about testing later today to see what the actual behavior is.

krisasmith, 2017-07-14:
I forgot about this, finally remembered to test. Reading though the chain of events I noticed the following:

1. The engine state "cranking" is inaccurate. It should indicate that the delay countdown has started.
2. The text message indicating the generator is "ready" seems accurate however it was still in a cool down state.
3. The active relays in the email alert seem delayed.

<br><br>

-When I tripped the breaker in the transfer switch I got this email.

------
_Time: 11:41:50 AM_
_Date: Fri 14-Jul-2017_

_Notice from Generator:_

_Switch State: Auto_
_Engine State: Cranking_
_Active Relays:_

------

-Then a minute later when my generated actually started (its set on a 60 second delay) I got this email.

------

_Time: 11:42:51 AM_
_Date: Fri 14-Jul-2017_

_Notice from Generator:_

_Switch State: Auto_
_Engine State: Starting_
_Active Relays:_

------

-and this test message at timestamp: 11:42:55

------

_Generator Running_

------

-and this email

------

_Time: 11:42:57 AM_
_Date: Fri 14-Jul-2017_

_Notice from Generator:_

_Switch State: Auto_
_Engine State: Running_
_Active Relays:_

------

-After restoring power I got this text message at timestamp: 11:43:01

------

_Generator Ready_

------

-and this email

------

_Time: 11:43:03 AM_
_Date: Fri 14-Jul-2017_

_Notice from Generator:_

_Switch State: Auto_
_Engine State: Cooling Down_
_Active Relays:_

------

-and this email

------

_Time: 11:43:59 AM_
_Date: Fri 14-Jul-2017_

_Notice from Generator:_

_Switch State: Auto_
_Engine State: Stopped_
_Active Relays: Ignition On, Starter On, Fuel Relay On_

------

-and this email

------

_Time: 11:44:05 AM_
_Date: Fri 14-Jul-2017_

_Notice from Generator:_

_Switch State: Auto_
_Engine State: Off - Ready_
_Active Relays: Ignition On, Starter On, Fuel Relay On_

------

jgyates, 2017-07-14:
Hi Kris,

This is good info. There are two distinct values in the main status register that show the status for starting vs cranking. These may be reversed in my logic.  This would appear to fit with the description you posted, i.e. the "Starting" state is when the countdown begins, and "Cranking" state is when the engine actually attempts to turn over.

I will do some testing on my end and see if I can figure this out. It may be a day or two before I can get to this, but I will keep you posted.

Thanks
Jason

jgyates, 2017-07-14:
one question: Do the active relays eventually show and empty status once the engine has stopped (after the cool down has ended)? I will look at the delay issue as well.

krisasmith, 2017-07-14:
Yes, when I view it in the web UI there is typically a delay both when starting and stopping for the relays. They eventually show the correct information.

jgyates, 2017-07-14:
I just checked in a new version of genmon.py that should help with the slow relay update. It still will have a delay since the web UI only updates every 5 seconds. The Web UI only polls genmon.py so there will be a 5 second delay in any update to the web page. Internally in genmon.py there are 2 list of registers to read. One list is read at a higher frequency than the other to improve responsiveness to critical events like power outages, etc. In the code the liquid cooled models were reading the output relay status in the higher frequency list and the air cooled were read at a lower frequency. I changed the code so both models will read the output relay status a the higher frequency. Again, this will not make it "real time" but it should improve it to some degree. 

I will look at the start vs cranking delay issue later today or tomorrow.

jgyates, 2017-07-14:
OK. Try the version that is checked in now. It has the following changes:

I changed description of some status conditions related to Nexus and Evo Air Cooled models.

At a power outage your should receive notices for the following conditions:
Power Outage
Start Delay Timer Activated
Cranking
Running
Cooling Down
Stopped
Off - Ready

Previously there was a state for Starting and Cranking that were mislabeled.

The state that was labeled cranking is now "Start Delay Timer Activated" and the state that was Starting is now Cranking. This appears to match the behavior better. Oddly enough, the air cooled evolution units do not appear to show the "Start Delay Timer Activated" state in the status register.

Let me know if this works for you.

krisasmith, 2017-07-14:
I should be able to test it tomorrow.

jgyates, 2017-08-04:
Hi Kris,

Did you get a chance to test this?

Thanks
Jason

krisasmith, 2017-08-05:
I've been out of town, I'm hoping to test tomorrow morning

jgyates, 2017-08-05:
ok. No rush, just following up. Thanks

jgyates, 2017-09-11:
Hi Kris, no rush, I am closing this issue for now. Feel free to reopen once you have validated and if the fix is not working for you. Thanks for your help in finding this.

closed 2017-09-11 by jgyates

krisasmith, 2017-09-13:
I had a power outage on Monday night, everything seemed to work fine.