jgyates/genmon #33: genserv.py problem with webpage

drandyhaas, 2017-07-15:
Other things working fine, but the genserv.py script gives a webpage problem for me, at http://192.168.55.105:8000/. See attached pic. This was from google chrome, but I also tried iexplore, and it's the same...

[image]

jgyates, 2017-07-15:
Hi drandyhaas,

You may have an old version of Flask. Try this:

Flask version 0.12 or higher is recommended. You can validate your version of Flask by the following command:

 pip freeze | grep Flask

The output should show Flask==0.12.0 or higher. If your version of flask is not at least 0.12.0 then update your Flask with the following command:

 sudo pip install Flask -U

Then restart genserv.py

drandyhaas, 2017-07-15:
That did it! Who are you?! Amazing software, and lightning-fast support, for free! I thank you again, sir.

closed 2017-07-15 by drandyhaas