jgyates/genmon #35: Nexus Liquid Cooled Utility Voltage

jgyates, 2017-08-07:
From JeanB51:

Here is the output of the status and maint commands.
My question: Why does it show 0V for the Line State Utility Voltage ?

generator: status maint
Status:
Engine State:
Switch State: Auto
Engine State: Off - Ready
Battery Voltage: 14.0V
RPM: 0
Frequency: 0.0 Hz
Output Voltage: 0V

Line State:
Utility Voltage: 0V, Min: 0V, Max: 0V
Utility Threshold Voltage: Low Voltage: 144V

Last Log Entries:
Start Stop Log: 07/31/17 10:39:42 Stopped - Auto
Alarm Log: 07/31/17 10:35:24 Unknown 0x01

General:
Monitor Time: Monday August 7, 2017 12:01:12
Generator Time: Monday August 7, 2017 11:43
Maintenance:

Generator Serial Number : 0008565313

Exercise:
Exercise Time: Tuesday 07:00 Quite Mode Off

Service:
Service Scheduled: Next service in 99 hours
Total Engine Run Hours: 54
Firmware : V2.12, Hardware : V1.01

Model : Generac QT036, 36KW, water-cooled, Nexus controller, Propane fueled. Installed 2014.

Here is the output from registers;

generator:registers
Registers:
Num Regs: 158
Not Changed: 6483 Changed 858 Total Changed 0.13
0038:01f4 0034:01f4
0037:01f4 0032:01f4
05f5:001e 05f4:0032
05f1:01f4 01f1:0010
05ed:0018 023b:0000
023e:0000 003b:0000
05fa:01fe 0059:01f4
0058:01f4 0053:03f2
0052:03f2 0057:01f4
0056:01f4 0055:01f4
0054:01f4 0208:03f2
005c:01f4 005a:01f4
005f:01f4 005d:01f4
003c:001e 000d:0000
000e:0a3b 000f:0807
000a:008c 000c:0036
0005:0700 0006:0200
0000:0006 0001:00000000
0008:0000 0009:0000
01f2:031d 0012:0000
0011:0090 0010:0111
0007:0000 0019:0000
002b:0000 002c:0000
002a:65d4 002f:01f4
002e:01f4 0020:0000
0021:0000 001f:46bc
001e:00f9 001d:4498
001c:0031 001b:454f
001a:0063
01f4:30303038353635333133

jgyates, 2017-08-07:
Jean B,

The issue with the utility voltage being zero is likely caused by the fact that you have a Nexus Liquid cooled model and to my knowledge this is the first liquid cooled Nexus model that has been tested and given feedback. Generac has four controller variants (Nexus Liquid Cooled, Nexus Air Cooled, Evolution Liquid Cooled, and Nexus Air Cooled). The way to proceed here is that I need to examine your registers that you sent and see if I can locate the proper register for the utility voltage for a Nexus Liquid Cooled model. If I can not our next option is to enable some debugging in the code and that my help us narrow things down. Let me know if you are seeing any other odd behavior. The code currently treats the Nexus Liquid Cooled and Air Cooled models the same. Let me review your registers and we can go from there.

Jason

jgyates, 2017-08-07:
OK. Let's try this:

From ClientInterface.py, issue the following command:

generator: getregvalue=001e

Issue this command repeatedly over the course of a few minutes. Does this value ever change. In the register values you sent the value of your register 001e was 00f9 hex or 249 decimal. This register is read my mobile link, but in my testing so far I have found no use for it so if this value changes over time and stays in the utility voltage range, then it is likely the utility line voltage fro the nexus air cooled model and I can make a change to the code to reflect this, if not I will keep looking...

JeanB51, 2017-08-07:
Ok, I have been typing the command for the last 10 minutes, and I always get 00f9.

jgyates, 2017-08-07:
ok. while that does not rule it out, it does make it less likely. Generally there is some fluctuation in the utility voltage.  Another thing to try, assuming you have your pi and network connection setup to be powered during an outage,  take a snapshot of the registers via the "registers" command, then turn your power off at your outside switch to simulate a power outage, then take another snapshot of the registers. We can compare the values, if this does not show us anything then we can move on to enabling the reverse engineering features in genmon.py.

I just posted the updated genmon.py that should fix the unknown values. Let me know if it does not.

JeanB51, 2017-08-07:
Ok, I will try this, it may take a lilttle while... I have not yet setup my UPS for this, but I could move one around. And I have to warn the household about the "power outage"..
I'll be back soon.
I will also update with the changed version of genmon.py.
JB

JeanB51, 2017-08-07:
ok - newbie github question here - what is the command to get the updated genmon.py onto my Pi ?

jgyates, 2017-08-07:
"git pull" from the genmon directory should do it.

JeanB51, 2017-08-07:
The Alert codes are all good now . Thanks !

I notice that when I start genmon, I get an email with Outage Notice - 
I guess because it sees the voltage at 0.

JB

jgyates, 2017-08-07:
yes, that is the case. I can add a flag to disable the outage monitoring if we can't figure this out, but we still have a few more things to try if you are willing to test. Since I don't have the generator controller you have,  we will do this at your pace so don't feel rushed to test if you have other things to do. Thanks, Jason

JeanB51, 2017-08-07:
Don't bother with a flag - we'll figure it out :-)
Thanks again for the quick support.

JeanB51, 2017-08-09:
Here is a question for you - when using the ClientInterface.py program, why do I have to prefix all the commands with "generator:" ? I understand that it is the same syntax as with the email queries, but as the CLI program does not seem to accept other kinds of input, this prefix seems superfluous, doesn't it ?

jgyates, 2017-08-09:
Good question. There are a couple of factors:

1) I reuse modules across other projects that I have not made public
2) These other projects use the same command line and email processing modules
3) I use gmail filters to move command emails to separate folders for each project (e.g. generator : status, othersystem: status)

If genmon were the only project you are correct this would be superfluous, but due to the way all of this development unfolded it was easier to maintain for me to add the prefix.

JeanB51, 2017-08-14:
Hi Jason

Ok, this morning I could do a full test, with my wifi on UPS. 
I switched the main power off, let the generator take over, then switched the power back on.

Here are the registers. 
Sorry for the longish post, I grabbed the registers at all the stages of the test:
- before the test
- at power off
- generator running
- utility power back on
- house switched  back to the grid power
- generator off after cooling

registers before:

>generator:status
Status:

Engine State:
    Switch State: Auto
    Engine State: Off - Ready
    Battery Voltage: 12.7V
    RPM:     0
    Frequency: 0.0 Hz
    Output Voltage: 0V

Line State:
    Utility Voltage: 0V, Min: 0V, Max: 0V
    Utility Threshold Voltage: Low Voltage: 144V

Last Log Entries:
    Start Stop Log: 07/31/17 10:39:42 Stopped - Auto 
    Alarm Log:      07/31/17 10:35:24 Low Oil Pressure 

General:
    Monitor Time:   Monday August 14, 2017 12:05:19
    Generator Time: Monday August 14, 2017 12:05

>generator:registers
Registers:
Num Regs: 158
Not Changed: 4653212 Changed 603674 Total Changed 0.13
0038:01f4                       0034:01f4
0037:01f4                       0032:01f4
05f5:001e                       05f4:0032
05f1:01f4                       01f1:0010
05ed:0018                       023b:0000
023e:0000                       003b:0000
05fa:01fe                       0059:01f4
0058:01f4                       0053:01f4
0052:01f4                       0057:01f4
0056:01f4                       0055:01f4
0054:01f4                       0208:03f2
005c:01f4                       005a:01f4
005f:01f4                       005d:01f4
003c:001e                       000d:0000
000e:0c06                       000f:080e
000a:007f                       000c:0036
0005:0700                       0006:0200
0000:0006                       0001:00000000
0008:0000                       0009:0000
01f2:031d                       0012:0000
0011:0090                       0010:0111
0007:0000                       0019:0000
002b:0000                       002c:0000
002a:65d4                       002f:01f4
002e:01f4                       0020:0000
0021:0000                       001f:46bc
001e:00f9                       001d:4498
[... 220 lines cut ...]
023e:0000                       003b:0000
05fa:01fe                       0059:01f4
0058:01f4                       0053:0430
0052:0430                       0057:01f4
0056:01f4                       0055:01f4
0054:01f4                       0208:03f2
005c:01f4                       005a:01f4
005f:01f4                       005d:01f4
003c:001e                       000d:0000
000e:0c13                       000f:080e
000a:0084                       000c:0036
0005:0700                       0006:0200
0000:0006                       0001:00000000
0008:0000                       0009:0000
01f2:0430                       0012:0000
0011:0090                       0010:0111
0007:0000                       0019:0000
002b:0000                       002c:0000
002a:65d4                       002f:01f4
002e:01f4                       0020:0000
0021:0000                       001f:46bc
001e:00f9                       001d:4498
001c:0031                       001b:454f
001a:0063                       
01f4:30303038353635333133

jgyates, 2017-08-14:
Can you send the following:

1) The output of the "generator: monitor" command
2) the contents of your /var/log/genmon.log file (or at least the last entries since you restarted the program)

The software may be detecting your controller incorrectly. Are you selecting a controller type in genmon.conf or do you let the software auto-detect? Register 0000 is used to detect the model, you have a value of 0006 in that register, which the software has never seen before. This value makes sense now that I see it, but I want to validate if you are specifying "evolutioncontroller=false" or if you are taking the default value in the configuration file and letting the software auto-detect.

JeanB51, 2017-08-14:
For the settings, I have:
liquidcooled=True
evolutioncontroller=False

For the outputs:

>generator:monitor
Monitor:

Genmon Stats:
    Monitor Health: OK
    Generator Type Selected : Nexus Controller, Liquid Cooled
    Generator Monitor running for 2 days, 3:17:30.

Serial Stats:
    PacketCount M: 5609690, S: 5609689, Buffer Count: 7
    CRC Errors : 0  Percent : 0.00
    Discarded  : 0  Restarts: 0  TimeOut: 0
    Packets per second: 60.76
    Average Transaction Time: 0.0316 sec

****And here is the genmon.log file** **
 more genmon.log
2017-08-08 22:59:16,853 : Error: timeout receiving slave packet for register 05c Buffer:0
2017-08-08 22:59:19,870 : Error: timeout receiving slave packet for register 01 Buffer:0
2017-08-08 22:59:22,878 : Error: timeout receiving slave packet for register 053 Buffer:0
2017-08-08 22:59:25,883 : Error: timeout receiving slave packet for register 052 Buffer:0
2017-08-08 22:59:28,925 : Error: timeout receiving slave packet for register 09 Buffer:0
2017-08-08 22:59:31,934 : Error: timeout receiving slave packet for register 5f1 Buffer:0
2017-08-08 22:59:34,938 : Error: timeout receiving slave packet for register 05a Buffer:0

And the previous, genmon.1

 more genmon.log.1
2017-08-08 22:59:13,852 : Error: timeout receiving slave packet for register 0d Buffer:0
2017-08-08 22:59:16,829 : Error in GeneratorDevice:MonitorThread integer division or modulo by zero
2017-08-08 22:59:16,869 : Error: timeout receiving slave packet for register 0e Buffer:0
2017-08-08 22:59:19,881 : Error: timeout receiving slave packet for register 0f Buffer:0
2017-08-08 22:59:21,846 : Error in GeneratorDevice:MonitorThread integer division or modulo by zero
2017-08-08 22:59:22,892 : Error: timeout receiving slave packet for register 0a Buffer:0
2017-08-08 22:59:25,904 : Error: timeout receiving slave packet for register 0c Buffer:0
2017-08-08 22:59:26,863 : Error in GeneratorDevice:MonitorThread integer division or modulo by zero
2017-08-08 22:59:28,915 : Error: timeout receiving slave packet for register 5f5 Buffer:0
2017-08-08 22:59:31,879 : Error in GeneratorDevice:MonitorThread integer division or modulo by zero
2017-08-08 22:59:31,924 : Error: timeout receiving slave packet for register 03c Buffer:0
2017-08-08 22:59:34,934 : Error: timeout receiving slave packet for register 1f2 Buffer:0
2017-08-08 22:59:36,893 : Error in GeneratorDevice:MonitorThread integer division or modulo by zero

jgyates, 2017-08-14:
Regarding the error "Error in GeneratorDevice:MonitorThread integer division or modulo by zero", to help narrow this error down, which of the following did you enable:  displaymonitor, displaystatus, displayregisters, displaymaintenance in genmon.conf? If you can't remember that is fine, these errors appear to have happened back on the 8th of August.

jgyates, 2017-08-14:
I think I found the divide by zero problem. If you can just let me know if you are using 2-wire start I can post an update. I have looked at the register output and I an not seeing a change that corresponds to the utility line voltage. The next option is to enable some additional debugging built into the program. In the genmon.conf file modify the line to set enabledebug = True, then reload the program. This option will send an email to you every 10 minutes with a list of registers that have changed. This option includes all registers, not just the ones that the program explicitly polls. Once this file is emailed to you by genmon, then post it here. You can attach a file to the comment to make it easier, the link to attach a file is below.

JeanB51, 2017-08-14:
For the display flags, I have displayoutput at false.
The following ones are at true , because they are documented as being ignored when the first one is false.
I had them True for my first few runs.

Except, displaymaintenance which is false.

JeanB51, 2017-08-14:
I think I have 2-wire start, although I wouldn't know how to verify that. But, the start seems to be commanded from the automatic transfer switch. The document for the switch says that it will send a start signal to the generator when it detects a power failure.
Also, it is from the transfer switch control panel that the weekly test run time is set.

Ok,. I will reload with the debug switch. Do you want only one sample of the file, or a bunch of them ?

jgyates, 2017-08-14:
Just one sample is fine.

Can you send the output of the maint and logs command?

Also, can you look at your logs (start/stop log and alarm log) from your controller and make sure the values match that are displayed in the app. From the main menu -> History. If there are any differences let me know the wording of the message. Are you getting a Two Wire start every time you start the generator?

Also please look at the "generator: maint" command and see if you can look at the menu in your controller and validate the engine run hours and version information matches what is displayed in the controller vs the application.  Run Hours is in the controller menu is located here:From the main menu -> Status -> Display.  The version info is located here: From the main menu -> Status - >versions (scroll between HW and Firmware versions)

jgyates, 2017-08-14:
What is the make/model of your switch?

JeanB51, 2017-08-14:
Switch : Generac QT-036 36KW water-cooled, propane-fuel.

I have attached on debug file

Here is the out put from the 2 commands. You can see from the start/stop log that all the weekly scheduled test runs are 2-wire. Manual runs are when I or the technician started it manually. 
The first entry is a manual run I did this afternoon just to see what the log entry
[genmon debug.txt](https://github.com/jgyates/genmon/files/1223759/genmon.debug.txt)

 would be.

Also you see in the maint command output that the exercice time is set for tuesday at 7:00. However, I just reset the weekly exercice time today, from the Transfer Switch, for mondays around 11:00. I think this has precedence.

I'll go look at the other stuff next.

generator:maint
Maintenance:

Generator Serial Number : 0008565313

Exercise:
    Exercise Time: Tuesday 07:00 Quite Mode Off

Service:
    Service Scheduled: Next service in 99 hours
    Total Engine Run Hours: 55
    Firmware : V2.12, Hardware : V1.01

>generator:logs
Logs:

Start Stop Log:
    08/14/17 17:33:30 Stopped - Auto
    08/14/17 17:33:19 Running - Manual
    08/14/17 12:40:32 Stopped - Auto
    08/14/17 12:24:10 Running - 2 Wire Start
    08/14/17 12:18:55 Stopped - Auto
    08/14/17 12:06:53 Running - 2 Wire Start
    07/31/17 10:39:42 Stopped - Auto
    07/31/17 10:35:49 Running - Manual
    07/31/17 10:35:04 Running - Manual
    07/31/17 10:33:57 Running - Manual
    07/31/17 10:31:41 Running - Manual
    07/31/17 09:39:12 Stopped - Auto
    07/10/17 17:19:39 Stopped - Auto
    06/20/17 11:07:53 Stopped - Auto
    06/20/17 11:07:28 Running - Manual
    06/20/17 11:04:27 Stopped - Auto
    06/20/17 11:04:08 Running - Manual
    06/20/17 10:58:07 Running - Manual
    06/20/17 10:54:49 Running - Manual
    06/20/17 10:22:24 Stopped - Auto
    06/20/17 10:10:14 Running - 2 Wire Start
    06/20/17 10:07:13 Stopped - Auto
    06/20/17 09:51:37 Running - 2 Wire Start
    06/20/17 09:42:41 Stopped - Auto
    06/20/17 09:40:20 Running - 2 Wire Start
    06/20/17 09:32:44 Stopped - Auto
    06/20/17 09:31:06 Running - 2 Wire Start
    06/20/17 09:27:48 Stopped - Auto
    06/20/17 09:12:44 Stopped - Auto
    06/11/17 18:15:06 Running - Manual
    06/10/17 18:20:15 Stopped - Auto
    06/10/17 18:03:53 Running - 2 Wire Start
    06/03/17 18:19:59 Stopped - Auto
    06/03/17 18:03:37 Running - 2 Wire Start
    05/27/17 18:19:40 Stopped - Auto
    05/27/17 18:03:18 Running - 2 Wire Start
    05/20/17 18:19:22 Stopped - Auto
    05/20/17 18:03:00 Running - 2 Wire Start
    05/19/17 07:01:05 Stopped - Auto
    05/18/17 23:04:33 Running - 2 Wire Start
    05/13/17 17:19:04 Stopped - Auto
    05/13/17 17:02:42 Running - 2 Wire Start
    05/06/17 17:18:46 Stopped - Auto
    05/06/17 17:02:24 Running - 2 Wire Start
    04/29/17 16:18:26 Stopped - Auto
    04/29/17 16:02:05 Running - 2 Wire Start
    04/22/17 16:18:07 Stopped - Auto
    04/22/17 16:01:46 Running - 2 Wire Start
    04/15/17 16:17:49 Stopped - Auto
    04/15/17 16:01:27 Running - 2 Wire Start
Alarm Log:
    07/31/17 10:35:24 Low Oil Pressure
    07/31/17 10:33:35 Low Cooling Fluid
    06/20/17 10:58:36 RPM Sense Loss
    06/11/17 18:09:28 RPM Sense Loss
    06/10/17 18:20:15 Inspect Air Filter
    06/03/17 18:19:59 Inspect Air Filter
    05/27/17 18:19:40 Inspect Air Filter
    05/20/17 18:19:22 Inspect Air Filter
    05/19/17 07:01:05 Inspect Air Filter
    05/13/17 17:19:04 Inspect Air Filter
    05/06/17 17:18:46 Inspect Air Filter
    04/29/17 16:18:26 Inspect Air Filter
    04/22/17 16:18:07 Inspect Air Filter
    04/15/17 16:17:49 Inspect Air Filter
    04/08/17 16:17:31 Inspect Air Filter
    04/01/17 16:17:09 Inspect Air Filter
    03/25/17 16:16:50 Inspect Air Filter
    03/18/17 16:16:31 Inspect Air Filter
    03/11/17 16:16:12 Inspect Air Filter
    03/04/17 16:15:53 Inspect Air Filter
    02/25/17 16:15:32 Inspect Air Filter
    02/18/17 16:15:13 Inspect Air Filter
    02/11/17 16:14:54 Inspect Air Filter
    02/04/17 16:14:36 Inspect Air Filter
    02/04/17 00:38:02 Inspect Air Filter
    07/30/16 23:29:10 Inspect Air Filter
    07/23/16 23:28:53 Inspect Air Filter
    07/16/16 23:28:37 Inspect Air Filter
    07/14/16 18:40:46 Inspect Air Filter
    07/09/16 22:28:19 Inspect Air Filter
    07/02/16 22:28:02 Inspect Air Filter
    06/25/16 22:27:45 Inspect Air Filter
    06/18/16 22:27:27 Inspect Air Filter
    06/11/16 22:27:10 Inspect Air Filter
    06/04/16 22:26:52 Inspect Air Filter
    05/28/16 22:26:35 Inspect Air Filter
    05/21/16 22:26:18 Inspect Air Filter
    05/14/16 22:25:59 Inspect Air Filter
    05/07/16 22:25:40 Inspect Air Filter
    04/30/16 22:25:21 Inspect Air Filter
    04/23/16 22:25:01 Inspect Air Filter
    04/16/16 22:24:42 Inspect Air Filter
    04/09/16 22:24:23 Inspect Air Filter
    04/02/16 22:24:03 Inspect Air Filter
    04/01/16 04:15:02 Inspect Air Filter
    09/26/15 19:05:15 Service Schedule A
    09/19/15 19:04:57 Service Schedule A
    09/12/15 19:04:40 Service Schedule A
    09/05/15 19:04:23 Service Schedule A
    08/29/15 19:04:06 Service Schedule A

JeanB51, 2017-08-15:
Ok, here is what I saw on the controller

- All the alarm log messages are OK.

- The start/stop logs are the same too. The only difference, is where the monitor says "2-wire start", my controller says "Remote Start". Although this is my interpretation, as my controller is set to French - I did not want to mess with the setup to try setting to English. But Remote Start makes sense to me..

-Hardware/Firmware versions are 1.01 / 2.12, as genmon reports.
- Run Hours - 55.2 - OK
- RPM = 0; HZ = 0, Battery = 14V.

There you are.

jgyates, 2017-08-15:
Generac appeared to change their labeling from Nexus to Evolution, remote and wire are essentially the same functionality (starting the generator from an external source). Two wire start is starting the generator by energising two wires from something external to the controller (like a transfer switch or something else), the "Remote Start" in the software or sometimes called Radio start is starting the generator via modbus commands over the serial interface. 

I have a theory of what may be the problem with the display of the utility voltage. It sounds like your transfer switch is making the decision regarding starting the generator if utility power is lost, hence the two wire start and the ability to set exercise times from the switch vs the generator controller. To contrast, my transfer switch only switches if a signal from the generator tells it to so the generator controller is monitoring the utility line voltage. My guess is that the utility lines are not connected to your controller so your controller will always read zero. Your transfer switch is monitoring the utility voltage and telling the generator to start so there is no need for the generator to monitor the line voltage. To validate this look in your controller menu, Main Menu -> Debug -> Inputs. From this screen you should see the utility voltage listed. My guess is that it is zero. If it is then this is all we can do since the controller can not read the utility voltage. If that is the case let me know and I can add an option to disable the checking of the utility voltage. If you do see a non zero voltage on the debug inputs screen let me know and we can go from there.

jgyates, 2017-08-15:
I uploaded an updated version of genmon.py and genmon.conf. You will need to uncomment and modify the new option disableoutagecheck in the conf file if it turns out that the utility voltage is not wired to your controller. Let me know once you get a chance to test. I also added code to handle the false alarm email. Your should get an email that states "two wire stop" occurred.

Let me know how your testing goes and if you have any other issues. Thanks for the help in testing.
-Jason

JeanB51, 2017-08-15:
Hi Jason

Your theory makes a lot of sense. The generator was added after the house was built, so they did not run the mains into it. The Transfer Switch is doing all the logistics. I can see that that all the generator has, is a regular 110V plug, for the block heater, to keep the battery charged, and to power the controller.

I will try the updated version, and keep you posted. Many thanks to you for the excellent support. As an open-source project, you beat a lot of paid-for software that I can think of !

An idea I had for an extra feature - a kind "I'm alive" signal: My generator is at my country house. This why I was looking for this monitoring solution. But then, if I don't hear from it for a long time, is it still running ? I know I can always email it to check, but it would be reassuring if it sent me an email periodically (once a week?) to tell me it is still up and running.
Just a suggestion.

JeanB51, 2017-08-15:
I just tested the updated version, and it is fine - no more warning of an outage.
I'll check the Alarm on stopping next time I do a run.
Thanks
JB

jgyates, 2017-08-15:
That is good, thanks for the feedback. Probably the best way to send a periodic email is via cron job. This link shows how to do that:

https://askubuntu.com/questions/332481/setup-cron-job-to-send-email-to-a-bunch-of-people-on-a-regular-basis

With this you can send an email with the status command at an interval of your choice.

I will close this issue but feel free to reopen it if you see any issues in your testing.

Thanks
Jason

closed 2017-08-15 by jgyates

jgyates, 2018-02-01:
@JeanB51 

If you have a moment can you look at this thread:

https://github.com/jgyates/genmon/issues/68

It would be helpful to get another data point from someone with a NExus Liquid cooled unit.

Thanks
Jason

JeanB51, 2018-02-01:
Hi Jason

Yes, I saw that. I am outside the country right now, but I will test this as soon as I get back. 

I’ll keep you posted. 

Jean

JeanB51, 2018-02-16:
Hi Jason 

I will be at my generator site tomorrow. 

How should proceed to test this?

I am still running last August version of the software. 

Jean

jgyates, 2018-02-16:
Hi Jean,

What I would recommend is:

1) Update to the latest software (https://github.com/jgyates/genmon/wiki/Updating-to-the-Latest-Version)

2) Start your generator (you can do this either manually or via the remote start feature)

3) Once the generator is running, make a note of the Frequency. Is this 60Hz or 120Hz. 

I had one person report that the frequency was 30Hz for a Nexus Liquid Cooled model (it should be 60Hz) and I changed the code to compensate. After looking at the past history of this thread it looks like you have 30Hz also so this fix should correct that for you. If it is 60Hz, we are all set, if not let me know.
FYI, the thread that discusses this is here: https://github.com/jgyates/genmon/issues/68

Thanks
Jason

JeanB51, 2018-02-19:
>    genmon/wiki/Updating-to-the-Latest-Version

Hi Jason

I ran the test this weekend, and everything seemed to run well.

- 1800 RPM
- 60Hz
- 240 V

I'll post the info on the thread.

Again thanks for the ongoing support on this project.

Jean