jgyates/genmon #36: Wondering if any genmon folks are using Indigo home automation

jeepjungle, 2017-08-27:
Greetings,
Been wanting to tie my Generac 36kw Nexus water cooled unit into my home network for some time and genmon is just the thing I need.  All my parts are ordered and I will be building it soon.

In the mean time, I'm wondering if anyone else here is uses Indigo home automation software.  I have this already setup and want to tie genmon into it and am seeking collaborators.

Thanks,
Rob

jgyates, 2017-08-27:
Hi Rob,

I have not heard of anyone integrating genmon with Indigo, but I feel sure it would be do-able. genmon.py interfaces with all other modules via a socket interface and the functionality of the command interface is already encapsulated in the ClientInterface.py library. I have a version of this library written in java that has not fully been tested. I do not use Indigo, but I could provide some insight regarding genmon to anyone who wanted to take this on. Personally I use nagios to monitor my network and I wrote a nagios plugin to let me know when the generator is in alarm, but I can see why someone would want to integrate this with a full home automation system.
-Jason

jeepjungle, 2017-09-06:
Jason,

I appreciate the response.  I will post on the Indigo forums to see if anyone there has backup generators and are interested in genmon!  Indigo is very flexible and extensible, more so than my Python skillz, so I will need to figure out the simplest way to do the interface.

Thanks,
Rob

jgyates, 2018-02-21:
Closing this thread for now. Feel free to comment if you have any other data points.

closed 2018-02-21 by jgyates