jgyates/genmon #37: Unknown Alarm

JeremyStevenson, 2017-08-28:
Got the following unknown alarm. I cleared the "check battery" and it went away. (Nice note in this email to do that!)

Time: 08:02:12 PM
Date: Sun 27-Aug-2017

Notice from Generator:

Switch State: System in Alarm
Engine State: Cranking in Alarm

Current Alarm: UNKNOWN ALARM: 00020014
System In Alarm: 0001:00020014

Last Log Entries:
    Start Stop Log: 08/23/17 20:31:35 Stopped - Auto
    Alarm Log:      08/23/17 20:31:35 Check Battery

To clear the Alarm/Warning message, press OFF on the control panel keypad followed by the ENTER key.

jgyates, 2017-08-28:
Hi Jeremy,
I just uploaded a new version of genmon.py that added this to the alarm logic. Thanks! Let me know if this does not fix it the next time a check battery alarm occurs.

-Jason

closed 2017-08-28 by jgyates

jgyates, 2019-01-15:
I updated Genmon to version 1.12.11 to include this alarm condition.

JeanB51, 2019-01-15:
Thanks for this, and the overall great support

Jean