jgyates/genmon #39: Superb project

Barrygordon, 2017-09-08:
I have a 22 KW guardian air cooled with the Evolution controller.  An RPI runs my entire home and is coded in Node.js. I must complement you on the degree of detail in your documentation. I am retired and spent over 50 years in the digital world.  I have never played with Python, now doing all my work in Node.js and Javascript.  My goal would be to have your project inform my RPi automation controller of all changes that occur.  The automation controller has full voice capability and can broadcast to the in-wall mounted iPads which are the basic HID in my home. Being a little lazy is there any chance a node.js version of the project exists? Have you thought about packaging this up as a product?  Generac charges almost $300 + fees for their version.  I will begin looking into this after Hurricane IRMA leaves FL. I would like to open a private dialog with you if that is all right. My name is Barry and I own the domain "the-gordons.net" so email should be obvious. If you would like to supply me with  a python version I could use I would be more than willing to pay for it as I would have to spend $300 with Generac. The version I would want would have no human interface but rather send out UDP broadcasts or TCP messages to a port (UDP) or to a specific TCP (the RPI automation controller). I use a simple UDP message protocol that makes things very easy and all devices in my home use it.  The protocol uses sequence numbered messages with no acknowledgement, messages are repeated N times with the same sequence number.  Very reliable in my environment
 Naturally I will contribute what I can, and if necessary start with what you have done and modify it to do what I need.

Great Job!!!!

jgyates, 2017-09-09:
Hi Barry,

I sent you and email. If you did not get it feel free to reopen this thread.

Thanks
Jason

Barrygordon, 2017-09-09:
Got it.  Perfect.

closed 2017-09-09 by jgyates