jgyates/genmon #40: Alternate Hardware Platforms

jgyates, 2017-09-10:
This thread is for people using non-Raspberry Pi based hardware. I would like to collect information about proven alternate hardware for use with this project so others can benefit. If you are using non-Pi hardware please post a link to the platform and if possible a picture of your final solution.

Thanks
Jason

jeepjungle, 2017-09-13:
This could likely be duplicated fairly easily with a Beagle Bone board.  They are very similar to the Pi and have Debian distributions available.  If anyone is thinking about that, I'd say go for it

LynnSchott, 2017-09-16:
I Found another board, but the setback is the board is running MicroPython. Which means that the application you have would have to be modified to Python 3.x to run off it.

http://emacinc.com/products/pc_compatible_sbcs/IOT-F407M
MitiPy run’s micropython 1.9 which is a Python 3 based program.

jgyates, 2017-09-16:
Currently genmon.py has not been ported to python3 so this would require some work before it would work.

jgyates, 2017-09-19:
@LynnSchott as a FYI, I have completed a quick and dirty port to python3. More testing is needed but all the basic functionality has been validated.

ColtB45, 2017-11-15:
@jgyates Thanks for creating and dedicating your time to this great project.

I built the first one for a friend with a Pi2. It works great. I have ordered a couple Orange Pi Zeros (512MB) to make two more for two other hams. (I don't actually have a generator to use it with, but I may have to get one now.)

The Orange Pi Zero 512 specs
https://www.board-db.org/product/139/

I expect they will work well. They are almost as fast as Pi3's with half the RAM (same as a zero). The real attraction is a detachable WiFi antenna and Ethernet on the same board at a lower price point than a pi zero w shipped. (They are just under $13 including shipping on banggood & aliexpress.)

I'll update here after I test one and detail any changes in config or code (unlikely) as well as pics of the HW config.

-ColtB45

jgyates, 2017-11-15:
@ColtB45, thanks for the kind feedback. The Orange Pi does look nice. Let me know how it works. The detachable WiFi antenna is a great feature. I use a USB WiFi adapter because it has an external antenna connection. At time the Pi does not play well with two WiFi adapters so this would be a good alternative.

ColtB45, 2017-11-21:
Ok, so I got the Orange Pi's in. armbian is a straightforward ubuntu/dabian based fork. Easy :)
The WiFi seems to be about as good as a Pi Zero with the stoke dipole antenna. The WiFi chipset is known to have problems with newer kernels so I am testing with the 'legacy' kernel.

Here is the blow-by-blow to replicate what I've done.

1. Download the legacy kernel distribution of armbian from here: https://www.armbian.com/orange-pi-zero/
2. Write the image to a sdcard and put it in the Orange Pi
3. Find the IP address and SSH in (user 'root', password '1234')
4. Follow prompts to change the root password.
5. Follow prompts to create a admin user account. Make sure you name the user pi for simplicity.
6. Run the following and set the timezone following the prompts. If you need to configure WiFi do that here as well.
`sudo armbian-config`
7. Run the following to disable the GPU, HDMI, and USB ports. (Unlike on the RPi, this has no effect on eth0 and wlan0)
`sudo h3consumption -g off -u off`
8. Reboot to make the changes take effect.
`sudo reboot now`
9. Install the following:
`sudo apt-get update && sudo apt-get install git python-pip python-serial`
10. The version of pip installed is outdated, perform an upgrade with:
`sudo pip install --upgrade pip`
11. Install setuptools for pip:
`sudo pip install setuptools`
12. Install the folowing:
`sudo pip install crcmod pytz configparser Flask`
13. Test crcmod:
`python -m crcmod.test`
14. Checkout genmon via git:
`git clone https://github.com/jgyates/genmon.git`
15. (optional, but advised) To loopback test the serial port;
        a. install a loop between UART1_RX/PG07 & UART1_TX/PG06 (as identified here: https://i1.wp.com/oshlab.com/wp-content/uploads/2016/11/Orange-Pi-Zero-Pinout-banner2.jpg):
        b. run the serial port test program:
        `python genmon/OtherApps/serialtest.py /dev/ttyS1`
16. Copy the config files to /etc/:
`sudo cp genmon/*.conf /etc`
17. Edit the config files being sure to set the serial port to: port = /dev/ttyS1:
`sudo nano /etc/genmon.conf`
18. Configure other .conf files as desired
19. Make startgenmon.sh excutable:
`sudo chmod a+rx genmon/startgenmon.sh`
20. Setup crontab to run genmon:
`sudo crontab -e`
        a. select nano
        b. Add the following to the bottom of the file:
        `@reboot sleep 30 && sh /home/pi/genmon/startgenmon.sh start`
21. Shutdown the Orange Pi with:
`sudo shutdown -P now`
22. Go install on the generator controller with an RS232-TTL converter.

jgyates, 2017-11-22:
Very nice, thanks @ColtB45!

cron410, 2019-01-11:
I'm currently testing with an Odroid C1+. Using a TTY to RS232 adapter with 4 pins on the 40 pin header, I have successfully tested the serial port. Now I just need to brave the cold and go hook it up to a generator. Using Odroid supplied Ubuntu 18.04, the serial port is /dev/ttyS2

ColtB45, 2019-01-11:
@cron410 I see no reason that wouldn't work. I would point out that has a LOT more horsepower than needed for genmon.

jgyates, 2019-01-11:
Sounds good, let us know how it goes. I don't see any issues.

daveh85, 2019-03-06:
Anyone able to comment on whether this will work with this board?

https://coral.withgoogle.com/tutorials/devboard/

cron410, 2019-03-06:
@daveh85 that would be a waste of the AI coprocessor as it won't be useful for genmon. Pretty much anything that runs Python and has a serial port. Either RS232 UART, USB RS232, or GPIO pins will work fine. The other issue is the 0 - 50°C temperature range. Yes most devices will operate outside their listed range, but I do not know how sensitive this board is.

ColtB45, 2019-03-06:
> @daveh85 that would be a waste of the AI coprocessor as it won't be useful for genmon. Pretty much anything that runs Python and has a serial port. Either RS232 UART, USB RS232, or GPIO pins will [...]

+1, complete waste of HW for genmon.

I went to the Orange Pi because it is dirt cheap and has Ethernet in addition to WiFi.

simslim2000, 2019-07-15:
I am actually running this as a VM on a vSphere cluster we have at the house (GenMon OS: Ubuntu Server 18.04 LTS ).  I have a USR-W610 serial gateway setup for comms, works great once I figured out the serial port part of it in the vm.

pridesleap, 2020-06-15:
Not sure where to post this. I'm new to GenMon, coming upto speed on Linux, and just started learning Python. My background is in industrial automation and PLC's.

I would prefer to not use a Pi in the generator compartment due to environemntal concerns (cold and damp where the generator is located). I plan on installing a serial to ethernet converter to get the generator on the local network and then running GenMon of a linux server on the local network.

My first question concerns installing GenMon on the Linux box. Given i'm still a bit new to linux, how do I install it? From what Ive been able to gather, I need to:

**Install Git:**
   sudo apt-get update
   sudo apt-get install git

**Doawnload Project Files:**
   git clone https://github.com/jgyates/genmon.git

**Run Install Script:**
   cd genmon
   bash ./genmonmaint.sh -i

Reboot and nvaigate to web interface.

QUESTION #1 - Am I correct in my assessment of the workflow above? Any suggestions would also be appreciated.

jgyates, 2020-06-15:
@pridesleap 

Yes, you are on the right track. 

The instructions here are up to date.

https://github.com/jgyates/genmon/wiki/3.3--Setup-genmon-software

If you have specific questions feel free to create a new issue thread rather than post to a thread that does not match the questions. This way it is easier for others to search past closed issues and find information on topics of interest. Most of the past issues threads are categorized so you can also search them by category or keyword. Of course you can create a new thread if you don't find the answer anywhere else.

liltux, 2020-06-15:
That looks about right.  May have to add sudo to the bash genmonmaint.sh.  Is this for your home or business?

pridesleap, 2020-06-15:
> If you have specific questions feel free to create a new issue thread rather than post to a thread that does not match the questions.

@jgyates - Thanks for the clarification. I'll create a new thread in the future.

>  Is this for your home or business?

@liltux - Just my home.

liltux, 2020-06-15:
@pridesleap on a home private network you should be ok.  I have done a couple for businesses, like a bank.  I ended up turning off ssh login, created ssh keys as back up, created a new user, installed a firewall to block all ports not needed.  Just some extra ideas if you are needing more security with the PI.

pridesleap, 2020-06-15:
@liltux I've enabled the firewall on the server and have only one port open for SSH, and it's restricted to the local subnet only. If you think of anything else let me know. I'm open to rsuggestions.

liltux, 2020-06-16:
That all looks good.  If security is an issue I would setup secure ssh keys and disable password login.  But on a private network you should be ok.

closed 2020-09-29 by jgyates