jgyates/genmon #41: Low profile WIFI antenna

jeepjungle, 2017-09-13:
I duplicated most all of jgyates original hardware design but for my situation the whip antenna was not the best option and looked for an antenna that was a little stealthier and low profile.  My generator is only 6 feet from the back of my house and about 45 feet from my WIFI access point.  The antenna I used is a lower gain (only 2db) but is just enough to get the signal from the generator to the access point.  Just mentioning it for anyone with similar requirement.
[image]

jgyates, 2017-09-14:
I have purchased antennas from this source before: https://www.data-alliance.net/
They have a good variety and pricing. I am using an external antenna also with a USB WiFi adapter. My generator is about 50 feet from the house, which is brick so wifi does not propagate that well through brick, hence the larger external antenna. I started with one like this:

https://www.data-alliance.net/antenna-2-4ghz-5dbi-weatherproof-directional-ip67-cable-to-rp-sma-or-u-fl-thru-hole-mount/

Then moved to the larger omni-directional. The uni-directional one would of been great, but I was trying to line up my signal with a window (better propagation through glass than brick), so I went with an omnidirectional.

jeepjungle, 2017-09-14:
That is an awesome site!  I could of saved some coin ordering from there.  I have it book marked now.  Will come in handy.

closed 2018-03-16 by jgyates