jgyates/genmon #43: Generac R200 controllers

ebzimny01, 2017-09-22:
Starting new thread about R200. Thank you for your last reply. I am aware of the Omnimetrix and the iGenerator products. Omnimetrix I believe is several hundred dollars plus monitoring fees. And iGenerator is $450, hence the reason I was interested in attempting to avoid those costs. My other searches for a DIY option with a Pi came up with ideas that seem similar to the iGenerator approach - which appears to be just basic monitoring of the voltage / load at the transfer switch. What I am most interested in knowing is that my generator weekly test ran as expected, and then possibly log/notify of the times that main power was lost and was switched over to generator.

jgyates, 2017-09-23:
Based soley on the information in [this](https://www.omnimetrix.net/wp-content/uploads/2017/03/installation_Generac_R100-R200_Residential.pdf) document, you may have some luck with a PI based solution that uses GPIP inputs to monitor the pins listed in the document. The picture of the cable looks like the end of the cable that connects to the generator are actually terminal style connectors that screw to the output of a signal for specific events or alarms. One thing that can help is to measure these with a multi-meter, while you disconnect some of the sensors. Not sure what is available out there for the R200, but searching for a service manual would be a big help in figuring out the best way to craft a Pi based solution for this generator.

jgyates, 2017-09-23:
This should help: [Controller Tech Manual for R200](http://soa.generac.com/manuals/5134155/0F9784)

tracerrx, 2017-10-23:
I'm also interested in getting this to work with an R200 controller, anyone try yet?

Lifeischoice, 2017-12-19:
I am interested as well..thanks

Lifeischoice, 2017-12-30:
For R200 Controllers, this may be a good possibility to [replicate](http://tyconsystems.com/index.php/products/tycon-power/tpdin-monitor-web/751-tpdin-monitor-web2) the functions of Omnimetrix  without monitoring fees!

jgyates, 2018-02-21:
I am closing this thread. Feel free to comment if you have any other data points to add. I will re-open the thread if there is more activity.

closed 2018-02-21 by jgyates