jgyates/genmon #45: Start and Transfer Process

sgtmaj27, 2017-10-07:
I have been successfully using your software on a Generac 5870 with a Nexus controller since June.  I have tried all of the functions successfully except the Start and Transfer...only because I am not sure how it will behave.  Will this function work just like the generator does for lost/restored utility power ie start the generator, transfer power, run until I select the stop function.  Then when stopping the generator, restore power, waits one minute for cool down, then shut generator off? 

Thank you for a GREAT Project!

jgyates, 2017-10-07:
Hi @sgtmaj27 

Yes, the start/transfer functionality will start the generator and activate the transfer switch so the generator can power the load instead of the utility line. Note that this may not work if you have a "smart switch". In most cases the generator makes the determination that the generator should run, with a smart switch the transfer switch monitors the utility voltage and make the determination to start the generator. 

Once the command is issued, the generator will start, and warm up for a short period of time (5 -10 seconds depending on the internal settings of the controller), then instruct the transfer switch to turn over. The stop command will undo this process and switch things back to normal operation (house powered by utility). Let me know if you have any other questions.

Thanks,
Jason

sgtmaj27, 2017-10-07:
Thanks for the response.  My transfer switch has no logic ckt, it is
controlled by the generator.  I will be using this function once a month
from now on to exercise my generator and ATS.

sgtmaj27, 2017-10-08:
I did the start and transfer function on my Generac 5870 Nexus today.  I t went perfectly.  Stated, transferred, (with no interruption of power to the protected ckts).  At shutdown it switched back to utility power, cooled down and shut off.  Received emails for each process.  So much better than having to turn off the utility power to test the transfer function.  To ensure that he generator recognizes losing utility power, I turned off the ckt breaker that feeds the transfer panel from utility power while the test was in progress.  A few minutes later, I received an email for loss of utility power.

You are a rock star, I can’t thank you enough for his project!

closed 2017-10-08 by sgtmaj27

jgyates, 2017-10-08:
Thanks! I am glad it is working for you. I appreciate the feedback, and thanks for the donation