jgyates/genmon #46: Start genmon.sh

rhombixt, 2017-10-22:
Having issues with trying to get the start genmon.sh script to run on a reboot. I tried putting in crontab but it will not run. I can run it from command line by putting the bash command in front. Forums are saying that you need the #!bin/bash as the first line of the script or it will not run in crontab. Can you provide some input to get the .sh to run on power up?

jgyates, 2017-10-22:
Hi @rhombixt 

The main requirement is that you must specify the full path of the python modules in the script. The script provided in the repository assumes you install everything in the /home/pi/genmon directory. If you use another directory you will need to modify the script to match your path.

To get startgenmon.sh to start every boot you need to add it as a cron job. The last couple of paragraphs on this page describe how to do that:

https://github.com/jgyates/genmon/wiki/Setting-up-your-Pi-and-genmon

Let me know if this does not solve your problem.

Thanks
Jason

jgyates, 2017-10-22:
Also, dont forget to change the permissions of startgenmon.sh so the file is executable.

     sudo chmod 775 startgenmon.sh

rhombixt, 2017-10-22:
I get permission denied using the ./startgenmon.sh start command.

Crontab is not working either

jgyates, 2017-10-22:
what does the output of this command look like:

     ls -la startgenmon.sh

also what directory is the file located ?

rhombixt, 2017-10-22:
I got it running - I was powering down and up .. I guess the pi does not consider that a reboot?

jgyates, 2017-10-22:
great, I will close this issue

closed 2017-10-22 by jgyates

rhombixt, 2017-10-22:
My error on the ./startgenmon.sh - I was not running it from the genmon directory..