jgyates/genmon #47: Vlan question

rhombixt, 2017-10-22:
I have multiple vlans.. the website that runs the data on port 8000 .. if I use a pc browser on the same subnet I can bring up the window and I can see the request on the client log window. When I try to open it via a browser from another vlan..it does not display anything . I can ping the device no problem and the client window shows no log for the event.. it’s almost as if it was listing for requests on just one subnet..

jgyates, 2017-10-22:
sorry, I am not sure what your question is...can you tell me specifically what is not working? If you are describing that you can not access the web interfacde from a separate VLAN then I would look at how you are routing your VLANs. This is dependant on your routing configuration. The software is agnostic of VLANs or subnets, it only assumes that Linux is setup properly. I would guess that ICMP (ping) packets are routed by default in your VLAN setup and other traffic is not. From what I think your are describing (not being able to access the web interface from a separate VLAN, this is the extra security that a VLAN provides, unless you specifically allow an exception in you VLAN routing.

rhombixt, 2017-10-22:
I can ping the pi from the other vlan no problem, when I try to open the web page from the other vlan, the web page does not respond. I can also vnc from the other subnet to the pi so I know the network is fine . The client interface also does not show a request from the ip from the other vlan like it does for a pc on its native subnet. It’s like the web server is only listing and acknowledges requests from one subnet. The unit is 192.168.110.45 .. PCs on .110 have no issue with the page, but a server on 192.168.130.20 subnet cannot open the page but can vnc to the pi.
The

jgyates, 2017-10-22:
Your VLAN is likely blocking non standard ports. You can either tell your  Router or Switch (which ever is enforcing your VLAN rules to allow port 8000 (the port that genserv.py uses, or you can modify line 73 in genserv.py.

To change the port that genserv.py uses change this:

      app.run(host="0.0.0.0", port=8000)

to this:

       app.run(host="0.0.0.0", port=80)

If this is your problem you may also run into issues running ClientInterface.py as it uses a socket number that is not commonly uses. Once you validate this is your problem I would test ClientInterface.py and see if you also have a problem. If you do then you can either modify your VLAN to allow port 9082 or you can modify genmon.conf:

Change this line to the port that you want to use

     server_port = 9082

and then change line 15 in myclient.py to match the port number you choose.

Select a port number to something that your VLAN will recognize (not 80 since that is in use by genserv.py), I don't know what you have enabled on your pi so the answer here is going to be dependant on your system.

This link has a list of well know ports that are usually open. You may try port 389 (LDAP).
http://www.pearsonitcertification.com/articles/article.aspx?p=1868080

The problem with using ports that are reserved for other apps is that it may cause problems if you try to use those apps (only one app can use a TCP or UDP port)

Let me know if this is your problem.

rhombixt, 2017-10-22:
Instead of changing the code, I setup a simple windows 7 pro machine with IIS - web page on port 8000. Works like a charm. All VLANS can open the web page on port 8000 using the same IP and the genmon

jgyates, 2017-10-22:
Can you connect ClientInterface.py across your vlan? This will let me know if the problem is with the flask server or your vlan. The IIS data point is interesting, but if the limitation is with the flask server then I do not have any control over that. The genserv.py app tells the flask server what page to serve, but the actual serving of the page is handled by flask, not the genserv.py code.

rhombixt, 2017-10-22:
I don't have any other pi controllers here so I can't install the .py on anything else. Maybe ill just move the unit to the server vlan and leave it there vs trying to figure this one out. It might be a flask issue like you said which is not really under your control.

jgyates, 2017-10-22:
Ok. Also, you can run ClientInterface.py on a windows or macOS system as as long as it has python installed. macOS has it installed by default. For windows you will need to install python 2.7 from here:

https://www.python.org/downloads/windows/

jgyates, 2017-10-22:
Also, if ClientInterface.py works from a windows machine (i.e. port 9082 is not blocked on your vlan), you can also run genserv.py from windows as well. All of the apps except genmon.py can be run from another machine since the communicate with genmon.py via a socket connection.

jgyates, 2017-10-22:
Correction, all the apps except genmon.py, gengpio.py and gengpioin.py can be run on a separate system. You would need to install the python Flask library on windows to run genserv.py

closed 2017-11-05 by jgyates