jgyates/genmon #48: DST clock change

krisasmith, 2017-11-08:
Not an issue, more of a feature request. Would be nice if genmon automatically changed the clock for DST.

jgyates, 2017-11-08:
This is a great idea, let me think on this and find an optimal way to implement...right now I am thinking of detecting the DST change of the linux system and then instructing the generator to sync to the linux system time. I don't want to get into detecting the DST for the linux system as there are already a lot of things to do that built into linux.

krisasmith, 2017-11-08:
Maybe also too an option to keep the generator clock in sync with the Linux clock (options to sync daily, weekly, monthly, etc.) I notice over a number of days it skews slightly.

jgyates, 2017-11-08:
ok. That would be even easier to implement, just add a timer that sync the time every hour.

One of the problems with the time that I noticed is that there is no (known) way to set the seconds, only day, hour and min. What the code does, in an effort to be consistent when setting the time, is I create a separate thread that waits until the system time changes to a new minute, then I set the generator time. This way the offset in seconds between the generator and the linux system is consistent, or at least that was the attempt...

Also, I always use NTP to have my time set on my linux system to the time at time.nist.gov or some other NTP server.

jgyates, 2017-11-08:
Try out the latest code. I just added the following two options to genmon.conf:

# (optional) This parameter will sync the generator time to the system time
# if there is a change in Daylight Savings Time. This does not change the system
# time, only the generator time.
#syncdst = True

# (optional) This parameter will sync the generator time to the system time
# once every hour. This does not change the system time, only the generator
# time.
#synctime = True

uncomment the option to enable it.

Also, you will need to install one additional library (a library to parse timezone information called pytz). To install type:  

    sudo pip install pytz

I have tested this out on my system and it works for me.

I also updated the docs. I will leave this issue open for a few days or until you have a chance to test. Let me know if you have any issues of if you think any changes are needed.

@krisasmith, Thanks for your help and feedback!!
Jason

krisasmith, 2017-11-08:
WOW, what took you so long! :)

closed 2017-11-13 by jgyates