jgyates/genmon #49: New Feature: Monthly, Biweekly, Weekly Exercise

jgyates, 2017-11-08:
I can add the monthly, weekly, etc setting, however the implementation is different for the four variants of the firmware (air cooled Nexus, air cooled Evolution, liquid cooled Nexus, liquid cooled Evolution). I would need one person with each variant (I already have a liquid cooled Evolution) to send me a few register dumps after setting the weekly,biweekly, monthly option via the controller interface, from there I should be able to determine what the settings would be. If there is interest and some volunteers we can start a new thread on this topic.

Background:
There are two types of registers in the controller(s): direct and indexed. If this setting is in the direct registers then it should be easy, if this setting is in the indexed registers then this will be harder to find. I can easily poll the direct registers, however the index registers appear to be write only so finding these would require a more indepth level of debugging.

As a point of reference, my generator (liquid cooled Evolution) does not have a setting for weekly, biweekly, monthly exercise.

If you want to volunteer to help provide data points, respond to this thread and let me know your model.

griffths, 2017-11-08:
I have a liquid cooled Nexus unit and would be happy to provide data points however I have not finished building my system yet so won’t be able to provide the data until next month or so.

 

Regards,

 

Jay

jgyates, 2017-11-08:
Sounds good. Once you are at a place where you have genmon running do the following and repost your findings here:

use the "registers" command (either in email or via the ClientInterface.py program) to take a snapshot of your registers. Ideally we will want a snapshot of the following conditions:

Weekly option is set
Biweekly option is set
Monthly option is set

The comparing the registers should tell us which bit in the registers should be changed to enable each option. Then I can update the software and you can try it out. Please note there is a chance that this option is set via the indirect registers or another register that the software is not currently monitoring. We have some options if that is the case, but let's start with the easy stuff first. 

Also, if anyone else wants to post information for another generator model feel free. At the moment I have one volunteer for a liquid coolded Nexus, @griffths. I have the liquid cooled Evolution covered, so any volunteers for an air cooled Nexus or Evolution would be appreciated. 

Also, please check that your model has this option (weekly, bi-weekly, monthly exercise), not all models and firmware versions support this.
 
Thanks
Jason

raymondjclark, 2017-11-08:
Generator 7033, Evolution Controller
Exercise set to 1st of each month

Num Regs: 208
Not Changed: 1831930 Changed 30171 Total Changed 0.02
0038:0000 0034:0000
0037:0f78 0032:4000
05f5:0000 05f4:0000
05f1:0000 01f1:0000
05ed:0064 023b:0000
023e:0000 003b:0000
05fa:0000 0059:0000
0058:0000 0053:0000
0052:0000 0057:0000
0056:0000 0055:0000
0054:0000 0208:03ec
005c:0000 005a:0000
005f:0000 005d:0000
003c:0000 000d:0000
000e:071f 000f:0b08
000a:0094 000c:0000
0005:0e00 0006:0000
0000:0009 0001:00000000
0008:0000 0009:00ed
01f2:0000 0012:0000
0011:009c 0010:0311
0007:0000 0019:000f
002b:0005 002c:0e00
002a:6a76 002f:0000
002e:0001 0020:0000
0021:0000 001f:49e0
001e:018f 001d:0000
001c:0000 001b:4706
001a:00c7
01f4:33303031373431313137

raymondjclark, 2017-11-08:
Generator 7033, Evolution Controller
Exercise set to 1400 on Sunday Biweekly

Num Regs: 208
Not Changed: 2681721 Changed 47762 Total Changed 0.02
0038:0000                       0034:0000
0037:0f78                       0032:4000
05f5:0000                       05f4:0000
05f1:0000                       01f1:0000
05ed:0064                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:0000
0054:0000                       0208:03ec
005c:0000                       005a:0000
005f:0000                       005d:0000
003c:0000                       000d:0000
000e:0f00                       000f:0b08
000a:008f                       000c:0000
0005:0e00                       0006:0000
0000:0009                       0001:00000000
0008:0000                       0009:00f0
01f2:0000                       0012:0000
0011:009c                       0010:0311
0007:0000                       0019:000f
002b:0005                       002c:0e00
002a:6a76                       002f:0000
002e:0000                       0020:0000
0021:0000                       001f:49e0
001e:018f                       001d:0000
001c:0000                       001b:4706
001a:00c7
01f4:33303031373431313137

raymondjclark, 2017-11-08:
Generator 7033, Evolution Controller
Exercise set to 0600 on Saturdays weekly

Num Regs: 208
Not Changed: 2689630 Changed 47937 Total Changed 0.02
0038:0000                       0034:0000
0037:0f78                       0032:4000
05f5:0000                       05f4:0000
05f1:0000                       01f1:0000
05ed:0064                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:0000
0054:0000                       0208:03ec
005c:0000                       005a:0000
005f:0000                       005d:0000
003c:0000                       000d:0000
000e:0f04                       000f:0b08
000a:008e                       000c:0000
0005:0600                       0006:0600
0000:0009                       0001:00000000
0008:0000                       0009:00ef
01f2:0000                       0012:0000
0011:009c                       0010:0311
0007:0000                       0019:000f
002b:0005                       002c:0600
002a:6a76                       002f:0000
002e:0006                       0020:0000
0021:0000                       001f:49e0
001e:018f                       001d:0000
001c:0000                       001b:4706
001a:00c7
01f4:33303031373431313137

jgyates, 2017-11-10:
That is good info. I have some ideas, just need to chew on them for a bit. I also need to make some changes to the code to make handling this a little easier. I don't have the exact sequence down so once I get my thoughts together I may ask for another test or data point. It looks like register 002e is key, but there may be another one involved also. I will be in touch...

raymondjclark, 2017-11-10:
Sweet

jgyates, 2017-11-13:
OK. I checked in a new version of genmon that has some additional debug functionality. Once you update to the latest genmon run these tests:

Exercise set to 2st of each month
Exercise set to 1400 on Wednesday Biweekly
Exercise set to 0600 on Wednesday weekly

For each of these cases please provide the registers for each as before. Also, I have added a new command to read un-cached registers (registers that are not saved by genmon). For each of the three cases above issue the following commands in ClientInterface.py

generator: readregvalue=002b
generator: readregvalue=002d
generator: readregvalue=002e
generator: readregvalue=002f

You can copy and paste each line individually from a command window shell. Let me know if you have any questions.

Also, since you are volunteering your time feel free to do this at on your schedule. 
Thanks for your help!
Jason

raymondjclark, 2017-11-13:
Exercise set to 2nd of each month at 0800
Registers:
Num Regs: 208
Not Changed: 14610 Changed 333 Total Changed 0.02
0038:0000                       0034:0000
0037:10a9                       0032:4000
05f5:0000                       05f4:0000
05f1:0000                       01f1:0000
05ed:0064                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:0000
0054:0000                       0208:03ec
005c:0000                       005a:0000
005f:0000                       005d:0000
003c:0000                       000d:0000
000e:0929                       000f:0b0d
000a:008b                       000c:0003
0005:0800                       0006:0000
0000:0009                       0001:00000000
0008:0000                       0009:00ef
01f2:0000                       0012:0000
0011:009c                       0010:0111
0007:0000                       0019:000f
002b:0005                       002c:0800
002a:6a76                       002f:0000
002e:0002                       0020:0000
0021:0000                       001f:49e0
001e:018c                       001d:0000
001c:0000                       001b:4706
001a:00c4
01f4:33303031373431313137

>generator: readregvalue=002b
0005
>generator: readregvalue=002d
0002
>generator: readregvalue=002e
0002
>generator: readregvalue=002f
0000
>

raymondjclark, 2017-11-13:
Model 7033 AirCooled Evolution controller
Exercise set to 1400 on Wednesday Biweekly

Registers:
Num Regs: 208
Not Changed: 21614 Changed 428 Total Changed 0.02
0038:0000                       0034:0000
0037:10a9                       0032:4000
05f5:0000                       05f4:0000
05f1:0000                       01f1:0000
05ed:0064                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:0000
0054:0000                       0208:03ec
005c:0000                       005a:0000
005f:0000                       005d:0000
003c:0000                       000d:0000
000e:092c                       000f:0b0d
000a:008c                       000c:0003
0005:0e00                       0006:0300
0000:0009                       0001:00000000
0008:0000                       0009:00ee
01f2:0000                       0012:0000
0011:009c                       0010:0111
0007:0000                       0019:000f
002b:0005                       002c:0e00
002a:6a76                       002f:0000
002e:0003                       0020:0000
0021:0000                       001f:49e0
001e:018c                       001d:0000
001c:0000                       001b:4706
001a:00c4
01f4:33303031373431313137

>generator: readregvalue=002b
0005
>generator: readregvalue=002d
0001
>generator: readregvalue=002e
0003
>generator: readregvalue=002f
0000

raymondjclark, 2017-11-13:
Model 7033 AirCooled Evolution controller
Exercise set to 0600 on Wednesday Weekly

Registers:
Num Regs: 208
Not Changed: 29536 Changed 606 Total Changed 0.02
0038:0000                       0034:0000
0037:10a9                       0032:4000
05f5:0000                       05f4:0000
05f1:0000                       01f1:0000
05ed:0064                       023b:0000
023e:0000                       003b:0000
05fa:0000                       0059:0000
0058:0000                       0053:0000
0052:0000                       0057:0000
0056:0000                       0055:0000
0054:0000                       0208:03ec
005c:0000                       005a:0000
005f:0000                       005d:0000
003c:0000                       000d:0000
000e:0930                       000f:0b0d
000a:008b                       000c:0003
0005:0600                       0006:0300
0000:0009                       0001:00000000
0008:0000                       0009:00ee
01f2:0000                       0012:0000
0011:009c                       0010:0111
0007:0000                       0019:000f
002b:0005                       002c:0600
002a:6a76                       002f:0000
002e:0003                       0020:0000
0021:0000                       001f:49e0
001e:018c                       001d:0000
001c:0000                       001b:4706
001a:00c4
01f4:33303031373431313137

>generator: readregvalue=002b
0005
>generator: readregvalue=002d
0000
>generator: readregvalue=002e
0003
>generator: readregvalue=002f
0000

jgyates, 2017-11-14:
@raymondjclark, question: when set the day of the month in the controller, do the days of the month go up to 31, 30, 29 or 28? i.e. what is the maximum day of the month you can set?

I think I have the registers figured out, I just need some time to make the code changes. There are quite a few sprinkled in genmon and in the web interface. I am guessing it will be a few days before I have something to test.

raymondjclark, 2017-11-14:
I looked and 28 is the max for day.
Thanks for all your efforts!

jgyates, 2017-11-15:
@raymondjclark, I just checked in the following files:
genmon.py, static/genmon.css, static/genmon.js, static/index.html

If you refresh you repositorty and add the following line in your genmon.conf file you should be able to test this out. Add this line to /etc/genmon.conf:

     # (optional) This parameter will enable the functionality to support biweekly
     # and monthly exercise times (weekly is the default). Note, your generator
     # controller  must support this functionality before enabling this option.
     enhancedexercise = True

Let me know how it works. I tested as much as I could. My generator does not support this feature so thanks for testing. Based on your testing results it looks fairly straight forward. Most of the updates were in the web UI.

For the command line interface, the way to set the exercise time was:

     generator: setexercise=Monday,14:00

Now the format for weekly is:

     generator: setexercise=Monday,14:00,Weekly

or for biweekly:

     generator: setexercise=Monday,14:00,Biweekly

or for Monthly:

     generator: setexercise=15,14:00,Monthly 

Once we validate this works for you I will add the documentation to the wiki for the new features. For now the code checked in should work as before unless the new genmon.conf option is enabled.

Let me know if you have any question.

raymondjclark, 2017-11-15:
Thanks. I am on an elk hunt and will be able to test next Monday.

jgyates, 2017-11-16:
OK. Sounds good. Good luck on your hunt.

raymondjclark, 2017-11-18:
Thanks for the update. 
I set to biweekly and it was reported in the gui as weekly, but it had set the generator.
I tried to set monthly via cli and I got "Invalid command syntax for command setexercise"
Tried to set monthly via gui and it did not report any change.
Set to biweekly in gui and it was reported weekly.
Manually set generator to monthly, on the 1st at 1500, reported weekly Sunday 1500

No elk dangit.

jgyates, 2017-11-19:
OK. I found a couple of things that were not right with my logic. I have updated genmon.py, /static/index.html and /static/genmon.js. Let me know how your testing goes.

raymondjclark, 2017-11-20:
Ah, looking better. Will test tomorrow.

raymondjclark, 2017-11-20:
Setting to weekly and biweekly works fine, however setting to monthly fails unless the command is given a day of the week (generator:setexercisetime=Sunday,12:00,monthly) vs a date (generator:setexercisetime=01,12:00,monthly). I looked through ParseExerciseStringEx in genmon.py but couldn't help with a fix. I did notice there is no bounds check for date (1-28).

jgyates, 2017-11-20:
All of the bounds checking takes place in the calling routine. I will go through it and see if I can find anything. Do you have any errors in /var/log/genmon.log?

raymondjclark, 2017-11-20:
From ClientInterface:
OK : Auto, Off - Ready
>generator:setexercisetime=01,12:20,monthly
Invalid command syntax for command setexercise

From /var/log/genmon.log
2017-11-20 13:27:58,871 : Validation Error: Error parsing command string in SetGeneratorExerciseTime (v3): setexercisetime=01,12:20,monthly

jgyates, 2017-11-20:
I think I found the problem. Line 1192 is:

     if ModeStr.lower != "monthly":

it should be:

     if ModeStr.lower() != "monthly":

jgyates, 2017-11-20:
I just checked in a fix for this. It may take a min or two, I checked it in but github has yet to show it is there..

raymondjclark, 2017-11-20:
That did it! Great, thank you. I will also send some beer money.

raymondjclark, 2017-11-20:
I updated genmon.py myself for my test.

jgyates, 2017-11-20:
Note to all who are using this feature: 

1) If you are using the web interface, you must update the files in /static directory as these files were updated to allow the UI to select monthly and biweekly allow
2) Your controller must support this functionality. Just enabling it in the software will not change the way your generator works. From the data that I have, Evolution air cooled models with the latest firmware support this feature. Liquid cooled Evolution does not. I am not sure about Nexus, you will have to check your controller. If you can set it via the controller you can enable this feature to set monthly, weekly or biweekly via genmon.
3) The old command format of setexercise=Monday,13:00 still works. Also the new command of:

     setexercise=Monday,13:30,Weekly 

is now supported. If you enable the new functionality in the genmon.conf file you can also use this format to set biweekly and monthly:

     setexercise=Monday,13:30,Biweekly

or

     setexercise=15,14:00,Monthly

Note the 15 in the above example is the 15th day of the month. You can set a day of the month between 1 and 28.

The readme, wiki and example genmon.conf have been updated. Let me know if there are any questions.

jgyates, 2017-11-20:
One other note:

This option is disabled by default in the genmon.conf in the repository. Also, if you enable this option you will need to restart gennon.py and refresh the web app for the weekly, biweekly, monthly controls to show up in the web UI.

closed 2017-11-25 by jgyates

jgyates, 2018-02-27:
@raymondjclark ,
Are you seeing the issue described in this thread: https://github.com/jgyates/genmon/issues/75

We are thinking that this may be related to a firmware change and it looks like you have FW V1.18
Thanks
Jason