jgyates/genmon #51: Nexus Monitor Screen Flashing

USAFPride, 2017-11-12:
I have built mine using a piZeroW and max3232 chip and powering off pins 1 & 2.  Tested the cable, loopback test passes.  Head out to the generator (Nexus air cooled), connect it up and everything works as expected, BUT the Nexus display is now just flashing it's light on and off with no message. Just a blank screen.  It should be showing the time and that I need to change the oil and filter. `Switch State: System in Alarm` shows on the webpage, and the orange maintenance light is still on.
Thoughts?

madisdev, 2017-11-12:
Does the generator act normal once you  disconnect the PI.

jgyates, 2017-11-12:
The alarm is likely the maintenance warning. You should be able to pull up the web page and look at the alarm log to tell the the last alarm condition. Can you validate the last alarm via the web page?

USAFPride, 2017-11-12:
@madisdev - the screen still just flashes the backlight on and off whether the pi is connected or disconnected.  It was working correctly before the connection.

@jgyates - I'm sure it is the oil change warning. The last alarm was back in June and that was due to leaving the gas off during some maintenance and then trying to start the generator (which I cleared).  Without the display working, I can't really see what the alarm is.  The webpage doesn't offer any specifics.  

Is there a way to a) get the specifics on the current alarm (triggered before the pi was connected) and b) a way to remotely clear the alarm(s) via modbus?  If b is possible, then if the screen just happened to go bad, I can keep all the same functionality without spending $ to replace it.

jgyates, 2017-11-12:
For you question a),  the web interface or ClientInterface.py ( command line) can both retrieve the alarm log history. Look in the "Logs" section of the web page (clickable menu on the left). You should see a start/stop log and an alarm log that shows the last 50 starts/stop events and the last 50 alarms.

For question b, no, I know of no way to clear an alarm via modbus.

madisdev, 2017-11-12:
What do you think of disconnecting battery to see if a power cycle helps.

USAFPride, 2017-11-12:
@jgyates, the last entry into the alarm log is the one from June. 

The controller works correctly, you just can't read it. With the backlight cycling on and off and the fuses all checking good, I think something in the display interface has gone bad. I need to figure out how to power down the display, which may involve switching to Off and pulling the fuses in the ATS.

USAFPride, 2017-11-12:
@madisdev, already did that (at least the negative). Backlight keeps cycling on and off. Pulled 7.5A fuse, same thing. I need to find where the power is supplied and cut it there to see if I can reset it.

jgyates, 2017-11-12:
@USAFPride, to power cycle the controller you will need to disconnect the controller or disconnect the battery and utility power and turn generator off. The controller will be powered from three sources (battery, generator power if utility is gone, or utility power). One of the connectors on the controller supplies power, not sure which one for Nexus. You might be able to get by with just disconnecting the battery (which will cause and alarm) and turning off your utility power on your outside switch (the inside switch will not disconnect power from the generator).

jeepjungle, 2017-11-15:
FWIW, when I had strangeness from my Nexus controller on my water cooled QT36, my maintenance tech suggested I pull the wiring harness connectors off the back of the Nexus for a couple minutes, then reconnect them to allow the Nexus controller to reset itself.  That actually helped with the problem I was having.

USAFPride, 2017-11-17:
Disconnecting the wiring harness fixed all the issues.  All working now.  Thought that was a $400 issue!!!

closed 2017-11-19 by USAFPride