jgyates/genmon #53: Logs: Start Stop Alarm

USAFPride, 2017-11-22:
I'm wondering if it makes sense to put them in 2 columns side by side as the start/stop log can get quite long and push the alarm logs to the bottom, which IMO is more important than the start/stop logs.  What is the limit of the number of start/stop entries will it show (mine is currently showing 50)?  Alternatively, limit it to the last 5 or 10  with an option to see all logs.

jgyates, 2017-11-22:
The limit of the controller is 50. Older entries, greater than 50 will be deleted by the controller. The Nexus controller has a  2 logs (start / stop log and an alarm log) while the Evolution controller has 3 logs (start/stop, maintenance, and alarm log). Supporting two columns my get a bit odd in the formatting for system with 3 logs, but I can make the start/stop log appear last. I am guessing a good order would be alarm, maintenance (if supported), and finally start / stop. Sound reasonable?

USAFPride, 2017-11-22:
Reasonable.  Alarms are more important than start/stop :)

jgyates, 2017-11-22:
Done, new version checked in.

USAFPride, 2017-11-22:
IMO, needs a space between last alarm log and Start Stop Log: label

jgyates, 2017-11-22:
Try it now.