jgyates/genmon #54: Connecting via a Serial Server

davidrule1969, 2017-11-22:
FYI - rather than install a Raspberry Pi inside my generator, I made a cable to connect to my existing serial server.  My serial server is a Digi CM32 but there are plenty of others out there that should all work the same.

I made a cable with pins 2, 7 & 8 on the molex connector on one end and pins 3, 5 & 6 of an RJ45 plug on the other.  This directly connects my generator to the serial server.

On the Raspberry Pi (running in my basement) I used SOCAT to create /dev/serial2 which is connected to the serial server via TCP.  Genmon runs perfectly!

-David

jgyates, 2017-11-22:
Hi David, Thanks for the info!

I guess for that matter you could run gen mon in a VM with Ubuntu or Debian with SOCAT. Once you decouple the serial port it is just a bunch of python processing code from a buffer.

UPS99, 2018-02-28:
I used a standard Raspberry Pi 3, but chose to mount it indoors and not within the generator.  I ran 25 feet of 4 conductor, 16 gauge, shielded, outdoor cable, grounding the shield at only one end, and using one of the wires as the common ground at both ends, as Generac did with their cables.  I mounted the Raspberry Pi in one the common plastic Pi cases and used a different RS232 to TTL adapter that could fit inside the case:
https://www.amazon.com/gp/product/B01DAR1ZOY/ref=oh_aui_detailpage_o07_s00?ie=UTF8&psc=1
To provide a bit of isolation from surges, and protection from shorts, I added 1K resistors in series with the RS232 data lines.  These are high impedance lines, and 1 K has no effect.  A 5 volt wall-wart supply powers it (connected to a UPS) and I used wired ethernet, not wi-fi.  I also proxied the web server through an existing internet-connected web server with a login page.  Still, all these are minor compared to the effort put into this by Jason.

closed 2018-03-16 by jgyates