jgyates/genmon #55: Edit Config via Webpage

USAFPride, 2017-11-24:
Obviously, this would take a bit of coding, but it would be really nice if the configs could be modified from an "admin" page.

jgyates, 2017-11-24:
Hi @USAFPride ,

I will think on this. There are a few issues off the top of my head that I will need to think about implementation. This does open a can of worms and the benefit may not outweigh the effort. Looking at the config file options, SiteName, Serial Port, sync time, DST are the main ones that would need to be changed by the user. There are a lot of debug related options and options that control output to the console so those do not make a lot of sense to put in the UI in my view. It may make more sense to add a subset of the conf file, if I impliment this. I would still need to address restarting, reloading and writing out the settings to the config file. In short, it is do-able, I just need to look at the benefit vs effort and then figure out the best way to implement the changes.

Thanks
Jason

madisdev, 2017-11-24:
I would have to agree with @jgates, I am not sure its really worth the effort. I can't say I have edited the config file after the initial install. A UI is nice, but don't think its worth the time unless someone wants to add it themselves. Pull request are welcome :)

proppilot, 2017-11-25:
I have zero experience with the js programming for the web pages.  Can anyone point me in the direction or suggestions/hints on how to modify the web page code to add a very simple username and password login before displaying the status page that would be based on a hard coded username and password ?  I hate asking general questions like this however the genmon is working perfectly and I don't want to screw it up !

jgyates, 2017-11-25:
Probably the easiest way is to add it to the Flask python app (genserv.py). This would be much more secure since if you hardcoded it into the javascript it would be visible to anyone and possibly make it a target for hackers.

This site has some info on adding authentication to the Flask python app:

https://flask-basicauth.readthedocs.io/en/latest/

I don't know what your needs are, however if you are wanting to expose the web app to the outside world I would recommend a VPN instead of adding authentication to the web page. It is much more secure and it will keep the traffic way from the pi, not just password protect it. Many routers have a VPN services built in and both iOS and Android both have VPN clients built into their mobile operating systems.

proppilot, 2017-11-25:
Never even thought about VPN ( I am an EE not an IT guy ! ) but see that my Synology DSM has a VPN server app that can be loaded.  I like this solution and am going to play with that.  Thanks !

jeepjungle, 2017-11-27:
FWIW, I use pfSense which is free firewall software based on FreeBSD to protect my network.  I run OpenVPN service from there so I can can connect to my internal home network from my phone and access the generator page that way.  You can issue a security certificate from the OpenVPN server and place it on your phone.

jgyates, 2017-11-28:
Closing issue

closed 2017-11-28 by jgyates

jgyates, 2017-11-30:
@proppilot 

I added HTTPS and a simple username password. You need to enable this in /etc/genmon.conf. I updatedd the readme / wiki. Look in the comments for genmon.conf for details or let me know if you have questions. User ColtB45 provided this code so I rolled it into the master branch. It is disabled by default but can easily be enabled.

Thanks
Jazson

proppilot, 2017-11-30:
> <reply+02039b7ac87806b566fe427034c4c33215b0759ab49ff88592cf0000000116370eb39
> 2a169ce107e1f19@reply.github.com>
> <mention@noreply.github.com>

Fantastic !

<https://github.com/jgyates/genmon/issues/55#issuecomment-348039775> , or
mute the thread 
I4rV5lfks5s7fKzgaJpZM4QqMcx> .

  
b","name":"GitHub"},"entity":{"external_key":"github/jgyates/genmon","title"
:"jgyates/genmon","subtitle":"GitHub
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
-9aed-b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/jgyates/genmon"}},"updates":{"snippets":[{
"icon":"PERSON","message":"@jgyates in #55: @proppilot \r\n\r\nI added HTTPS
can easily be enabled.\r\n\r\nThanks\r\nJazson"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/55#issuecomment-34803
9775"}}}

proppilot, 2017-11-30:
Jason,  this is going to be a stupid question.  When I first configured the
project I downloaded all of the files from the project via the GUI interface
and the pi local with the HDMI cable, keyboard and mouse.

Now the Pi is remote and locked up.  How do I download these updated files
via SSH into the Pi ?  Is it a one shot 'all files' deal and I have to redo
the config and mail files etc ?

Thanks, Tony

jgyates, 2017-11-30:
Hi Tony,

No problem at all. You have a couple of options. Both assume a PC is connected to the same network as your pi. Both of these examples assume an IP address of IPADDRESS (replace this with the actual IP address of your pi)

If you use Windows, I would install PuTTY as it contains a graphical ssh / scp tool. These tools (ssh and scp) are included in linux and macOS.
http://ged.msu.edu/angus/tutorials/using-putty-on-windows.html

1) Use a secure shell (ssh) to make a terminal connection on your PC into your pim then refresh your git repository. 

Open a command prompt on a linux or macOS system and type (use PuTTY for Windows):

     ssh pi@IPADDRESS

you will be prompted for the username of the pi user on your pi. Then change directories into your genmon directory and update the git repository by typing:
    
    git pull
    
this will update to the latest, or, if you have modified any files and want to overwrite them then type:

    git reset --hard origin/master

This command will update your entire repository to the latest. It will not overwrite your /etc/genmon.conf file.  You will need to change the If you want to use any new options you will need to update your file manually.  If you do a "git reset"  you will need to type "sudo chmod 755 ./startgenmon.sh" to change the permissions again like you did during install. Once your files are updated you can restart genmon via the ./startgenmon.sh scripts, or just reboot your system (sudo reboot now)

The second option is to download the files and copy them from your PC with the secure copy command (scp). From a terminal / command windows on Linxu or macOS (or use PuTTY for Windows) copy the files over with the terminal command:

    scp fileToCopy pi@IPADDRESS:./genmon

In the above directory ./ is the home directory of the pi user on your system with and IP address of IPADDRESS and the file will be copied into the genmon directory.

The first option is the easiest, but it is good to know the second one as it comes in handy sometimes.

Let me know if you have any other questions.

proppilot, 2017-11-30:
> <reply+02039b7a7bfbb42ef898fbdfe7142ae00b25c66a1ddb11bf92cf00000001163744389

Hi Jason,  excellent.  I am using a MAC and do all of my programming/playing
using TERMINAL and SSH.  I will update everything this weekend.

Cheers, Tony

Date:  Wednesday, November 29, 2017 at 10:59 PM

http://ged.msu.edu/angus/tutorials/using-putty-on-windows.html

k22PJ8Eks5s7ig4gaJpZM4QqMcx> .

  
"icon":"PERSON","message":"@jgyates in #55: Hi Tony,\r\n\r\nNo problem at
(replace this with the actual IP address of your pi)\r\n\r\nIf you use
macOS.\r\nhttp://ged.msu.edu/angus/tutorials/using-putty-on-windows.html\r\n
\r\n1) Use a secure shell (ssh) to make a terminal connection on your PC
into your pim then refresh your git repository. \r\n\r\nOpen a command
prompt on a linux or macOS system and type (use PuTTY for Windows):\r\n\r\n
ssh pi@IPADDRESS\r\n\r\nyou will be prompted for the username of the pi user
the git repository by typing:\r\n    \r\n    git pull\r\n    \r\nthis will
overwrite them then type:\r\n\r\n    git reset --hard
origin/master\r\n\r\nThis command will update your entire repository to the
now)\r\n\r\nThe second option is to download the files and copy them from
with the terminal command:\r\n\r\n    scp fileToCopy
pi@IPADDRESS:./genmon\r\n\r\nIn the above directory ./ is the home directory
will be copied into the genmon directory.\r\n\r\nThe first option is the
sometimes.\r\n\r\nLet me know if you have any other questions.\r\n \r\n
\r\n       "}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/55#issuecomment-34807
6096"}}}

proppilot, 2018-01-19:
Hi Jason,

Apologies up front as I am not an experienced Unix/Pi user thus the dumb
question I am sure.

I wrote a .sh script to detect loss of wifi connect by pinging my router and
then initiating a reboot ( screen shot of script enclosed )

The .sh script is located in the 'OtherApps' directory off of the genmon
directory

The other shot is of my crontab file that I open by typing 'sudo crontab ­e'

On reboot, it successfully runs the shell that powers the heartbeat light on
the enclosure and runs the genmon software as well as the checkwifi.sh
script which can be seen running in the attached screen shot of htop (red
box around the process)

When I shut down the wifi, no reboot happens  I only have a 'wait 30' in
there (was wait 600) for testing as I have the unit on my desk now to play.

Anything jump out at you ?  Does the root process not have the privilege to
force the shutdown ?

I have seen improved versions of the wifi disconnect code posted and will
play with them once I figure out why the script does not seem to work.

Thanks !  Tony

jgyates, 2018-01-19:
Hi Tony,

I don't see an attachment with the screen shot. You can post if and I will take a look.  Below is my script that I use to check for wifi disconnect:

#router ip
/bin/ping -I wlan0 -c4 192.168.1.1 > /dev/null

if [ $? != 0 ]
then
#make sure you have the correct wlan
/sbin/ifdown --force wlan0
/sbin/ifup wlan0
/usr/bin/mail -s "Generator WLAN Notice" myemail@gmail.com <<< "WLAN0 Restarted"
fi

I this runs every 5 min via crontab. I add this line to crontab:

*/5 * * * * /home/pi/checkwifi.sh

Also, your script must have executable privileges. You can set the privilege with this command:

sudo chmod 755 /home/pi/checkwifi.sh

proppilot, 2018-01-19:
> <reply+02039b7a7e8c4c0e3e45ca483cca40e0e70e826288be035492cf00000001167a2e759

Hi Jason,

Here are the shots.  I am trying your code below now as well.

Thanks, Tony

Date:  Friday, January 19, 2018 at 5:08 PM

gfYmtFHks5tMRJ1gaJpZM4QqMcx> .

  
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png&quot;,&quot;avatar_ima
ge_url&quot;:&quot;https://cloud.githubusercontent.com/assets/143418/1584216
6/7c72db34-2c0b-11e6-9aed-b52498112777.png&quot;,&quot;action&quot;:{&quot;n
ame&quot;:&quot;Open">https://cloud.githubusercontent.com/assets/143418/1749
2498112777.png","action":{"name":"Open</a> in
GitHub","url":"https://github.com/jgyates/genmon&quot;}},&quot;updates&quot;
:{&quot;snippets&quot;:[{&quot;icon&quot;:&quot;PERSON&quot;,&quot;message&q
uot;:&quot;@jgyates">https://github.com/jgyates/genmon"}},"updates":{"snippe
ts":[{"icon":"PERSON","message":"@jgyates</a> in #55: Hi Tony,\r\n\r\nI
disconnect:\r\n\r\n#router ip\r\n/bin/ping -I wlan0 -c4 192.168.1.1 \u003e
/dev/null\r\n\r\nif [ $? != 0 ]\r\nthen\r\n#make sure you have the correct
wlan\r\n/sbin/ifdown --force wlan0\r\n/sbin/ifup wlan0\r\n/usr/bin/mail -s
\"Generator WLAN Notice\" mailto:myemail@gmail.com">myemail@gmail.com</a>
\u003c\u003c\u003c \"WLAN0 Restarted\"\r\nfi\r\n\r\nI this runs every 5 min
via crontab. I add this line to crontab:\r\n\r\n*/5 * * * *
/home/pi/checkwifi.sh\r\n\r\nAlso, your script must have executable
privileges. You can set the privilege with this command:\r\n\r\nsudo chmod
755 /home/pi/checkwifi.sh \r\n\r\n\r\n"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/55#issuecomment-35910
3128&quot;}}">https://github.com/jgyates/genmon/issues/55#issuecomment-35910
3128"}}</a>}

jgyates, 2018-01-19:
Hi Tony,

I still do not see your attachments. If you are replying to this thread in email, then email attachments will not show up. If you go to the issue section in the project forum then you can attach files. The URL for this thread is:

https://github.com/jgyates/genmon/issues/55

Or you can navigate to the thread from the main project page.

proppilot, 2018-01-19:
sorry.  here are the shots
[image]
[image]
[image]

jgyates, 2018-01-20:
It looks like your checkwifi.sh is only running on reboot. Change this line:

@reboot sh /home/pi/gemon/OtherApps/checkwifi.sh

to 

*/5 * * * * /home/pi/checkwifi.sh

The remove the do / while loop from your script.

The "5" in the line above is for "every 5 minutes", you can change this to another number if you want to. 

The @reboot, tells chrontab to run once at every reboot. 

It looks like you are ask chrontab to run a script once at start and expect the script to keep running forever. I think chron will kill the task after a period of time as a cleanup measure. I would make the checkwifi a single task that runs once and let chrontab run it again.

Everything else looks OK to me,

proppilot, 2018-01-20:
> <reply+02039b7a731f5458d48710e19133b1322651573ba080e57c92cf00000001167a4b1f9

Hi Jason, just to see if this is running every five minutes I added another
line of code at the bottom to send me an email (my real email address is in
there) and I am not getting any messages

#!/bin/bash

# Script: checkwifi.sh

# Purpose : Check for wlan(0) active and reset if disconnected

# _________________________________________________________________

#router ip

/bin/ping -I wlan0 -c4 192.168.0.1 > /dev/null

if [ $? != 0 ]

then

#make sure you have the correct wlan

/sbin/ifdown --force wlan0

/sbin/ifup wlan0

/usr/bin/mail -s "Generator WLAN Notice" xxxx@xxxx.com <<< "WLAN0 Restarted"

fi

/usr/bin/mail -s "Generator WLAN Notice" xxxx@xxxx.com <<< "WLAN0 Tested"

Date:  Friday, January 19, 2018 at 7:11 PM

Kc7z5uqks5tMS8fgaJpZM4QqMcx> .

  
"icon":"PERSON","message":"@jgyates in #55: It looks like your checkwifi.sh
is only running on reboot. Change this line:\r\n\r\n@reboot sh
/home/pi/gemon/OtherApps/checkwifi.sh\r\n\r\nto \r\n\r\n*/5 * * * *
/home/pi/checkwifi.sh\r\n\r\nThe remove the do / while loop from your
script.\r\n\r\nThe \"5\" in the line above is for \"every 5 minutes\", you
can change this to another number if you want to. \r\n\r\nThe @reboot, tells
chrontab to run once at every reboot. \r\n\r\nIt looks like you are ask
let chrontab run it again.\r\n\r\nEverything else looks OK to
me,"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/55#issuecomment-35912
4592"}}}

proppilot, 2018-01-20:
I also notice I don't have a /mail directory off /usr/bin

Sorry, I know these are really elementary issues and I appreciate the help !

jgyates, 2018-01-20:
/usr/bin/mail is a program not a directory. it this the command line way to send mail. It should come with the OS so there is nothing to install. If you still dont have it let me know.  To test just type:

sudo /usr/bin/mail -s "Generator WLAN Notice" xxxx@xxxx.com <<< "WLAN0 Restarted"

at the command prompt (fill in with your email instead of xxxx)

proppilot, 2018-01-20:
> <reply+02039b7a8bde6dcd8c0c705462d83fb80db7f6019523dee292cf00000001167a89f29

Hi Jason,  thanks very much.  I did figure that out last night and when I
try to send the email I get an error

mail: cannot send message: Process exited with a non-zero status

I believe the mail is configured in /etc/ssmtp/ssmtp.conf ?

I have seen that there are issues with gmail not authorizing (my domain is
dot mail) so that may be the issue.

Is there a debug mode or log where the failure or error would be registering
?

BTW I do see in SYSLOG that cron is running the wlan watchdog every 5 so
that is working.  I just like the idea now of the email of failure and want
to get that working

Date:  Friday, January 19, 2018 at 11:39 PM

jJWA1o7ks5tMW3ygaJpZM4QqMcx> .

  
"icon":"PERSON","message":"@jgyates in #55: /usr/bin/mail is a program not a
know.  To test just type:\r\n\r\nsudo /usr/bin/mail -s \"Generator WLAN
Notice\" xxxx@xxxx.com \u003c\u003c\u003c \"WLAN0 Restarted\"\r\n\r\nat the
xxxx)\r\n"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/55#issuecomment-35914
4971"}}}

jgyates, 2018-01-20:
Hi Tony,

I used this article to setup sending mail on my pi.

http://www.pontikis.net/blog/gmail-smarthost-exim4-debian

Use the guidance in the article to setup sending mail but prepend "sudo " before the following commands listed in the article:

dpkg-reconfigure exim4-config

nano /etc/exim4/passwd.client

proppilot, 2018-01-21:
> <reply+02039b7a7e01ad07a286f5c241e513f56061ded4fc23f4e692cf00000001167b5b689

Looks like my domain mail(dot)com does not allow login from less secure apps

When I change the ssmtp config to my gmail address and enable 'less secure
apps' on the gmail account it works great

Date:  Saturday, January 20, 2018 at 2:32 PM

http://www.pontikis.net/blog/gmail-smarthost-exim4-debian

baCLJ5Qks5tMj9ogaJpZM4QqMcx> .

  
"icon":"PERSON","message":"@jgyates in #55: Hi Tony,\r\n\r\nI used this
pi.\r\n\r\nhttp://www.pontikis.net/blog/gmail-smarthost-exim4-debian\r\n\r\n
before the following commands listed in the article:\r\n\r\ndpkg-reconfigure
exim4-config\r\n\r\nnano
/etc/exim4/passwd.client\r\n"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/55#issuecomment-35919
6299"}}}

proppilot, 2018-01-21:
Hi Jason,  email issues seem to be related to the mail(dot)com domain so
that is resolved and thanks for the tip to us crontab to run the .sh task to
check wifi every 5 mins vs running it once and having it have its own timer

Last question (I hope !) is something I see in SYSLOG when I went to check
to ensure the checkwifi was running.  Issue seems related to the
avahi-daemon and the software constantly updating the host name ?

I did look at various sources including some other postings on GitHub but
not sure how to resolve this ?

Jan 21 06:25:02 GenMon liblogging-stdlog:  [origin software="rsyslogd"
swVersion="8.24.0" x-pid="335" x-info="http://www.rsyslog.com"] rsyslogd was
HUPed

Jan 21 06:25:02 GenMon liblogging-stdlog:  [origin software="rsyslogd"
swVersion="8.24.0" x-pid="335" x-info="http://www.rsyslog.com"] rsyslogd was
HUPed

Jan 21 06:29:10 GenMon avahi-daemon[375]: Withdrawing address record for
fd00:9050:cac7:bec2:4fb:a843:a937:a3d on wlan0.

Jan 21 06:29:10 GenMon avahi-daemon[375]: Withdrawing address record for
2607:fea8:601f:fb2e:1b45:a7f0:b7e8:4cc5 on wlan0.

Jan 21 06:29:10 GenMon avahi-daemon[375]: Withdrawing address record for
192.168.0.203 on wlan0.

Jan 21 06:29:10 GenMon avahi-daemon[375]: Host name conflict, retrying with
GenMon-1340

Jan 21 06:29:10 GenMon avahi-daemon[375]: Registering new address record for
2607:fea8:601f:fb2e::3 on wlan0.*.

Jan 21 06:29:10 GenMon avahi-daemon[375]: Registering new address record for
fd00:9050:cac7:bec2:4fb:a843:a937:a3d on wlan0.*.

Jan 21 06:29:10 GenMon avahi-daemon[375]: Registering new address record for
2607:fea8:601f:fb2e:1b45:a7f0:b7e8:4cc5 on wlan0.*.

Jan 21 06:29:10 GenMon avahi-daemon[375]: Registering new address record for
192.168.0.203 on wlan0.IPv4.

Jan 21 06:29:11 GenMon avahi-daemon[375]: Server startup complete. Host name
is GenMon-1340.local. Local service cookie is 2927119540.

Jan 21 06:29:11 GenMon avahi-daemon[375]: Withdrawing address record for
fd00:9050:cac7:bec2:4fb:a843:a937:a3d on wlan0.

Jan 21 06:29:11 GenMon avahi-daemon[375]: Withdrawing address record for
2607:fea8:601f:fb2e:1b45:a7f0:b7e8:4cc5 on wlan0.

Jan 21 06:29:11 GenMon avahi-daemon[375]: Withdrawing address record for
192.168.0.203 on wlan0.

Jan 21 06:29:11 GenMon avahi-daemon[375]: Host name conflict, retrying with
GenMon-1341

Jan 21 06:29:11 GenMon avahi-daemon[375]: Registering new address record for
2607:fea8:601f:fb2e::3 on wlan0.*.

Jan 21 06:29:11 GenMon avahi-daemon[375]: Registering new address record for
fd00:9050:cac7:bec2:4fb:a843:a937:a3d on wlan0.*.

Jan 21 06:29:11 GenMon avahi-daemon[375]: Registering new address record for
2607:fea8:601f:fb2e:1b45:a7f0:b7e8:4cc5 on wlan0.*.

Jan 21 06:29:11 GenMon avahi-daemon[375]: Registering new address record for
192.168.0.203 on wlan0.IPv4.

Jan 21 06:29:12 GenMon avahi-daemon[375]: Server startup complete. Host name
is GenMon-1341.local. Local service cookie is 2927119540.

http://www.pontikis.net/blog/gmail-smarthost-exim4-debian

jgyates, 2018-01-21:
Hi Tony,

After googling this issue it appears this is a widespread issue that appears to effect other non-raspberry pi systems as well. It appears to be a bug that is related to multiple modules interacting, but that is a guess. Unless it is causing any functional issues I would ignore this and update your system software periodically. I typically update my system software once a week with the following commands:

First, run this once to install the debian goodies package. This will put a program called checkrestart on your system:
sudo apt-get install debian-goodies

Then, once a week (or some other period of time)
sudo apt-get update
sudo apt-get dist-upgrade
sudo checkrestart

If there are updates to packages then the 2nd step will prompt you to install them. Once your run the 3rd step it will tell you if any packages are not using the newly updated versions. If any packages are not able to update then just restart your pi via this command:

sudo reboot

In time the issue you describe will likely be resolved. Linux is notorious for small issues like this so unless there is some functional issue I typically avoid worrying about an error in syslog like this. These issues typically get resolved over time.

proppilot, 2018-01-22:
> <reply+02039b7a5ea25f99af71b721b74b55a9490b902828362a4592cf00000001167ce7eb9

Thanks again for the great support.  I figured it was a system bug and will
do the updates as recommended.  It is not causing any issues.  I like your
wifi watchdog code with the email when it restarts the wlan0.  Just got an
email this evening of a restart so I can get an idea of how often it
happens.  It does seem to resolve the disconnect.

Date:  Sunday, January 21, 2018 at 6:44 PM

EfIAsACks5tM8vrgaJpZM4QqMcx> .

  
"icon":"PERSON","message":"@jgyates in #55: Hi Tony,\r\n\r\nAfter googling
with the following commands:\r\n\r\nFirst, run this once to install the
system:\r\nsudo apt-get install debian-goodies\r\n\r\nThen, once a week (or
some other period of time)\r\nsudo apt-get update\r\nsudo apt-get
dist-upgrade\r\nsudo checkrestart\r\n\r\nIf there are updates to packages
via this command:\r\n\r\nsudo reboot\r\n\r\nIn time the issue you describe
time.\r\n\r\n"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/55#issuecomment-35929
2824"}}}

jgyates, 2018-02-11:
Since it was asked in this thread, email related settings are now editable via the web interface.

proppilot, 2018-03-10:
Jason,  I finally took the plunge to upgrade to the latest checked in
software.  I have a couple of issues.

1) The sh file startgenmon.sh is failing in the crontab @reboot with an
error (I see it on the command line as well )

Startgenmon.sh 3: startgenmon.sh: Bad substitution

2) The web interface still looks like the original one.  None of the new
'stuff' that has been added.

Did something change in the nomenclature for the startgenmon.sh script file
?

buschauer, 2018-03-10:
see #78 for discussions on this.