jgyates/genmon #57: Syslog Support

davidrule1969, 2017-11-29:
Hi,

Would you consider adding support for logging to a remote syslog server?  As well as error messages, if events that generate email, SMS or pushover notifications could also be sent to syslog, that would be great.

This would be a useful addition for monitoring the status of genmon and the generator alongside the ClientInterface.py library/Nagios interface.

Thanks!
David

jgyates, 2017-11-29:
Hi David,

That is a good idea. I added a new utility to log the generator state to syslog. See gensyslog.py in the repository and docs.

Note: This does not log software errors, only generator state. Software errors are still logged in /var/log/genmon

Thanks
Jason

davidrule1969, 2017-11-29:
Thanks!  I'll have a play with it now :-)

jgyates, 2017-12-05:
Closing this issue. Feel free to comment in this thread if you have any issues or questions.

closed 2017-12-05 by jgyates