jgyates/genmon #58: Adding native TLS support within flask

ColtB45, 2017-11-29:
1. Run the following to install the necessary components.
```
sudo apt-get update && sudo apt-get install libssl-dev libffi-dev
sudo pip install pyopenssl
```
2. Edit genserv.py

- For a self-signed cert
Change:
`app.run(host="0.0.0.0", port=8000, threaded = True)`
To:
`app.run(host="0.0.0.0", port=443, threaded = True, ssl_context='adhoc')`

- For a proper certificate
a. Place your cert.pem & key.pem in the genmon folder
b. Change:
`app.run(host="0.0.0.0", port=8000, threaded = True)`
To:
`app.run(host="0.0.0.0", port=443, threaded = True, ssl_context=('cert.pem', 'key.pem'))`

jgyates, 2017-11-29:
I like this a lot! I will add a config file option to enable this in genserv.py. I will let you know once it is there. 

As part of the writeup for this I was planning on including this link for details on how to create the cert and key:

https://workaround.org/ispmail/jessie/create-certificate

Sound reasonable?

ColtB45, 2017-11-29:
Sounds great.

I was going to go down this road as well to add authentication. Would you be interested in the complete changes once I get it done?

https://pythonspot.com/en/login-authentication-with-flask/

jgyates, 2017-11-29:
Sure

jgyates, 2017-11-29:
I updated genserv.py and the docs. There are two new gennon.conf options:

    # (Optional) This parameter, if true will enable the use of HTTPS 
    # (secure HTTP) in the Flask web app. This option is only applicable to the 
    # web app.
    #usehttps=true

    # (Optional) This parameter is used with usehttps. If 
    # userhttps is true, then this option will signal the type of certificate 
    # to use. If this option is true  a self signed certificate (supplied by 
    # Flask library) will be used. If false, then the a key file (key.pem) and 
    # a certificate file (cert.pem) must be created and placed in the same 
    # folder as genserv.py. For information on how to create your own 
    # certificate please see this site: 
    #     https://workaround.org/ispmail/jessie/create-certificate 
    #useselfsignedcert=true

Let me know if this works for you.

Thanks
Jason

jgyates, 2017-11-29:
I tested using HTTP and HTTPS self-signed. I do not have a full cert/key pair so I did not test the non self signed option.

ColtB45, 2017-11-29:
Thanks, it works great.

Here is a simple patch to apply for authentication.
```
diff --git a/genmon.conf b/genmon.conf
index 524b4a3..eda1ed8 100644
--- a/genmon.conf
+++ b/genmon.conf
@@ -125,7 +125,15 @@ outagelog = /home/pi/genmon/outage.txt
 # (Optional) This parameter, if true will enable the use of HTTPS
 # (secure HTTP) in the Flask web app. This option is only applicable to the
 # web app.
-#usehttps=true
+#usehttps = True
+
+# (Optional) This parameter, if usehttps is true will use the set username.
+# This option is only applicable to the web app.
+#http_user = admin
+
+# (Optional) This parameter, if usehttps is true will use the set password.
+# This option is only applicable to the web app.
+#http_pass = password

 # (Optional) This parameter is used with usehttps. If
 # userhttps is true, then this option will signal the type of certificate
@@ -135,4 +143,4 @@ outagelog = /home/pi/genmon/outage.txt
 # folder as genserv.py. For information on how to create your own
 # certificate please see this site:
 #     https://workaround.org/ispmail/jessie/create-certificate
-#useselfsignedcert=true
+#useselfsignedcert = True
diff --git a/genserv.py b/genserv.py
index 9fd3409..ed51fdd 100644
--- a/genserv.py
+++ b/genserv.py
@@ -9,7 +9,7 @@
 # MODIFICATIONS:
 #------------------------------------------------------------

-from flask import Flask, render_template, request, jsonify
+from flask import Flask, render_template, request, jsonify, session
 import sys, signal, os, socket, atexit, configparser
 import mylog, myclient

@@ -20,36 +20,49 @@ app = Flask(__name__,static_url_path='')
 log = mylog.SetupLogger("genserv", "/var/log/genserv.log")

 #------------------------------------------------------------
-@app.route('/')
+@app.route('/', methods=['GET'])
 def root():
-    return app.send_static_file('index.html')
+    if not session.get('logged_in'):
+        return render_template('login.html')
+    else:
+        return app.send_static_file('index.html')
+
+@app.route('/', methods=['POST'])
+def do_admin_login():
+    if request.form['password'] == HTTPAuthPass and request.form['username'] == HTTPAuthUser:
+        session['logged_in'] = True
+        return root()
+    else:
+        return render_template('login.html')

 #------------------------------------------------------------
 @app.route("/cmd/<command>")
 def command(command):

-    if command in ["status", "outage", "maint", "logs", "monitor", "getbase", "getsitename", "setexercise", "setquiet", "getexercise","setremote", "settime"]:
[... 57 lines cut ...]
         if bUseSecureHTTP:
             HTTPPort = 443
+
+            if config.has_option('GenMon', 'http_user'):
+               HTTPAuthUser = config.get('GenMon', 'http_user')
+
+            if config.has_option('GenMon', 'http_pass'):
+               HTTPAuthPass = config.get('GenMon', 'http_pass')
+
             if config.has_option('GenMon', 'useselfsignedcert'):
                 bUseSelfSignedCert = config.getboolean('GenMon', 'useselfsignedcert')

@@ -95,8 +117,9 @@ if __name__ == "__main__":
     MyClientInterface = myclient.ClientInterface(host = address,port=clientport, log = log)
     while True:
         try:
-
-            app.run(host="0.0.0.0", port=HTTPPort, threaded = True, ssl_context=SSLContext)
+            if bUseSecureHTTP:
+               app.secret_key = os.urandom(12)
+            app.run(host="0.0.0.0", port=HTTPPort, threaded = True, ssl_context='adhoc')

         except Exception as e1:
             log.error("Error in app.run:" + str(e1))
```

jgyates, 2017-11-30:
Thanks @ColtB45 

I added this to the main branch. I tweaked the config file reading logic to make https required for username / pw login, but still allows https to be used without a username/password. Let me know if anything looks out of place or does not work. I appreciate you taking the time to make this work.

Jason

jgyates, 2017-12-05:
Closing issue. Thanks for you help @ColtB45

closed 2017-12-05 by jgyates