jgyates/genmon #59: Nexus - Set Exercise Time Not Working

USAFPride, 2017-12-08:
Nexus - Air Cooled - 14kW
Log from genmon.log

`2017-11-29 19:13:16,771 : Validation Error: Error parsing command string in ParseExerciseString:  setexercise=sunday,09:00,weekly
2017-11-29 19:13:16,783 : invalid literal for int() with base 10: '00,weekly'
2017-11-29 19:13:16,786 : Validation Error: Error parsing command string in AltSetGeneratorExerciseTime (v1):  setexercise=sunday,09:00,weekly
2017-12-01 19:11:19,526 : Validation Error: Error parsing command string in ParseExerciseString:  setexercise=sunday,09:00,weekly
2017-12-01 19:11:19,528 : invalid literal for int() with base 10: '00,weekly'
2017-12-01 19:11:19,530 : Validation Error: Error parsing command string in AltSetGeneratorExerciseTime (v1):  setexercise=sunday,09:00,weekly
`

jgyates, 2017-12-09:
I think I found a problem. I just checked in a fix. Let me know if this corrects the problem for you. I tested the best I can but I do not have an Nexus controller.

USAFPride, 2017-12-09:
It now changes the time, but...  When I set it to say 9AM, it actually sets it to 8:59AM (and no log errors)

jgyates, 2017-12-09:
Yes, that is known issue.  The method for setting the time on the Nexus is very convoluted. As a result, I do not have it down 100 percent in the code so you can be off by 1 or 2 min. Generac appeared to make the method to set the exercise time more straightforward with the evolution controller. See the docs for the genmon.conf entry for uselegacysetexercise. This option is enabled if you have a Nexus controller. The docs describe this issue.

FYI you can set it multiple times for the same time and it may get set it exactly at times. This has to do with the way the controller sets this time internally and how it exposes it to the registers. The register interface for this value on the Nexus requires the software to take the difference between the current generator time (which we do not know the seconds value, only the hour and min) and then create weighted bit values to write to the Nexus register.

USAFPride, 2017-12-09:
Ok.  Read and understand.  As a note, if I set the time for 9:01 it stores it as 9:00.  8:59 -> 8:58  8:30 -> 8:29
Always 1 minute less.

jgyates, 2017-12-09:
OK. that is a good observation, but if you set it to 1,2,3,4,5 or 6 days in the future or at different time in the day do you get the same offset of 1? If memory serves then it is not consistent, but honestly this testing was done on an evolution so the result may be different on a Nexus, but I doubt it considering how the numbers are calculated (See AltSetGeneratorExerciseTime and CalculateExerciseTime in genmon.py) I found an alternate way to set the exercise time on the evolution, but it does not work on Nexus so I only had this to fall back on. BTW, this unreliable method was the way mobile link was doing it. I think it is very close, just some of the bit interpretations are off.

USAFPride, 2017-12-09:
I was changing the time on the existing day and observed those results.  Changed it to the next day, same time, and it set correctly.  Now it will always set to the correct time.  I've pounded it over and over and now it is working correctly.  It's almost if changing the time on the same day triggers the error, but changing the day sorted it out.  I'll keep playing with it and see if I can find some consistency.  

Feel free to close this and if I sort it out, I'll reopen this ticket.

jgyates, 2017-12-09:
ok. Thanks for the help in finding issues.

closed 2017-12-09 by jgyates

jgyates, 2017-12-09:
FYI, I added one more line of code that performs another consistency check for the Nexus set exercise time. No real improvement in functionality but I at least wanted to let you know I just updated this code. Let me know if it causes a problem in any way.

USAFPride, 2017-12-09:
I tested it and didn't see any issues.  The randomness is back and then not (was +1 minute, then -1 minute, and now it is back)

jgyates, 2017-12-09:
Yeah, I remember chasing this one for a bit. The problem is that the value you write to the register is relative to the current time, which keeps changing. Thanks for testing.