jgyates/genmon #60: mymail.log - Multiple entries

USAFPride, 2017-12-23:
The mymail.log file is getting hammered with `Error selecting mail folder! (select)` entries every 15 seconds.  I'm using gmail and the messages are getting through to my inbox.  Any thoughts in pointing me the right direction would be helpful.

jgyates, 2017-12-23:
Based on your description I would assume your issue is that you do not have a folder in your Gmail with the same name as the incoming_mail_folder entry in genmon.conf. genmon.py is using mymail.py to receive email commands from your gmail account. Instead if reading your normal inbox you have the option of using an alternate email folder to process incoming mail. 

To resolve you have a copule of options:

Make the imap_server entry in mymail.conf blank. This will disable the receiving of email.

or 

Make a folder in your gmail that matches the incoming_mail_folder and processded_mail_folder entries in genmon.conf

Let me know if you have any other questions.

USAFPride, 2017-12-24:
That was the problem.  I didn't realize that the mail was also setup for remote control via email as well. I went back and re-read the instructions a few times to gain some clarity.  For a bit of simplicity, I have the following three recommendations:
1. Move all the mail config stuff to `mymail.conf` to keep all related items together.

> the name of the folder in the mailbox for searching for incoming email commands (required)
> incoming_mail_folder = Generator
> 
>  once an email command has been processed by genmon.py the email is moved to  this mailbox folder (required)
> processed_mail_folder = Generator/Processed

2. Revise the `README.md` and move the description of the mail `A note about email and security: ` to the Configuring mymail.py so all mail descriptions are in one spot.

3. Relocate `#disableemail = false` to the top of the `mymail.conf` file for ease of finding

As far as a feature request, it would be nice to have a "send mail" function only, for not only security, but it is an option that you may not want (and it doesn't cause errors if the file doesn't exist).

Again, this is a great program and thanks for the time you have put into this.

jgyates, 2017-12-24:
Glad to hear it is working for you. I made change 2 and 3 you mentioned above. I did not make change one as it will cause mymail.py to reduced functionality. Keeping the incoming and outgoing folder in out of mymail.conf allows mymail.py to be rescued by other applications to send and receive mail (I currently use it for two other apps to send and receive email). Thanks for you feedback. It is always good to here how we can make this project better.

closed 2017-12-24 by jgyates

USAFPride, 2017-12-24:
Maybe make a note in `mymail.conf` to point out that the folders in `genmon.conf` need to be set/verified?

jgyates, 2017-12-24:
I added the following text to the section on configuring mymail.conf:

Please note that if you use IMAP to receive email (i.e. allow genmon to receive commands via email) you will need to also set the incoming_mail_folder and processed_mail_folder options in genmon.conf.