jgyates/genmon #61: HTTPS with latest commit

sgtmaj27, 2018-01-01:
I just updated tothe latest version to try to take advantage of the added features.  Everything appears to work as it should except for the https.  I am sure that I am doing something wrong and not being well versed in the programming language I am floundering in my trouble shoorting the cause.  Here is what I have and done:

I added in the four remarks in the genmon.conf.  Usehttps= true. Useselfsignedcer=true, created cert.pem and key.pem and placed hem in the same directory as genserv.py.  Httpuser and http_pass both set.

When I access the web page, I get the web page with undefined in the data block.  The genserv.log has a total of six logs with multiple entries all indicating errorno 2 no such file or directory and errorno 98 address already in use.  I have tried useselfsignedcert= false as well and also tried just the ipaddress:443 with no luck.

Flask ver is 0.12.2

Can you direct me to a resource that I can use to help me diagnose my inability to get the https working.  I really only need a login for the web page not necessarily the https connection.

Thanks...Happy New Year
Foster

jgyates, 2018-01-01:
Hi @sgtmaj27,

Good to know about the browser cache. If there are any other issues you want to resolve now please reply back to this thread or open a new thread. Also, if you have errors in your log it is helpful to copy and paste them into the issue.

closed 2018-01-01 by jgyates

sgtmaj27, 2018-01-01:
Thank you for all your hard work on this project.  I am impressed and love
the functionality.  Felt good to “work” through my problems, albiet minor
to most.  Now on to learning Flask and HTTPS!

jgyates, 2018-01-01:
I took a closer look at genserv.py and I think I may of found an issue that would cause the key file and cert file not to be found if run from a cron job (for example reboot), but it would work if loaded in a normal context. I have moved the name and location of the key and cert file to genmon.conf. Now to use your own certificate you must set useselfsignedcert to False and specify keyfile and cert file as well. Let me know if you have any questions.

Thanks
Jason