jgyates/genmon #62: Hardware

rlaferla, 2018-01-03:
Does it work with a Generac Guardian Series 5875 - 20kW?

Also, I have a Raspberry Pi but would rather buy a completed device or a kit.  Anyone offering this?

jgyates, 2018-01-03:
Yes, the 5875 appears to use the Nexus controller so it is supported. I do not know of anyone offering completed kits.

madisdev, 2018-01-04:
Robert,

I am pursuing the idea of putting some preassembled kits together. How far is your generator from Wifi?

gardiner4, 2018-01-04:
I’d be interested too(Generac Guardian 5884). Wi-Fi AP is 10 feet away. I have extra Pi’s lying around I just not into making a cable.

madisdev, 2018-01-04:
So seems like we have two groups of people. Some who need whole kit, and some who just need the molex connector with leads. 

For those who just need molex with leads. Do you need dupont connectors on the other end? Would you like the max3232 board also included?

proppilot, 2018-01-05:
Not directly related to this topic but 'hardware' seemed like a good thread.  Is anyone having issues with low outdoor temperature operation ?  It is about -2 F here and the wifi (after 30 days up) is disconnecting (and the auto reconnect routine which works is not reconnecting) however I know the processor is still running (heartbeat LED on the enclosure).  I have a small utility routine that monitors the CPU and GPU temperatures (cannot see it now due to the disconnect) and as of last night when I could connect, the temps were 8 degrees C ( 46 F ) so its pretty cold !  Anyone else having wifi issues due to the cold ?

JeremyStevenson, 2018-01-05:
Hi, propilot. I just picked up an SR22 and working on my IFR right now. Fun stuff.

I have a generator that is next to a body of salt water. It's experience cold (0 degrees F). I just went to check and the Pi is not on my network. Unfortunately, I'm not local to it. DOH!

I don't know if this due to water entry with the extreme weather or wifi disconnect. 

Will the system reconnect when temperature comes back to normal? Or, is it going to require a reboot.

Looks like I need to set up a monitoring system to monitor my monitoring system.

gardiner4, 2018-01-05:
I would want the all the connectors and the max3232.

Are your Pi's covered in snow? Water/Ice easily blocks wifi. Pis are rated for -40 if I remember correctly. You might just need an external antenna.

proppilot, 2018-01-05:
Keep your airspeed UP !  The 22 has a slippery wing.  I own a Commander and fly IFR.  There is definitely some cold weather issues with the pi wifi so curious to see what pops up

proppilot, 2018-01-05:
Box is high and dry and I have an external antenna.  If I reboot the wifi is back.  

I know someone uploaded some code that reboots the Pi if wifi is lost.  Right now I just force a wlan restart.  I may try the reboot after x ping attempts that fail.

JeremyStevenson, 2018-01-05:
I found the following solution from https://tech.scargill.net/pi-zero-wi-fi-automatic-reconnect/

I have yet to try it because, sigh, I need to figure out how I'm going to reboot my non-wifi connected pi remotely now.

This can be solved with no extra scripts.

In /etc/network/interfaces put:

allow-hotplug wlan0
iface wlan0 inet manual
wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf
iface default inet dhcp

The roaming allows the interface to self-heal.

dprus, 2018-01-18:
@madisdev I'd like to purchase a kit as well (I assume by kit you mean the cable that connects to the Pi on one side and to the generator on the other side. Also assume it will use the Generac's pins to power the Pi?). I just had a Generac Guardian 7043 installed which I think has a Nexus controller and should work with genmon? My WiFi isn't far away at all (I would guess ~50ft although there is a wall in between).

kalex1114, 2018-02-07:
I would be interested in a kit as well.  no need for the actual pi but other stuff

gardiner4, 2018-03-02:
Can somebody make me a cable?

madisdev, 2018-03-07:
I am getting in some parts on Friday. I will see what I can get done this weekend.

kalex1114, 2018-03-07:
Nice.  let me know

gardiner4, 2018-03-07:
Great.

madisdev, 2018-03-10:
I'm just gathering some info here, I have parts, still figuring out logistics.

What kind of length of cable do you need from the molex connector?
My initial thought is to solder the cable from molex cable onto the MAX3232 connector, but was thinking if you are going to be putting this in a project box that's sort of problematic.
I could just put [dupont ](https://goo.gl/images/dbZBzJ)connectors on the end of the molex cable that way you can feed them easier through a hole. Thoughts?

kalex1114, 2018-03-10:
i would say maybe a foot maybe 18" to give it some room.  I'm planning of attaching the waterproof box where mobile link currently is.  You have a good point on the dupont connectors.  it will be hard to put it into the box with max3232 connector on it.

proppilot, 2018-03-10:
> <reply+02039b7ac1959452ae3956f845495c520fc1d4e783088b7992cf0000000116bc01609
> 2a169ce110856ce@reply.github.com>
> <comment@noreply.github.com>

My box is mounted where the MobileLink would have been.  I notched the box
at the bottom to allow the cable to lie in the notch and then put the lid
on.  That way I did not have to pass the connectors through a big hole.

hYXIQ9Wks5tdDbggaJpZM4RSCuL> .

  
b","name":"GitHub"},"entity":{"external_key":"github/jgyates/genmon","title"
:"jgyates/genmon","subtitle":"GitHub
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
-9aed-b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/jgyates/genmon"}},"updates":{"snippets":[{
"icon":"PERSON","message":"@kalex1114 in #62: i would say maybe a foot maybe
on it.  "}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/62#issuecomment-37206
3364"}}}

madisdev, 2018-03-11:
@kalex1114 How will you make connections to PI, do you have headers? Regular dupont connectors will do?

kalex1114, 2018-03-11:
@madisdev i have headers. regular dupont connectors will do

dprus, 2018-03-14:
@madisdev Right now I just have a Generac 7043 and a Raspberry Pi 3. Haven't really put any thought yet into where I'm gonna put the Pi. Ideally I guess I would just like to find a box that I can fit inside my generator housing to hold the Pi and I would like the Pi to be powered by the generator. Hoping you can make me up a cable to satisfy my needs - I'm open to any suggestions / recommendations you have on the best way for me to get it all going. Thanks!

lqallison, 2018-03-26:
Just getting started. I am looking at using a usb port on the Pi. Using a usb to DB9 adaptor and port 
/ttyUSB0. Gets an ok  if I loopback  Tx-Rx on DB9 connector.  What about adding current sense in Genmon? How would the read outs be added to Genserv.py? Also indicators for switch position looking at aux contacts on on switch?  I have 5875 and Nexus controller. Just had stepper motor failure during outage.

jgyates, 2018-03-26:
@lqallison, Are you asking about displaying the current (amperage) load on the generator? If you are we are in the process of adding  that to genmon, the first release, assuming testing goes well, will be for Evolution controllers. I need to do some research to determine if we can support this on Nexus. This will be displayed both graphically and text in the web interface. 

Also, USB ports work fine. I used a few in my initial development. Let me know if you have any other questions.

Jason

lqallison, 2018-03-27:
yes, I want to put current transformers on the lines to breaker box and be able to read them whether or not the generator is running. Also math could be done to give a wattage rating. I don't see any current sense wires going into Nexus controller. So I figured split core current transformers were the way to monitor load and they would go directly to the Pi to be read. Not talking to Nexus.

jgyates, 2018-03-27:
That could work, you may be better off with an adreno instead of a pi if you use your own CT since the adreno has built in AtoDs. and the pi does not, you would need a daughter card on the pi. Most 
CT are analog out.

I had thought about creating an energy monitor or using one like openenergymonitor.org.
I have not looked into the nexus for CTs yet. In the evolution there are 2 CTs that help with the over load protection. Once I get the CT stuff working and released on the Evolution I am looking for volunteers to help me test to see if I an get it working on Nexus as well.

Now I am reading the CT info in the Evo controller and it appears to be working (+- 3A on a 200A generator). Once you get the CT (your own or from the controller) getting kw is just V*A for single phase. Reading it from the generator is probably not as precise as having your own CTs but it is pretty close since I don't know the step value of the A/D. I am basically just measuring the current with a clamp on meter and taking a register reading to reverse engineer the A/D step value.

I assume there are 2 CTs internally in the Nexus as well, although there is a chance that they are not exposed via the serial modbus link.

lqallison, 2018-03-27:
I am willing to be a test station.  I have not purchased the CT's yet and just ordered the Molex connectors today. Hope to have something online next week.

jgyates, 2018-03-27:
Sounds good. That will work out well timing wise for me. Basically I will just need you to run a few test with the software, and make a measurement of your current with at clip on meter if you have one, then I will code something up and you can test it out. I can force genmon to think it is running on a Nexus on my controller but that only gets me so far in testing so having people to help test is very useful. Thanks

jcherney, 2018-04-01:
@madisdev  I am also interested in an assembled finished kit if available. I have a Nexus controller. Is this something you might be able to offer?

lqallison, 2018-04-05:
Does anybody have the pinouts for the Nexus controller port? I have seen where the RS232 connections are, just wondering what the other 5 pins do. Going to use shielded 6 conductor wire to run RS232 into the change over switch where I will locate the Raspberry Pi and CT's  and have 3 left over wires for other things. I could use a manual on the controller/ control panel. Also I have two neighbors with the Evolution controller, so I would like the same information for it too. I am going to set them up with this setup also.

jgyates, 2018-04-05:
The pin out on the aux plug on the back of both the Nexus and Evolution is here:
https://github.com/jgyates/genmon/wiki/Making-a-Cable

I don't know if what the other pins are (other than RS-232 tx and rx, power and ground)

Look in this doc has some additional info:

http://www.generator-parts.com/manuals/home-standby/0H9172.pdf

On page page 37 (as printed on the bottom of the page) for some info on the connector.

kalex1114, 2018-04-05:
Did anyone try to use cable kit and enclosure from the mobilelink?  my subscription is expiring soon and was thinking if its possible to reuse these

jgyates, 2018-04-05:
The cable will work on the controller end and the end that connects to the battery, you will need to figure out a way to mate it to your pi, the cable is just 2 connections (tx and rx) from ML to the controller, and  power and ground to the ML come from the battery. I did not reuse my cable but I did examine it to make my initial cable.

lqallison, 2018-04-11:
Got Genmon working on Nexus controller. Got it working in time to watch the  weekly exercise. Battery voltage, utility voltage have readings with gen not running. On exercise output, frequency, rotation showed the normal lower readings of the exercise state. Pi is located in garage 25 ft from generator. Cable is 6 conductor shielded 18 awg. use 3 wire for rs232, 2 data 1 ground and 12 volts from the battery to power pi thru 12v to 5v usb connector. Maint. page does not show serial number, exercise time shown and can be set by window, total hours correct. Controller Hardware Version : V1.02  Firmware Version : V1.43.  Generac 5875 20 kw (18 Natural Gas) 107 hrs installed Feb 2013
 Generator Monitor Stats : 
        Monitor Health : OK
        Detected Controller : Nexus, Air Cooled
        Run time : Generator Monitor running for 0:00:02.
        Generator Monitor Version : V1.5.10
Would like to be able to block email password from being seen on review of setup page. Working to setup current monitoring for each incoming line. Going to use Explorer Hat for analog to digital conversion.
Software written in Python for testing now, need to integrate into Genmon. Will be test subject for 
what ever you want to do.

jgyates, 2018-04-11:
Hi @lqallison 

Adding a non visible pw to the email password is an easy thing. Will add that soon.

If your Serial Number is not showing up it is mostly likely due to a controller being replaced. Replacement controllers controllers do not have a serial number. If this is not the case let me know.

First step in helping out would be to submit your registers in this thread:

https://github.com/jgyates/genmon/issues/10

lqallison, 2018-04-11:
This is original controller. serial # 7434429
Registers : 
    Num Regs : 162
    Not Changed : 220598
    Changed : 46648
    Total Changed : 0.21

    Base Registers : 
        0038 : 0001
        0034 : 0000
        0037 : 1195
        0033 : 0064
        0032 : 2e6b
        05f5 : 03ec
        05f4 : 03ec
        05f1 : 03ec
        01f3 : 120a
        01f1 : 000e
        05ed : 000e
        023b : 0000
        023e : 0000
        003b : 0000
        05fa : 0000
        0059 : 0000
        0058 : 0000
        0053 : 03ec
        0052 : 03ec
        0057 : 0000
        0056 : 0000
        0055 : 000e
        0054 : 0000
        0208 : 03ec
        005c : 03ec
        005a : 03ec
        005f : 03ec
        005d : 03ec
        003c : 0000
        000d : 0000
        000e : 1024
        000f : 040b
        000a : 008e
        000c : 006b
        0005 : 0c0f
        0006 : 0301
        0000 : 0003
        0001 : 00000000
        0008 : 0000
        0009 : 00ed
        01f2 : 0101
        05ee : 000e
        0012 : 0000
        0011 : 009c
        0010 : 0312
        0007 : 0000
        0019 : 0004
        002b : 0000
        002c : 0000
        002a : 668f
        002f : 0000
        002d : 0000
        002e : 0000
        0020 : 465c
        0021 : 013f
        001f : 00a0
        001e : 0077
        001d : 465c
        001c : 006e
        001b : 465c
        001a : 00a0
        01f4 : 30303037343334343239

jgyates, 2018-04-11:
@lqallison , Please post to the appropriate issue topic thread. This thread is for pre-assembled cables. This makes it difficult for me to respond and keep up if all of the topics are mingled together. If you are still not seeing your serial number open a new issue and I will be happy to help.

dprus, 2018-04-13:
Speaking of pre-assembled cables, do we have anyone supplying them yet? :)

-Richard.

kalex1114, 2018-04-24:
@gardiner4 nice.  Did you use battery connection or did you get power from evolution port?

gardiner4, 2018-04-24:
I used the battery to USB connector. No problem with Wifi signal.

jcherney, 2018-05-06:
Wondering if someone can assist in loading the program onto my Raspberry PI3. I can make cables. Everything else is built-just can get this program on my PI3. 
Thanks in advance.

kalex1114, 2018-05-06:
@jcherney i can help out if you want.  email me my userid @ gmail and i can help you

zentec, 2018-07-14:
I didn't want to open a new issue, as this concerns the cable.  Does anyone have access to documentation that details the other 6 pins of the AMP connector?  My new generator (16kW air cooled) has pins 1,2,5,6 in use.

It would be nice to know what I'm going to disturb when I start messing with the connector.

zentec, 2018-07-15:
@jgyates 
It is an Evolution 2.0, and these wires are for the wifi link.  Initially, I thought they went into box for transfer switch interconnect, but they pass through the box wall and to the house side to the wifi module.

I can provide photos if you like.

jgyates, 2018-07-15:
Photos would be great. I will start a new thread related to Evolution 2.0. I have heard some things about it but I don't know much that is conclusive.

https://github.com/jgyates/genmon/issues/149

KortP, 2018-07-26:
I’m am also looking to just buy a unit assembled and ready to go. If it works well I have a market to sell more.

milton100, 2018-08-16:
I am also interested in a complete ready to go device with cables. I was thinking a 3b+ pi paired with either a poe tophat or a Gigabit-Ethernet-Active-PoE-Splitter-5V-Micro-USB-for-Raspberry-Pi-3B-3B-Plus. I was also hoping to start a dialog with jgyates off of GitHub about this project and provide pictures and measurements. (emkellamjr@gmail.com) As I am old now and suffering from the aftermath of several events I am no longer able to tinker like I used to and want to, but this project has great potential.

KitRo421, 2018-08-20:
You can add me to the list of those that would purchase it pre-assembled!

swvalenti194, 2018-09-17:
I am interested in a pre-assembled unit for a 2017 model 16kw with evolution controller.  Can anyone help me out?

liltux, 2018-09-18:
https://www.ebay.com/itm/263943699203
This is for me to do as time permits.  I left as OBO, so if you need just a cable you can make offer and we can discuss your specifics privately through ebay.  Also I am no business so it will take time for me to assemble and test any built components.  This is just me trying to help out and as you can see I will donate 50% of bottom line, minus ebays contribution back to the project.  If anyone is interested.  This is wifi version only.

swvalenti194, 2018-09-18:
Sir, thanks for reaching out.  I can definitely build the cables and hardware but when it comes to loading the program on the RPI and doing scripts etc that's where I don't know without being coached.  I assume I access the webpage via the static IP I assign to the RPI then enable port forward to access via WAN?  I will gladly donate if you can help me with that?  Also below is a list of the components I got on the way for my build minus the enclosure...that look about right?  My unit is 3ft away from my house and when I place my phone in the generator and run a speedtest my strength is good and I get about 180mbps on 5g band so I don't think I will need the external antenna.  

1 x Hammer Header Male - Solderless Raspberry Pi Connector
1 x Raspberry Pi Zero W
1 x Row of Breakaway Headers
1 x ATX-8 Pin Male
1 x SparkFun Transceiver Breakout - MAX3232 
1 x ATX - 4 Pin Female
1 x ATX - 4 Pin Male

liltux, 2018-09-18:
If you are good with assembly, Jason here has a real good wiki follow along.  Between the wiki and raspberry pi .org documentation pages it should almost walk you directly through setup.  Looks like you have most of what you need to get going.  Check out the thread #31 for some pictures on some different setups for more ideas.

swvalenti194, 2018-09-27:
Couple questions, I setup the Pi Zero W and installed genmom by using the .git command.  I am returning the Pi Zero W cause it can't connect to 5ghz and replacing it with a 3b+.  If i drop the already configured card into new Pi will that work?  Other than running the git for genmon and changing serial port is there anything else I need to do? Also I didn't connect the MAX3232 yet cause they sent me wrong one, is that why I can't change the serial port?  Keep getting errors.

jgyates, 2018-09-27:
I am not 100% sure, but my guess is it will not work. I know that you can swap SD card between ***some*** pi models, but it is very specific and you also risk the OS setting up some things specific to the ZeroW and leaving out other things for the 3b+ during the first boot. Of course it is easy enough to try it and see if it will work.

jgyates, 2018-09-27:
@swvalenti194 

The pins are the same. While 5V will work, it is recommended to use 3.3V (pin1) to power the MAX3232 regardless if the raspberry pi variant.

swvalenti194, 2018-09-27:
Thank you sir

swvalenti194, 2018-09-27:
jgyates can you pm with your email?

jgyates, 2018-09-27:
github does not support PM. You can reach me at the alias generatormonitor.software@gmail.com

liltux, 2019-02-01:
@jgyates , I just wanted to put something on here about the Ebay listing.  I am working on redoing my Ebay listing and once the one that is up is done, i.e. sell the last one on that listing.  I will be putting up a new listing.  But I would also like to point out if anyone is interested in a pre-built unit or needs a cable made up, etc. Feel free to contact me at liltux (at) liltux dot com.  Through Ebay's system there is little I can do about communication and I do not want to incur extra fees to have a listing on Ebay.  Ebay does not allow exchange of contact information between seller and buyer.  On one-ofs such as cable making I do not have a set fee, just need to discuss what is needed and come up with a solution together.  Thanks.  liltux.

jgyates, 2019-02-01:
Thanks for the info @liltux !

[7 more replies]

closed 2019-07-06 by jgyates