jgyates/genmon #63: USB to TTL

mijairaf, 2018-01-05:
Sorry if this is the wrong forum to ask a question but I could not find an alternative.

Can a USB serial to ttl device be used in place of the current configuration ? For example this device ?
 https://www.amazon.com/USB-TTL-Adapter-Interface-Development/dp/B075N82CDL/ref=sr_1_5?ie=UTF8&qid=1515093021&sr=8-5&keywords=usb+serial+ttl 

Or one of the many from China on Ebay for a couple of bucks or lower ?

jgyates, 2018-01-05:
Hi @mijairaf, 

If you are asking if you can use the item in the link to connect a Pi to the generator controller, the answer is no. This device connects USB to TTL level signals. You could use a USB to RS-232 device and connect the USB end to the Pi and the RS-232 end to the controller. 

I have not used this exact model, but based on the description this should work:

https://www.amazon.com/dp/B072KJSS5C/ref=sspa_dk_detail_3?psc=1&pd_rd_i=B072KJSS5C&pd_rd_wg=Nd6FS&pd_rd_r=VZ5RZZSGJ0JP5JMS1AYK&pd_rd_w=UqGO1

Jason

jgyates, 2018-01-05:
I updated the Hardware wiki page with this link and added a description to help find alternatives.

Regarding finding an appropriate RS-232 to TTL converter:
"The critical thing here is that the converter must support RS-232 to 3.3V TTL. If it supports RS-232 to 5V TTL only then it will not work with the Pi."

mijairaf, 2018-01-05:
> 3?psc=1&pd_rd_i=B072KJSS5C&pd_rd_wg=Nd6FS&pd_rd_r=
> VZ5RZZSGJ0JP5JMS1AYK&pd_rd_w=UqGO1

Hi Jason,

 Thanks for the quick response. I am not a hardware guy by any means.
  What is the difference between the two devices. I thought the USB FTDI
TTL device presents itself to the linux os (python) as /dev/ttyUSB0 which
is like a rs232 port ?  These are the devices that I've used to interface
to esp/ardunio boards. I guess I need to learn a bit more about
microprocessors.

Thanks Again

jgyates, 2018-01-05:
The PI has a built in serial port that exposes TTL level signals. TTL can be either 3.3 volt or 5 volt signals. The PI uses 3.3Volt TTL signals as the external interface to it's serial port.  The generator controller does not accept TTL signals, it needs RS-232 level signals (between 7volts and 12 volts)

You can add another serial port via the USB interface. They make USB serial ports that have an external interface of TTL or RS-232,  However, the original link you provided will add a second serial port with TTL level signals, which is what the PI has already. The issue in using the device you provided a link to is not the PI, it is the generator controller. The controller expects RS-232 level signals (typically between 7 volts and 12 volts).

As a side note, you can use the the device in your link to interface a PI to a PC or any other device that has a USB port. You would connect the TTL interface on the device to the TTL serial interface on the PI, then connect the USB serial port to a PC, for example. But, if you want to connect to the generac controller, you need RS-232 level signals.

Let me know if you have any other questions.

Thanks
Jason

mijairaf, 2018-01-05:
Thanks for explaining the differences.

 And thanks for making genmon available.

closed 2018-01-05 by jgyates