jgyates/genmon #64: Add JSON support?

madisdev, 2018-01-07:
Jason,

Did you ever consider returning data from genmon in a more structured format, specifically JSON? I think it might open up the app a little more for consumption from other clients. I think it also might make modifications to the webpage a little easier.

I looked at the code and it seems like its pretty baked in how you build the strings, but I don't think its impossible to add an option to return JSON.

I'm not asking you todo the work, just want your opinion. Thanks.

PS I think we could also add a PI CPU temp measurement to the webpage, to monitor how frozen our generators are getting.

jgyates, 2018-01-07:
Hi Madison,

That is a good idea.  No parsed data, only register values, are stored in genmon so it may be best to just create separate routines that dump JSON formatted data. This approach would allow the support to be worked in without disrupting the current functionality.

One approach would be to use the "registers" command and have an external python module parse the data into JSON format. I have thought about moving all of the string parsing into a separate module and just make genmon only monitor the registers. This method would fit better into that approach. 

The CPU temp is also a good idea, we would just need to make sure that it is optional as there are some who are not using the PI and I would rather keep this as open to other platforms as possible.

In short, my only real concerns are that we not break existing functionality, we are mindful of the maintainability and open to other platforms.
 
Thanks
Jason

proppilot, 2018-01-07:
> <reply+02039b7af048aea0ee6a7520075f98ec9303a7268ff8b8aa92cf00000001166a2cd29
> 2a169ce1115035c@reply.github.com>

Here is a script I use for some housekeeping monitoring of temp, wlan and
disk 

#!/bin/bash

# Script: genmontemp.sh

# Purpose : Display the ARM CPU and GPU temp of the Raspberry Pi 3

# _________________________________________________________________

count=1

while :

do

 cpu=$(</sys/class/thermal/thermal_zone0/temp)

 echo "$(date) @ $(hostname)"

 echo _____________________________________________

 echo "GPU => $(/opt/vc/bin/vcgencmd measure_temp)"

 echo "CPU => temp=$((cpu/1000))' C"

 echo "  "

 echo "Uptime $(uptime)"

 echo " "

 echo "WLAN status"

 iwconfig wlan0

 echo " "

 echo ____________________________________________

 echo "Disk Status"

 df -h

 echo " "

 sleep 1

 clear

done

TK00zKfks5tIRDSgaJpZM4RVvh0> .

  
b","name":"GitHub"},"entity":{"external_key":"github/jgyates/genmon","title"
:"jgyates/genmon","subtitle":"GitHub
repository","main_image_url":"https://cloud.githubusercontent.com/assets/143
418/17495839/a5054eac-5d88-11e6-95fc-7290892c7bb5.png","avatar_image_url":"h
ttps://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6
-9aed-b52498112777.png","action":{"name":"Open in
GitHub","url":"https://github.com/jgyates/genmon"}},"updates":{"snippets":[{
"icon":"DESCRIPTION","message":"Add JSON support?
(#64)"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/64"}}}

jgyates, 2018-01-07:
For what it is worth, this is how I monitor temp and other things on my pi (and everything else):

I use [Nagios](https://www.nagios.org/) to monitor all of my computer / network equipment. Attached is a screen shot of the things I am monitoring on the pi attached to my generator:

[image]

Nagios will notify me via email or mobile app if anything is out of the expected range. I have another [pi that host Nagios](https://community.spiceworks.com/how_to/68159-install-nagios-on-a-raspberry-pi), but it could be hosted on the same pi.

ColtB45, 2018-02-25:
@madisdev I really like the JSON idea!

Regarding the temp and _other_ monitoring, I don't know. I kind of think that's better left to other utilities. It's well outside of genmon's core functionality, but I do understand the desire to have it all present in a centralized place.

jgyates, 2018-03-01:
Just and update: I have stared the process of adding JSON support. At best it will be another week, depending on the spare time I have. I will likely roll this out in phases. Base genmon.py changes first, then genserv.py and genmon.js (web UI) second. javascript is not my best language, so that may go a little slower.

madisdev, 2018-03-01:
Oh I didn't mean for you todo all the work I can contribute if you start it.

jgyates, 2018-03-01:
No worries, I did not take it that way. One thing that I am not sure about that you may can help with: it has to do with the javascript component. Currently genmon sends over the requested data in ASCII and the javascript mangles it to be more HTML friendly (replacing spaces and CR/LF with HTML like items), the java script. I was wondering it it would be better / easier to maintain, etc, to send JSON instead of ASCII. The "plan" with genmon.py was to make the core output components (status, maint, outage, logs, monitor) have the option of outputting JSON in addition to ASCII. I have converted status and maint over so far. I am making most everything internally that deals with output use python dictionaries and lists and then converting them to JSON or ASCII depending on what is requested. I am attaching the output of the "status" command as printed from my working copy. This will give you and idea of the basic format of the JSON data. If it turns out we should leave the web UI alone, that is fine too. I just have not got around to researching that yet as my experience is not with javascript / HTML. The short version of the question is this: does it make sense to use JSON in the web app or is it fine as is? Let me know if you have any questions.

[out.txt](https://github.com/jgyates/genmon/files/1769741/out.txt)

madisdev, 2018-03-01:
Currently we are just dumping the text to a html page, I have some experience with Angular 4 and was thinking I might give it a shot to render the JSON result using databinding in the page. Obviously it would make the HTML page a little heavier. Ill do some research to see if anyone is hosting angular page using the Flask framework.

It's a good scenario because we already have something that's working, trying something new shouldn't cause any issues.

jgyates, 2018-03-01:
Another question I have is this: If I return a JSON string via json.dump() python library call in genmon.py through to the python Flask app, do I need to call the jsonify() function as with other calls in genserv.py? Presently I am assuming that I do not need to call jsonify() for data that should already be JSON formatted.

madisdev, 2018-03-02:
Looks like jsonify is used by flask to return a correct Json http response. Json.dump is just a generic method to make a Json object from string.

jgyates, 2018-03-03:
OK. I have completed the JSON support in genmon and genserv. It is checked in and I tagged a new release. I added a new page to the web UI that uses JSON. The link to the new page is at the bottom of the web UI page. The new page is mostly just for debugging / reverse engineering functionality so I made it low profile. It displays the registers of the generator and highlights changes in the registers as they occur. The new page is not the best looking thing to look at, but it is useful for technical reasons. Once you install the update you can reach the new page by the link at the bottom of the web UI labled "Generator Registers".

Implementing this gave me a chance to clean up some of the code that deals with string parsing and output. Now most data that was output was just a lot of string manipulation. Now the data is stored in python dictionaries and either converted to strings or JSON before outputting, which is easier to maintain and better interoperability, so thanks for making this feature request.

I am not 100% convinced we need to update the web interface to fully use JSON just yet, but I am open to ideas if anyone has any. In general the web interface is fairly easy to maintain now. If JSON can make this easy to maintain and not complicate the setup for people then I am all for it. 

Let me know if anyone has any issues with the new release or questions on the JSON implementation.

proppilot, 2018-03-03:
> <reply+02039b7a53fe7a2f9beca3ba7dd39778d5932c7bf5442fe492cf0000000116b1c42d9

Hi Jason,  there have been some terrific upgrades and development done.
With a stable platform (including wifi link break detection and restart) I
now have 39 solid days of running so I am (quite) a bit gun shy to mess with
success.  I know you have posted some updates on how to do the update so my
question is this.  Is there an easy way to clone the SD card that is in my
PI and then pop in the clone and play with it with a backup (revert to the
original SD card should I mess it up ?  Is it a simple copy of all files (I
am on a MAC platform) or are there hidden files that need special handling ?
Sorry, a bit off topic but maybe an interesting topic for others Leary to
mess with a stable operating GENMON !

Date:  Friday, March 2, 2018 at 9:02 PM
Cc:  Anthony Varga <tonyvarga@mail.com>, Comment
<comment@noreply.github.com>

TcAuytpks5tafmtgaJpZM4RVvh0> .

  
"icon":"PERSON","message":"@jgyates in #64: OK. I have completed the JSON
Registers\".\r\n\r\nImplementing this gave me a chance to clean up some of
thanks for making this feature request.\r\n\r\nI am not 100% convinced we
setup for people then I am all for it. \r\n\r\nLet me know if anyone has any
implementation.\r\n\r\n"}],"action":{"name":"View
Issue","url":"https://github.com/jgyates/genmon/issues/64#issuecomment-37010
7100"}}}

proppilot, 2018-03-03:
Sorry, should have asked the Google first.  Found a nice utility for free
called PiCloner that does exactly what I wanted.  Thanks.

jgyates, 2018-03-03:
Cloning ther SD card works, you can also just make a copy of your genmon directory before you update. You can also use the git command to revert to any tagged branch. I tag all the major releases. Release tags are found here https://github.com/jgyates/genmon/releases.

I have cloned SD cards before (I use a Mac also). This article tells how to do it with the built in command line tools on a mac. Of course the GUI you mentioned should work too.

https://computers.tutsplus.com/articles/how-to-clone-raspberry-pi-sd-cards-using-the-command-line-in-os-x--mac-59911

Upgrades are always optional, but I think this one is pretty safe. The core serial logic is unchanged. I have been running it on my system for a couple of days. If you want to play it even safer, wait a week or two then upgrade to see if any issues come up.

jgyates, 2018-03-15:
As a final note on JSON support. The web UI now uses exported JSON data from genmon. Also, for folks who want to develop applications around genmon this wiki has some starting information:

https://github.com/jgyates/genmon/wiki/Interfacing-GenMon-to-External-Applications

closed 2018-03-15 by jgyates