jgyates/genmon #65: Monitor Health: Not receiving data  =(

GitHubUserToo, 2018-01-23:
Hello,

First, thank you for a great project!!  More than I ever expected to find.

Unfortunately, on first setup, on the Monitor page I get:  "Monitor Health: Not receiving data."

/var/log/genmon.log continues to produce like:
 2018-01-22 20:15:49,809 : Error: timeout receiving slave packet for register 415 Buffer:8
 2018-01-22 20:15:49,821 : Data Error: CRC check failed: 0062  9d62
 2018-01-22 20:15:49,822 : Error Receiving slave packet for register 416

I have a Generac 22KW and followed the Wiki and other posts for making a cable and setup.

/etc/genmon.conf set
port = /dev/serial0 (and tried ttyAMA0)

/boot/config.txt set
enable_uart=1
dtoverlay=pi3-disable-bt

serialtest.py returns:  PASSED! Loopback successful

pi@rpi-genmon:~ $ ls -l /dev/serial*
 lrwxrwxrwx 1 root root 7 Jan 22 20:13 /dev/serial0 -> ttyAMA0
 lrwxrwxrwx 1 root root 5 Jan 22 20:13 /dev/serial1 -> ttyS0

 sudo systemctl stop serial-getty@ttys0.service
 sudo systemctl disable serial-getty@ttys0.service

Removed "console=serial0,115200" from /boot/cmdline.txt

I have tried swapping rx and tx, pin 08 and 10 on the PI3, which did not produce any different results.

I connected the Molex pins with a jumper and tested continuity at the RS232 - no problem.

I'm concerned maybe have the serial setup wrong, or some wiring issue.  Any suggestions are much appreciated!

[image]
[image]
[image]

krisasmith, 2018-01-23:
Are you getting power for the pi from the generator battery?

GitHubUserToo, 2018-01-23:
I see that now in your documentation - my bad. :(

Pi is powered inside the house.  I connected molex #2 to the RS-232 #5 GND and now no more CRC errors. Monitor Health: OK :) Wow - had no clue about ground like that!

Thank you!

closed 2018-01-23 by GitHubUserToo